AllskyTeam/allsky #484: help mode is broken

ckuethe, 2021-09-16:
I expected that `-h` would print a help message, then exit with no other side effects. Additionally, help mode should not require any correct configuration - I might be running help mode to figure out how to configure the camera, right?

Unfortunately, when I do that on my actual allsky camera, it interferes with capture, causing the running process to get `ASI_TIMEOUT`

```
$ ./capture -h

 ******************************************
 *** Allsky Camera Software v0.8 | 2021 ***
 ******************************************

Capture images of the sky with a Raspberry Pi and an ASI Camera

Add -h or -help for available options

Author: Thomas Jacquin - <jacquin.thomas@gmail.com>

Contributors:
-Knut Olav Klo
-Daniel Johnsen
-Yang and Sam from ZWO
-Robert Wagner
-Michael J. Kidd - <linuxkidd@gmail.com>
-Chris Kuethe

*** ERROR: No Connected Camera...
scripts/copy_notification_image.sh: line 11: /config.sh: No such file or directory
scripts/copy_notification_image.sh: line 12: /scripts/filename.sh: No such file or directory
scripts/copy_notification_image.sh: line 13: /scripts/ftp-settings.sh: No such file or directory
*** copy_notification_image.sh: ERROR: File 'notification_images/NotRunning.' does not exist or is empty!
     ***** Stopping AllSky *****
```

```
$ ./capture -help

 ******************************************
 *** Allsky Camera Software v0.8 | 2021 ***
 ******************************************

Capture images of the sky with a Raspberry Pi and an ASI Camera

Add -h or -help for available options

Author: Thomas Jacquin - <jacquin.thomas@gmail.com>

Contributors:
-Knut Olav Klo
-Daniel Johnsen
-Yang and Sam from ZWO
-Robert Wagner
-Michael J. Kidd - <linuxkidd@gmail.com>
-Chris Kuethe

*** ERROR: No Connected Camera...
scripts/copy_notification_image.sh: line 11: /config.sh: No such file or directory
scripts/copy_notification_image.sh: line 12: /scripts/filename.sh: No such file or directory
scripts/copy_notification_image.sh: line 13: /scripts/ftp-settings.sh: No such file or directory
*** copy_notification_image.sh: ERROR: File 'notification_images/NotRunning.' does not exist or is empty!
     ***** Stopping AllSky *****
```

```
$ ./capture --help

 ******************************************
 *** Allsky Camera Software v0.8 | 2021 ***
 ******************************************

Capture images of the sky with a Raspberry Pi and an ASI Camera

Add -h or -help for available options

Author: Thomas Jacquin - <jacquin.thomas@gmail.com>

Contributors:
-Knut Olav Klo
-Daniel Johnsen
-Yang and Sam from ZWO
-Robert Wagner
-Michael J. Kidd - <linuxkidd@gmail.com>
-Chris Kuethe

*** ERROR: No Connected Camera...
scripts/copy_notification_image.sh: line 11: /config.sh: No such file or directory
scripts/copy_notification_image.sh: line 12: /scripts/filename.sh: No such file or directory
scripts/copy_notification_image.sh: line 13: /scripts/ftp-settings.sh: No such file or directory
*** copy_notification_image.sh: ERROR: File 'notification_images/NotRunning.' does not exist or is empty!
     ***** Stopping AllSky *****
```

EricClaeys, 2021-09-16:
Thanks for pointing it out.  I'll fix in the next batch of updates I make.
I agree that capture.cpp and capture_RPiHQ.cpp should exit after displaying the help message.

Is the standard to accept either "-h" or "--help"?

ckuethe, 2021-09-16:
Not just exiting after help, but doing so in a way that does not have side-effects. What if the user is running some other program that's using the ZWO camera (eg. ASIStudio) - we don't want to disturb that.

I'm used to at least `--help` doing something sensible. Some tools might use `-h` to specify hostname, and might interpret `-help` as the same as `-e -h -l -p`. If it's not too much hassle, it would be user-friendly to support all of `-h`, `-help`, and `--help`.

In terms of argument parsing rework, do we want to support any short form arguments (single letters), or just long form (--histogram-box, --angle, --cooler-enabled, --small-font-color, ...)? I'm thinking long form only.

EricClaeys, 2021-09-22:
I don't see a problem with supporting only long form arguments.

PR #499 (2021-09-28): Update capture.cpp
Fixes #484.  Fixes #498
Adds camera ID (for USB 3 cameras) and camera serial number to log output.  These will be used in a future update that has different settings for different cameras.
commit 1f755cc9 (2021-09-28): Update capture.cpp

Fixes #484.  Fixes #498
Adds camera ID (for USB 3 cameras) and camera serial number to log output.  These will be used in a future update that has different settings for different cameras.
closed 2021-09-28 by ckuethe