AllskyTeam/allsky #485: Weather and SQM data overlay integration

Mouzarty, 2021-09-16:
First of all thanks for the amazing job you’re doing! This software is incredible as it is. 

I’m wondering if integrations for weather data (ASCOM compliance devices and/or OpenWeatherMap) and SQM data to add as an overlay on the images is on the roadmap? Would be amazing to have all this in the same solution.

linuxkidd, 2021-09-16:
I've actually achieved this with NodeRED ( to pipe weather data to a json file ) and some python code that uses ImageMagick to do the overlay.  You can see a sample on my [public allsky site](https://allsky.linuxkidd.com/).  In the newest WebUI, there's an option to provide a text file to overlay.  While I've not tried that new feature (since I already solved it), it may be able to provide what you're looking for if you can get the data to a text format.

EricClaeys, 2021-09-17:
@Mouzarty, you can do that now if you can get the data into a text file on a regular basis; in fact, some people already do this.
See the "Extra Text File" option in the GUI or the "extratext" and "extratextage" options at [https://github.com/thomasjacquin/allsky](url).
Basically, you create a text file with one or more lines, then tell the software to display that file.
We don't have any plans to build that capability directly into the software though - there would be too many options and it would take a fair amount of time.

If this answers your question, please close the issue.

closed 2021-09-17 by EricClaeys