AllskyTeam/allsky #486: RPi4 + RPi HQ camera - high CPU load?

rodrigoluk, 2021-09-17:
I just want to report on an interesting issue with RPi4 + RPi HQ camera. The default camera setting after a fresh install has:

"nightdelay":"10" # which is 10ms!

For some reason the PI4 goes into a high CPU load (ut above 4) and a kernel worker process stays at the top of the processing table. I think this is basically managing raspistill and it is basically running all the time. Day or Night. The process is something like:
kworker/u8:3-brcmf_wq/mmc1:0001:1

Through trial and error I found that changing the nightdelay to a value above 1 second:

"nightdelay":"2000" # which is 2 seconds

gets rid of the problem and the Pi4 goes back into more normal load (ut below 1).

I don't understand why this is happening. My hardware is running the latest firmware, which I update if needed with:

sudo rpi-update

EricClaeys, 2021-09-20:
@rodrigoluk,
A lot of people who use the software like a very short night delay since they are looking for meteors and other such things and don't want to miss any due to a long delay.  For people like me (and you?) that only want to see what the cloud coverage is at my observatory, a much longer delay is appropriate.

I have also seen high CPU usage with kworker but never figured out why and don't know what kworker is (it's not in $PATH).

I try to reproduce what you're seeing.

robmorin, 2021-09-20:
Hey @rodrigoluk 
What lens are you using with the Pi HQ Camera? I cant really find any fish eye ones. 

Thanks, and have a great day!

rodrigoluk, 2021-09-21:
Hi @EricClaeys,

Thanks for looking into this issue 😊 The kworker (kernel workers) are not in the $PATH as they are not binaries the user calls. These are spawned by the kernel itself (as I understand these but I could be wrong). I understand some people really want continuous frames without delay but 10ms is too aggressive for some reason on my setup.

Thanks!

rodrigoluk, 2021-09-21:
Hi @robmorin,

I tried to reply by referencing a new thread but I don't think that worked.
I'm using StarDot LEN-2M1.55MMCS 1.55-1.55mm f/1.5-1.5 Fisheye Body only Camera Lens, Black https://www.amazon.co.uk/dp/B01CX4U7DK/ which seems to be a clone of popular lens made by Arecont. There are many Chinese clones but basically the same lens.

The following page has an interesting set of tests https://skyinspector.co.uk/allsky-camera-ii/.

I hope this helps,

R.

robmorin, 2021-09-21:
Thanks @rodrigoluk 

I shall take a look at those URLs..
I update the post when i pick something up.

Have a great evening.( i think evening, your 6 hours ahead of me, I'm from Montreal, Canada)

EricClaeys, 2021-10-04:
@rodrigoluk, I'm on vacation and don't have access to my Pi so haven't been able to test.  I'll test when I'm back home.

rodrigoluk, 2021-10-05:
No problem. Enjoy the break! 😁

EricClaeys, 2022-06-30:
@rodrigoluk   Is this still an issue for you?  I don't see it with my test Pi, although it's using libcamera-still and not raspistill.  And no one else has reported this, even though a lot of people us a 10 ms delay between pictures.

EricClaeys, 2022-06-30:
@rodrigoluk   Allsky doesn't have direct control of the RPiHQ camera like it does with ZWO cameras.  Instead, it needs to call a program to actually take pictures.  On Buster it the older `raspistill` command and on Bullseye it uses `libcamera`, which replaces `raspistill`.  I believe the intent is that `libcamera` will eventually support more cameras and many more features.  One example is it will very soon support sensor temperature so Allsky can create darks like it does for ZWO.

I believe `libcamera` is a total re-write of the  `raspistill` code so they likely behave quite different "under the hood".
The `libcamera` tools include several commands - the one Allsky uses is actually called `libcamera-still` (for still pictures); there's also `libcamera-video` and others.

Since your original problem is resolved I'm going to close this Issue.  If it happens again, let's open a new Issue.

closed 2022-06-30 by EricClaeys