AllskyTeam/allsky #487: General setup question

maserowik, 2021-09-17:
AllSky is working fine for me but want to get a better understand of one of the folders. 
In the ftp-settigns.sh there is a directory that is called IMGDIR.  That folder is created on the hosting company site.  Been looking in that folder and it appear that is empty.  

So is that folder a temp folder for file processing of images? should there be something in that folder? is something missing on the current setup being used?

Thanks
Mike

EricClaeys, 2021-09-18:
@maserowik, good questions, and they may help us improve the documentation, so thanks for asking.

The variables in ftp-settings.sh should point to whatever directories you create on your FTP server.  These are totally independent of the folders on your Pi, and you could, if you were crazy, create a "startrails" folder on the FTP server and have Allsky copy the keogram to that folder.
The defaults in ftp-settings.sh when you install Allsky are there simply to give you an idea of what to put in the variables.
On my FTP server, I put everything in the topmost directory, and the FTP server balks if I use "/", so all my variables look like this:     IMGDIR=''"

Allsky doesn't create any folders on your FTP server - **you** have to do that before you try to upload files, otherwise uploads will fail.

Let me know if you have more questions about this, otherwise, please close the issue.  Thanks.

maserowik, 2021-09-19:
I have those folder on my web hosting site.  I have data in the startrails and keograms but I have nothing in the images folder.  Just trying to figure out what files or ?? go in to that folder.  The data.json and the image.jpg are in the root of the allsky folder on my hosting site.  

here are my current settings I use
[image]

EricClaeys, 2021-09-19:
@maserowik, do you have `UPLOAD_IMG="true" `in config.sh?  That will upload the current image (e.g., "image.jpg") to IMGDIR so your web page can display it.  Assuming the name of the uploaded image doesn't change, IMGDIR will only ever hold one file, the most recent image.
The other 3 directories on your web server will have many files - each with the date in its name.

Do you have "allsky-website" installed on your web server (it's the one that allows you to overlay the star field on the image)?  It knows about those directories.

Let me know if you have more questions.

maserowik, 2021-09-19:
Eric, 
I had a typo in the setup on my end.  I did not add the images to the imgdir in the ftp-settings.sh.  

Thanks for you help!!

closed 2021-09-19 by maserowik