AllskyTeam/allsky #488: GUI Question

maserowik, 2021-09-17:
So the version number of allsky can be retrieved by looking at that allsky log.  Is there a way to add the version to the be displayed for both allsky and allsky-website on the allsky admin portal.

Mike

EricClaeys, 2021-09-19:
@maserowik, you want the GUI to display the version of the three packages?  allsky, allsky-portal, and allsky-website?
I will give it some thought.  The way Git works, allowing users to download single files might make it hard to define what "version" someone is running.  Let me think about it and discuss with Thomas.

maserowik, 2021-09-19:
That what I was thinking, if it is doable.

Jonk2, 2021-09-20:
I have a reservation, this should only be visible after you've logged into the admin gui, otherwise it's a security risk.
I did start a thread asking for a logout button to be added, which is also a security thing, as anyone accessing their pi outside of their local LAN leave it wide open (on a mobile for example, where they perhaps don't close a session properly like a computer browser does).

EricClaeys, 2021-09-20:
@Jonk2, It only displays the settings file path name on the administrative GUI which you can only get to by logging in.
As far as I know, you can't get to the GUI until you log in.  If I'm wrong, please correct me.

It does NOT display on the web page that allows you to overlay a star map.

I' added "add a logout button to the admin GUI" to the "possible to-do" list.