AllskyTeam/allsky #489: Tip for use ramfs for storing image.jpg

KrisAstro, 2021-09-18:
Hi,

i wanted use a ramdisk  for storing the current image. I created a tempfs (a ramdisk) of 4mo (i have an asi 120 mono) and i modified the filename (in the camera settings screen) to adding before the filename the complete path to the ramdisk (/mnt/ramdisk/image.jpg).
I discovered a small bug in saveImageDay.sh, saveImageNight.sh and copy_notification_image.sh; FILENAME should be prefixed with ##*/ for take the filename like in cp "$IMAGE_TO_USE" "${IMG_PREFIX}${FILENAME##*/}.${EXTENSION}".

With this small tip i reduce the writes on the sd card.

ckuethe, 2021-09-18:
Check out `scripts/allsky_mfs.sh`

EricClaeys, 2021-09-19:
One of the things I am working on is putting all the files that change often into a single directory.  This would make it much easier to use a ram disk or USB drive.

KrisAstro, 2021-09-19:
Thank Chris, i seen this file and is very useful for create the ramdisk. But by default, the program check the image in the allsky directory and if you change this, we'll have problem.

@Eric : My tip work fine with/without a path to filename. I think is a good stuff to do. I just see a problem is for the webapp; the path for seen is incorrect (is the same problem with the concatenation of the filename with the imageprefix variable) but i doesn't find where this stuff is done.

ckuethe, 2021-09-19:
Let's just say that some folks who have contributed to allsky agree that being able to use a ramdisk/tempfs to save microsd wear, and that we've just about got the time and energy to develop a proper solution. Whatever we come up with, it'll be less hacky and easier than current approaches

EricClaeys, 2021-09-20:
@KrisAstro, the version of AllSky I'm working on does away with the $IMG_PREFIX variable and only uses the file name you specify (default "image.jpg").  All the old "resized", "processed", "liveview-", and other variations of "image.jpg" are gone.

The capture program will write a file called `image-DATE.jpg` to the new allsky/tmp directory, and all the scripts will process the file in that directory, then move the final version to `allsky/images/DATE/image-DATE.jpg`.  A symbolic link called "image.jpg" will be created in allsky/tmp to point to that final file.  The websites can then access allsky/tmp/image.jpg.

Gone also is the `temperature.txt` file which was written to every exposure.  Instead, the temperature is passed to the `saveDayImage.sh / saveNightImage.sh` scripts, which does away with another write.

If you want to use a ram disk or USB stick, (I think) you'll only need to map it to the allsky/tmp directory.  You shouldn't have to add a directory to the file name if the default is in allsky/tmp.

KrisAstro, 2021-09-21:
@EricClaeys it's a very very good new :+1: i'm very impatient to test this new version.

dbainbridge, 2021-09-29:
If you are on Pi4 you can boot straight off of USB/SSD drive with no need for SD card.  When my Pi4 arrives to replace my Pi3 currently being used for allsky I will be trying this since it would basically alleviate need for ram disk.

EricClaeys, 2022-03-03:
@KrisAstro The current version of Allsky implements what I described earlier.  All writes, except the final one, are to the allsky/tmp directory.  [This Wiki page](https://github.com/thomasjacquin/allsky/wiki/Miscellaneous-Tips) describes how to set up allsky/tmp as a `tmpfs` to significantly reduce wear on the SD card.

closed 2022-03-03 by EricClaeys