AllskyTeam/allsky #492: sometimes notification images appear in the nightly movies

ckuethe, 2021-09-21:
Guessing it's a race condition of some sort. Maybe notification images could be symlinked rather than copied, then `saveImageNight.sh` could detect that and ignore the image. Dunno, just noticed it, haven't had time to investigate deeply.

EricClaeys, 2021-09-21:
@ckuethe, I _think_ I fixed it in the version of AllSky I'm developing/testing.  I'll keep this issue open until the updated version is out for a while and people don't report any issues.

ckuethe, 2021-09-21:
Cool, you should update your fork with your changes so I can test them.

ckuethe, 2021-09-21:
It occurs to me that having notification images in the allsky directory is the wrong approach. They should go into the web server directory as a web asset.

PHP can calculate the best image to display, possibly with a banner that indicates the status of the capture process.

* If there's no image, PHP displays a black box, or maybe a cloud partly covering the sun/moon, and says "No images captured yet."
* If an image is present, display it.
* but if it's older than, I dunno, more than 4 capture intervals add a caution/warning/problem banner than says the image is stale
* and also checks for the existence of a .pid file created by capture (also in your forthcoming working directory changes) and verifies that that pid corresponds to a running capture process. Do nothing, or show a banner as appropriate.

EricClaeys, 2021-09-21:
Not everyone has the GUI or web interface so we can't count on them being there.  And some people put the web interface on a separate machine and FTP files to it.

The capture programs display several notification images:
* Taking dark frames
* Camera off during day
* Software not running

allsky.sh displays:
* Software starting up
* ERROR - see log file
* Software not running

There's already communication between allsky.sh and the capture programs; moving the notification files elsewhere would require creating that communication, and to be honest, I don't really see a benefit.
One thing I like about the notification images is that they are very obvious.  Black screens and iconic-like images can be interpreted all sorts of ways.  The website used to display a small aurora thumbnail if there was no actual thumbnail - it wasn't obvious at all what the picture was or what it had to do with a missing thumbnail.

Since Thomas created the "dev" branch (and now separate branches for each PR) I haven't been using my own forks.  I never got them to work correctly anyway - whenever Thomas merged things my fork was both ahead and behind his in commits and neither of us could figure it out.
I can try again for testing purposes.

ckuethe, 2021-09-21:
Branch-per-PR is generally a good practice. It keeps the diffs small and easy to review, so you know exactly what you're changing and whether nor not you've missed something. You can make one branch depend on another branch if you're stacking two major changes on top of each other - and rebasing to account for changes that were merged while you were developing is easier (often trivial) with smaller diffs.

Once the PR is merged, you can delete the branch so your repo doesn't get too cluttered.

EricClaeys, 2022-02-10:
Issue fixed so closing.

closed 2022-02-10 by EricClaeys