AllskyTeam/allsky #497: Time delay between Pictures

sebmueller, 2021-09-23:
As I looked the pictures of a night, I found an airplane trail on them.
Between 2 Pictures, there were a big gap in the airplane trail. As I watched at the timestamp of the pictures, I saw, there is also a gap of app. 6 seconds.
I investigated the code an found, that the "raspistill" command is called without the "--timeout" parameter. That means, raspistill waits for about 5 seconds before it starts to take the picture.
If I add the parameter "--timeout 1" it starts more or less immediatly. 

"--timeout 0" means, it will never start.

So maybe, you can add this parameter for taking faster sequences of the picture.

kind regards
sebastian

EricClaeys, 2021-09-23:
@sebmueller this is a hard one.  I played around with different values and settled on 2 seconds during the day and 10 seconds at night.  If the number is too short, raspistill doesn't have time to do the auto exposure.  I figured during the day the exposures are normally very short so 2 seconds should give more than enough time.  At night, exposures are much longer and could really use more than 10 seconds, but 10 seemed like a good compromise.

What I'll do in the next version is allow users to change the day and night values, which will only apply if auto exposure is on.  Does that sound reasonable?

sebmueller, 2021-09-24:
@EricClaeys thanks a lot for your fast response.

I try to catch comets, so a gap of 10 seconds between 2 pictures is to long.

As we use a manual exposure time and no auto gain (in the night), the sensor shouldn't need time for setting up these parameters, or I'm wrong?

To have this option in the setting user menu is a good Idea, but why only with autoexposure on?

EricClaeys, 2021-09-24:
@sebmueller if auto exposure is OFF it would ignore the timeout values entered by the user and use the smallest possible value.
If auto exposure is ON it will use what the user specified, or defaults if they didn't specify anything.

Eric

sebmueller, 2021-09-25:
@EricClaeys ok, I see it.
but for finding comets, a 10 second delay is to long. so to set the parameter by users is a good solution :)

I'm going to play around with some different parameters, maybe there is a better solution.

kind regards
sebastian

EricClaeys, 2021-10-20:
@sebmueller,
Where did you get the raspistill.c program?  It doesn't have all the options as the one that comes on the Pi.  And your startnightview.sh has several options that aren't in your raspistill.c program.

On line 1659, "sig" is never set.  It's commented out on line 1655.  Is that a bug?
What sends SIGUSR[12] to raspistill?  Something other than the Allsky software?  If so, what regulates the delay between pictures in Allsky, or are they taken as fast as possible?

Interesting about the airplane.  Thanks for sharing.

sebmueller, 2021-10-21:
@EricClaeys 
Some options of raspistill are defined in RaspiCommonSettings and other files

Pedritoprint, 2022-01-27:
Any news? i saw on my pictures from yesterday's evening (26.1.2022 18:15) when ISS crosed my "allsky", that between two pictures, there are about 5sec delay, allthogh my "Delay"-parameter is set to 10 (ms). The Exposure-Time is 30" but the Timestamp between the saved pictures is 35".
I'm working with RPiHQ, Raspberry Pi 4B Bullseye.

EricClaeys, 2022-02-10:
@Pedritoprint I'm not ignoring you; I've been on the road for about a week and at busy at my observatory the week before.

Are you use auto-exposure?
If you, you can remove the `* MS_IN_SEC` on lines 444 and 448 in src/capture_RPiHQ.cpp.  That will set the delay to 2 and 10 milliseconds rather than 2 and 10 seconds.  Run `cd src; make capture_RPiHQ` then stop Allsky and `cp capture_RPiHQ ..` and restart Allsky.

If you are using manual exposure, I'm not sure how to decrease the delay between pictures.  Allsky adds 10 milliseconds, plus whatever time it takes raspistill to start up and take the picture.

EricClaeys, 2022-06-01:
Closing Issue.
The next version of Allsky will set the delay to the shortest value possible in manual exposure mode and when using the "Mean Brightness" auto-exposure mode.  When using the camera's auto-exposure, there will be a 2-second delay during the day and 10-second at night, to give the camera time to determine the correct exposure.
We expect people will either use manual exposure or "Mean Brightness" auto-exposure.

closed 2022-06-01 by EricClaeys