AllskyTeam/allsky #501: Question about the allsky and allsky-gui

maserowik, 2021-09-23:
I remember reading a post about maybe grouping the allsky program and the allsky-gui(portal) to one install program.  Is this still on the to do list?

EricClaeys, 2021-09-29:
@thomasjacquin, @linuxkidd, @ckuethe 
I think this would be a good idea to merge the GUI portal into the base "allsky" install so we can force people to use the GUI to configure everything.  The GUI would enforce type checking and range boundaries (e.g., for camera-specific values).  That would also make documentation and debugging easier (could remove all the variable descriptions from README.md since they are in the "description" field of the config file.

Comments?

WirthmU, 2021-10-01:
It would be really good to go ahead this way.

Also, and considering the ever increasing pace of commits it would be very practical to have a one-click update procedure.

A huge "thank you" to all the curageous contributors!

Regards from Switzerland

EricClaeys, 2021-10-01:
@WirthmU Having an upgrade script is on the "to do" list.

linuxkidd, 2021-10-01:
We're actually looking at providing an apt repository with compiled packages ready to install via `apt` or `apt-get`.  The `allsky` package will have a dependency on `allsky-portal` so that both are installed at the same time.  Keeping them separate however, will allow each to iterate separately as needed.

ckuethe, 2021-10-01:
Thoughts

* what if I'm already running lighttpd on my pi for something else? Can we make this auto-installed web ui (reasonably) claim only the `/allsky` namespace and not mess with anything else lighttpd might be doing? do we set up a /etc/allsky/lighttpd directory?
* what if I have a different web server installed? Is the gui compatible with that?
* what if I'm already running completely different daemons on ports 80, 8000, and 8080?
* I'm not fond of dragging in too many dependencies: gpsd-clients comes to mind. I think that could end up installing X on what should be a headless, text only machine because one of the sample apps is a GUI tool and all I wanted was `cgps`... a little text-based "you are here" tool. I know, we're linking x and opencv and a bunch of other stuff, but part of me always whines "why for I gotta install that?!"

I think I'd prefer to keep allsky and allsky gui separate,

EricClaeys, 2021-10-01:
Chris,
When you say you prefer to keep the GUI separate do you mean it shouldn't be installed automatically, or it should but it should remain a separate package?

Eric

linuxkidd, 2021-10-02:
I seem to recall that the Allsky WebUI used to consume /allsky on the web server.  ( At least, on my prior install it did -- unsure if that was an edit I made or it came that way ).  I think it makes sense to do that for just the reasons @ckuethe mentioned.

Further, I agree that it would be great if the installer and/or package knew an existing webserver was present, and didn't try to re-invent the wheel.  ( Historically, I'd comment out the install lines for lighttpd, php, etc... and front it with nginx that was already present ).

Much investigation to be done..  :)

EricClaeys, 2021-10-02:
@linuxkidd Michael, did Chris forward the "Allsky enhancements-version2.txt" file to you?  It includes what I'm currently working on and all the possible enhancements he and I had when I created the file.  One of them is "Support upgrades" of allsky, allsky-portal, and allsky-website.  For example, installing allsky-portal blows away allsky-website's files.

I'd be happy to add your possible enhancements to the list.

EricClaeys, 2021-10-02:
@ckuethe

`* what if I'm already running lighttpd on my pi for something else? Can we make this auto-installed web ui (reasonably) claim only the /allsky namespace and not mess with anything else lighttpd might be doing? do we set up a /etc/allsky/lighttpd directory?`
The allsky-portal webscript should check if lighttpd is already installed and configured for allsky.  There are two aliases in the lighttpd config file for "/current" and "/images".  I don't know what else is in the file that allsky needs.

`* what if I have a different web server installed? Is the gui compatible with that?`
This is harder - I assume there are lots of different web servers?  If so, we'd need to check for all of them and add the aliases as needed.  I would be tempted to say AllSky only works with lighttpd, although we could add support for other servers if people request it.

`* what if I'm already running completely different daemons on ports 80, 8000, and 8080?`
Is there a way to check?  If so, we could use a different port and let the user know.  Does allsky-portal use anything other than 80?

`* I'm not fond of dragging in too many dependencies: gpsd-clients comes to mind. I think that could end up installing X on what should be a headless, text only machine because one of the sample apps is a GUI tool and all I wanted was cgps... a little text-based "you are here" tool. I know, we're linking x and opencv and a bunch of other stuff, but part of me always whines "why for I gotta install that?!"`
Are you referring to installing the GUI when installing allsky?  The "why" is because IMHO it makes documenting the settings much easier and allows us to do real-time error checking.

ckuethe, 2021-10-03:
Using gpsd as an example again, we got really frustrated with incomplete bug reports so we wrote a tool to gather all the required information in one go. We could write a similar diagnostic tool for allsky. And if the problem is obvious, we could remediate it too.

As for web stuff, I don't think that gui/portal/website know about the bind address or listen port. Php runs, processes some environment variables, and then spews out some content to stdout. With any luck, there's a web server that will pass it on to a web browser... It'd be easy enough to check for that.

I volunteer to write some sort of cli based config check tool.

And "why I gotta install all that?" - basically I just want to be able to do minimal installations of stuff. I can `apt install` the bits I need when I'm about to use them.