AllskyTeam/allsky #510: Problem with automatic night exposure

bribon2, 2021-09-26:
Hello good Morning. I have a problem that I cannot find where the solution can be ...
Days when there is a Moon for example, the automatic exposure changes constantly producing a "flicker" in the sequences, such as:
This happens to me since I update to the latest version 0.8
In version 0.7 this did not happen to me.

Thanks!
https://observatoridepujalt.cat/allsky/videos/allsky-20210925.mp4

linuxkidd, 2021-09-26:
Hi @brianboru82 -- Can you attach the `/var/log/allsky.log` file file to this issue.  Note: Attach can be done by dragging the file into the compose window for your next comment, please don't copy/paste the file contents.  

@EricClaeys will likely want to see that.

I'm wondering too, if you could upgrade to the latest code ( just merged last night ) and see if that helps at all.

Thanks!

bribon2, 2021-09-26:
Thank you! This is the file:
[allsky.log](https://github.com/thomasjacquin/allsky/files/7232054/allsky.log)

How to update the latest code? Do I have to delete everything and reinstall?

ckuethe, 2021-09-26:
You don't have to delete, you can just move the allsky directory out of the way. Additionally, if you're comfortable with git, you can do something like `git stash && git pull && git stash pop` (you might have to resolve some merge conflicts if you've made local changes).

EricClaeys, 2021-09-27:
@bribon2 We didn't do anything to the nighttime exposure per se, but it's possible the 0.8 exposure algorithm has a side effect.  Try disabling the Version 0.8 Exposure - in the GUI, go to the bottom of the Camera Options and click on Show Advanced Options, then set the 0.8 Exposure Mode to No.
If it's aldeady No I will have to look at your log file in a couple days when I have my PC.

Eric

EricClaeys, 2021-09-28:
@bribon2 It looks like auto gain is on, and I suspect auto exposure as well since the sleep times vary, and that only occurs when the exposure length changes.  Try disabling auto gain, and set it to a manual 220 which is what the auto gain was using.  Some people have reported problems when both auto gain and auto exposure are set.
At night the camera handles the exposure time and gain value, our software doesn't.
Let me know how it goes with manual gain.
Thanks - Eric

bribon2, 2021-10-04:
Removing the automatic gain happens to me the same.
The option to remove the algorithm 0.8 does not appear to me ... What's more, I can't apply the settings from the GUI, I have to do it from the config file and the default settings appear from the GUI, although it actually applies those from the config.
I updated it from version 0.7 to 0.8 by deleting the files by hand and installing it from 0, I don't know if I did something wrong ...

Example of tonight: https://observatoridepujalt.cat/allsky/videos/allsky-20211003.mp4

Jonk2, 2021-10-04:
I've also noticed this, but not mentioned it due to the ongoing daily changes, I was hoping that it would already be on the to-do list.
You can see by the attached compressed video excerpt and keogram from last night, that the exposure is hunting when not at the maximum exposure value - zoom in and look at the keogram, the 2nd half of it where it's cloudy and the sky brightness changes continuously - it ramps up over 5 or so exposures, before reverting back and hunting again.
You can also see this on the video, watch the exposure value on the overlay - you can see it in the numbers.
In another allsky program I used in the past, you could set the number of exposures the software used for it's target exposure, the minimum being 2 frames, maximum, I don't remember. the idea was that it could be fine tuned by the user to filter out these ramps and settle on the best for their camera / setup.

I also remember there was an option to set the autoexposure minimum / maximum changes, for example during the night you might want the exposure to only change by 0.5s, to reduce the flickering, and only use the last 3 frames to settle on an exposure.

I'm using a version of this allsky that is probably only 3 or 4 days old (i.e. not the very latest).

[image]

https://user-images.githubusercontent.com/77614536/135823196-ae3ca0b2-fde0-4338-adb7-709166cecb66.mp4

EricClaeys, 2021-10-04:
@bribon2 Is the bright light on the right side the moon?  It doesn't appear to be moving.
Do you notice the "flicker" when there are no clouds?  I would guess that some of the flicker is due to the clouds coming and going.  When there are lots of clouds the image is dark, but when the clouds clear around the moon the next picture will be bright, then if the clouds come back before the picture after that, that picture will be dark.  It takes the auto exposure algorithm some time to adjust to changes in brightness.

In order to see the option for disabling the 0.8 exposure method, you'll need to upgrade the GUI (allsky-portal).  When you do that you'll see a "Show advanced options" button at the bottom of the `Camera settings` page.  Click on that and you'll then see the version 0.8 option (it's considered an "advanced" option).

@Jonk2 Do you notice the "flicker" when there are no clouds?  Does changing the version 0.8 exposure method make any difference?

Jonk2, 2021-10-05:
I don't notice a flicker when the exposure is at maximum which I suppose is expected, as nothing's changing. It just appears to ramp up over 4 or 5 frames, then returns to the value at the beginning of the ramp when it's at anything less than the max (30 seconds in my case).
I tend to do all my testing on the PI indoors, leaving the one outside as is, but I've just changed the one outside to v0.7 exposure mode and will report back tomorrow, to see if there are any differences.

EricClaeys, 2021-10-05:
@Jonk2.  Thanks in advance.  I'm curious what the outdoor one does with and without clouds, both with 0.7 and 0.8.  It might be difficult to test for the next couple weeks since the moon isn't out.

EricClaeys, 2021-10-08:
@Jonk2 if daytime and nighttime gain are different and aren't set to "auto", then gain should be adjusted from its night value to its day value when going from night to day, and from the day to night value when going from day to night.  It begins at the day/night transition point as specified by your "angle".  The transition length is user-settable.  It doesn't take the image brightness into account when adjusting the gain, it just does simple math - I'm at 100 and need to get to 0 in 5 minutes so I'll decrease gain by 20 every minute.  If you want to attach /var/log/allsky.log when this is happening I'll take a look. The exposure algorithm I think has a bug that may be contributing to the problem.

I'm not able to play the videos.  Tried with 2 browsers on my PC and with my phone.  Can you try attaching them rather than including the URL?

I assume you're "angle" is such that sunrise and sunset are during daytime.  If so, the histogram exposure algorithm is being used.  You can try adjusting where the histogram box is (the part of the image that's looked at to get the brightness), and adjust the size of the box.  On smaller sensors the box size is too big (15% of the chip size in pixels is a good amount).  In the WebUI on the Camera Settings page, enable "Show Histogram Box" and then adjust the "Histogram Box" numbers.
My camera is positioned such that the sun and moon go right through my histogram box and when that happens the pictures get very dark, so I moved my box and on my ASI290 with a small chip, decreased the size.

Jonk2, 2021-10-08:
@EricClaeys strange, must be the encoding. Try these?

Although they're quite compressed, you can see the numbers and sunset / sunset flickering.

https://user-images.githubusercontent.com/77614536/136597293-8641cdc3-69b4-4bf2-a81d-75316a68bd7f.mp4

https://user-images.githubusercontent.com/77614536/136597435-245445fa-048b-460b-9567-b22d4a1a02b1.mp4

https://user-images.githubusercontent.com/77614536/136597451-f6ea709b-e63e-477f-b650-0f00f2656cdb.mp4

My 'nighttime' is set to an angle of 0, so my timelapse will always show the transition from day / sunset into full night (which is usually over a 2 hour period and timelapses showing this to me look great - it's called the holy grail method).
I get it's just a timer, but this timer starts at sunset in my case (angle of 0), so why doesn't it finish at angle 0 (i.e. sunrise) rather than start the transition back to gain 0 at sunrise? Does that make sense? I would have thought the transition back to 0 gain for daytime would start 120 minutes before sunrise (the angle of 0).

EricClaeys, 2021-10-09:
@ckuethe Chris, can you take a look at the changes in keogram size above, between version 0.7 and 0.8?  The version 0.8 one seems to be scrunched up in width.

EricClaeys, 2021-12-02:
@Jonk2, there's a new option to the keogram program that causes it to always create an image that's as wide as your sensor, so you shouldn't get images like below.
I think it's the --extend option, but run `./keogram --help` to be sure.

[image]

EricClaeys, 2021-12-29:
@Jonk2, have you had a chance to try the new keogram option to expand the image?

@Apennine2, have you tried the most recent capture.cpp and other files?  I'm curious if they help with the nighttime strobing.

Jonk2, 2021-12-29:
I haven't done anything manually, and I have only glanced at the automatically generated keograms every other day or so, but checking this morning's I notice 3 changes from the images above:

1 - The text is now white (better)
2 - There's a date (which I don't like personally)
3 - The top of the image is trimmed so it doesn't show the entire height.

This morning's is attached.

Regarding the strobing, it's still there and is caused by exposure hunting over 5, sometimes 6 frames for some reason. You can see it clearly in the keogram when you zoom in.

[image]

EricClaeys, 2021-12-30:
@Jonk2 For # 1, are you using the default settings in config.sh:
`KEOGRAM_EXTRA_PARAMETERS="--rotate 0 --font-size 1.0 --font-line 1 --font-color '255 255 255'"`
I removed the font-size and font-line and let the program use its default, which are 2.0 and 3, which I think make the text much easier to see.

For numbers 2 and 3, would you please enter a new Issue stating the top of the keogram image is trimmed, and include the picture above?  Also request that displaying the date be user-configurable.

For the strobing, please see [this Wiki page](https://github.com/thomasjacquin/allsky/wiki/Reporting-Issues) and attach your log file.  Do you see the strobing during the day AND night?

Apennine2, 2021-12-30:
@EricClaeys , I do have the latest software versions installed and I still see the strobing at night. I do not see the strobing during the day, although it's not making videos during the day. But, I say that it isn't since I always see a nicely exposed constant exposure during the day. 

I keep the max night exposure at 60s, since that makes great videos on moonless clear nights. Here is last night's video where you still see the strobing caused by a ~+/-3s exposure bounce during an overcast night. 

[Video](http://www.thewanderers.xyz/allsky/videos/allsky-20211229.mp4)

-Chris

EricClaeys, 2021-12-31:
@Apennine2 I'll need your log in order to troubleshoot.

Apennine2, 2021-12-31:
@EricClaeys , the log is attached. This is for the subsequent (last) night. Also a constant cloudy/foggy night. I would have attached the log yesterday, but it was huge since I had not cleared it in a while. I truncated it yesterday, and collected data for about 24hr.

-Chris
[allsky.log](https://github.com/thomasjacquin/allsky/files/7796573/allsky.log)

Jonk2, 2021-12-31:
> @Jonk2 For # 1, are you using the default settings in config.sh: `KEOGRAM_EXTRA_PARAMETERS="--rotate 0 --font-size 1.0 --font-line 1 --font-color '255 255 255'"` I removed the font-size and [...]

Yes the keogram_extra_parameters is the same as you mention (default).

I will start a new issue about the keogram trimming.

The 'strobing' happening during the day, I don't know as it doesn't save the images to create a day timelapse.

EricClaeys, 2022-01-01:
@Apennine2 You are probably getting pretty bad strobing during the day as well.  Try saving daytime pictures so they are included in a timelapse and look for daytime strobing.  Or you may be able to see stobing by zooming in on the keogram.

Your camera appears to be reacting unpredictably to changes in exposure time.  During the day the first picture gave a mean brightness of 250 (out of 255) at 500 microseconds (us) exposure so was way too bright.  The next couple pictures used shorter exposure times and the brightness went down some, then all of a sudden it got too dark.  Adding a little exposure then made it too bright.  It then kept ping-ponging like that (which causes the strobing) for quite a while then settled down, then started again.

At night the camera uses the ZWO auto exposure so we have no control over it.  It was also doing the ping-pong but also had shots of the same exposure but somewhat different brightness.  That again could be caused by rain drops on the dome.

It's very possible that the strobing was caused by rain drops - slight movements in them can cause bright spots that cause the camera to try and darken the next picture, then the bright spot is gone and the camera brightens the next picture.

Do you see the strobing when it's clear outside?  Or when it's fully cloudy but not raining?

ennui2342, 2022-03-29:
I'm in a similar position as reported. I had to recently do a complete rebuild of the pi and upgrade from a version of allsky about a year ago to a fresh install. Now nighttime cloudy videos flicker as the exposure ramps up and back over a few frames as described, before they were nice and stable. This is with an asi224mc.

ennui2342, 2022-03-30:
Hey Eric. So the setting certainly made some improvement - the flickering isn't as deep, but is still there as the exposure changes enough per frame to be noticeable. Here are videos to show what's happening:

The first is an equivalent night on my approx year old setup before I had to rebuild...
https://user-images.githubusercontent.com/912310/160779809-8a8f05b0-277e-4a71-a4d1-722ee0bb8b0c.mp4

The second is the default setup without the config change you suggested...
https://user-images.githubusercontent.com/912310/160780004-0b4c0497-8f38-4978-a298-f57d615a97b2.mp4

And the third is the improvement with the 0.8 config change, but still not back to the quality I had in the 2021 version...
https://user-images.githubusercontent.com/912310/160780172-bb458100-8c02-4526-9128-bf750341d3a2.mp4

EricClaeys, 2022-04-01:
@ennui2342   If you are using the latest version of Allsky, try setting
```shell
CAPTURE_EXTRA_PARAMETERS="--aggression 50"
```
in config.sh.  This will apply half the expected change in exposure time and should even out the flickering.  You can experiment with the number.  The default is 75.
Let me know how it goes.
If you are still getting flickering, please follow the instructions in the Wiki on Reporting Issues, and upload the log file.

ennui2342, 2022-04-22:
Hi Eric. I gradually ramped this incrementally all the way up to 100. And while the night images aren't too bad, when there's cloud and particularly the moon, it still can't match the performance of the old version (which is a shame because these kind of dramatic images were the type I was posting on http://www.tiktok.com/@photonfog).

Here's the current setup with aggression at 100

https://user-images.githubusercontent.com/912310/164638602-6b6487af-26d5-4c62-8253-547d6a7e6f70.mp4

and here's how the old system performed with the moon in frame, you can notice the exposure stability - it is changing but it's much more stable

https://user-images.githubusercontent.com/912310/164639350-7bd13c95-8c97-4c52-9fc5-f56a996d5bba.mp4

Is there anything more I can try apart from downgrading? Thanks.

-Mark

EricClaeys, 2022-04-23:
@ennui2342    Mark, you want to _decrease_ the aggression to even out the changes.  100% means it makes the full change in exposure every time.  If one picture is very bright it will darken the next picture a lot, which you don't want.

Try decreasing aggression.  Also, change **Debug Level** to 4 so we make sure aggression is working.

ennui2342, 2022-04-23:
> @ennui2342    Mark, you want to _decrease_ the aggression ...

Omg, how silly of me - I assumed it was how aggressively the changes were damped... Doh. I'll let you know how it goes...

ennui2342, 2022-05-07:
Hey Eric

I've decreased aggression all the way down to zero, but it hasn't really improved the situation when there's anything other than a clear sky. Here's last night for example:

https://user-images.githubusercontent.com/912310/167260524-c0ccb4a6-126d-4ed3-a483-036c086f5feb.mp4

Are there any other settings worth trying?

-Mark

EricClaeys, 2022-05-08:
@ennui2342   Mark, would you please turn on **Show Mean Brightness** in the WebUI (it's an advanced option), then follow the instructions in the Reporting Issues Wiki page and attach the log file?

If you look at the houses on the right, do they flicker in brightness, or is it just the sky?  You may need to look at the actual pictures instead of the video.

ennui2342, 2022-05-10:
Will do Eric - I assume you mean turn on the historgram for ZWO as mean brightness is only an option if using the raspberry pi camera.

Anything that's a bright light noticeably flickers because of the rapidly changing exposure - so when the lights are on in the house you can see this. You can see the exposure moving quickly in the annotation in the top left. If you compare this to the video above from the older version of allsky with the moon in it you can see that the exposure is a lot more stable and there is no noticable flickering.

ennui2342, 2022-05-11:
Here's the video from last night and the log file (I've cut it down to 2am to 3am)...

https://user-images.githubusercontent.com/912310/167832619-d20efea8-e6f2-4fb4-839a-b3f2073fc73b.mp4
[allsky.log.txt](https://github.com/thomasjacquin/allsky/files/8669217/allsky.log.txt)

It looks like there's a pattern that the mean jumps up adding a big jump to the exposure then jumps back down, the exposure gradually ramps back down to match and then the mean jumps back up and down again causing another cycle of jump and slow ramping down. Seems it adds a big jump up but a slow ramp down.

This is with 0.8 exposure set to on.

EricClaeys, 2022-05-11:
Thanks @ennui2342.  Can you try with 0.8 exposure OFF ?

EricClaeys, 2022-05-11:
@ennui2342   Mark, would you also please attach the log file from when the video above was happening?
Thanks

ennui2342, 2022-05-11:
> @ennui2342   Mark, would you also please attach the log file from when the video above was happening?
> Thanks

The log and video in the post above are both from last night. Do you mean you want the whole nights log since I trimmed it to just the 2am to 3am window?

EricClaeys, 2022-05-11:
@ennui2342   Mark, I didn't see the link to the log file - I only saw the video link.  Never mind.

When a much brighter image occurs, the ZWO camera driver says the next exposure should be HIGHER than the prior one, which is exactly opposite of what I'd expect.  Unfortunately I can't do anything about that until we totally replace the ZWO auto exposure/gain algorithm with our own, which won't be for a few months.

Was your **Debug Level** at 4?  There aren't any log entries about dampening the exposure change due to aggression so I can't tell if it's working on not.

ennui2342, 2022-05-11:
Hmm that's interesting. The debug level is at 4. Maybe that suggests the aggression flag isn't being applied? This is the line from the config:

CAPTURE_EXTRA_PARAMETERS="--aggression 0"

EricClaeys, 2022-05-11:
@ennui2342   Mark, please set aggression to 1.  The log file should have an error message that the value must be between 1 and 100.

ennui2342, 2022-05-11:
Doh - I should have checked that. I'll try with it at 1 tonight with 0.8 exposure on and post the logs.

EricClaeys, 2022-05-11:
@ennui2342 When I looked at the log message in the code I realized it said it was setting the 0 value to 1, but it didn't actually do that, so I just submitted a bug report.

ennui2342, 2022-05-14:
Hi Eric, heres the results at aggression set to 1 with 0.8 exposure on

https://user-images.githubusercontent.com/912310/168421034-2440933a-c5ec-4ced-8cbf-812f90693bea.mp4

[allsky.log.txt](https://github.com/thomasjacquin/allsky/files/8692682/allsky.log.txt)

Since there didn't seem to be much difference in the log , but I did notice in the day, I had a quick look at the code and I wonder whether most of the aggression code isn't actually being used at night. I notice this comment before the conditional statement that limits a whole chunk to daytime: 

https://github.com/thomasjacquin/allsky/blob/c8a8082830e4d0655d75318f94caed68d7e0967b/src/capture.cpp#L2167

I'll try tonight with the same settings but 0.8 exposure off and let you know how that goes

ennui2342, 2022-05-15:
With 0.8 exposure off it was much worse. Is it worth me trying to find an older version of the ZWO driver and downgrading? I'm not clear whether this is something that's change in the driver or with the allsky code. I'm off to a dark sky site later this month, so I'd really like to get it good enough for the trip :)

EricClaeys, 2022-05-16:
@ennui2342   Mark, would you please attach the top of the log file, where it shows all the settings?  It'll start with `***** Starting AllSky *****` and go through `*** Stop the allsky service to end this process. ***`.
I'm working on a test version of the capture program that should give me more info.
Thanks

EricClaeys, 2022-05-16:
@ennui2342   Mark, the log file you attached a couple days ago appears to have aggression at 75% which is the default.

Something weird is going on with the camera.  Notice in the following entries, the Mean brightness (from 0=black to 255=pure white) changes even though the exposure is the same.  For example, the first two shots are both at 1.8 seconds but the Mean goes from to 165.  I would expect some changes when there are clouds that come and go with a bright moon, but not this consistently.  It's pretty much up, down, up, down, etc.  
Also, the "Next" exposure is what the camera suggests for the next exposure, in order to auto expose.  That number is adjusted per the aggression.

Some things to try while we debug this:
* Turn Auto Gain off if it's on.
* Disable stretching in config/config.sh.  That won't solve the problem but may make the changes less noticeable.
* Try the attached file.  To do so:
   ~~~shell
   sudo systemctl stop allsky
   cd ~/allsky/src
   mv capture.cpp capture-ORIGINAL.cpp
   # move the attached file and rename to "capture.cpp"
   make capture
   cp capture ..
   sudo truncate -s 0 /var/log/allsky.log
   sudo systemctl start allsky
   ~~~

~~~
STARTING EXPOSURE at: 2022-05-12 21:21:39   @ 1.8 sec    > Mean: 127.  Next: 1.8 sec
STARTING EXPOSURE at: 2022-05-12 21:22:39   @ 1.8 sec    > Mean: 165.  Next: 2.2 sec
STARTING EXPOSURE at: 2022-05-12 21:23:41   @ 2.1 sec    > Mean: 129.  Next: 2.1 sec
STARTING EXPOSURE at: 2022-05-12 21:24:41   @ 2.1 sec    > Mean: 166.  Next: 2.7 sec
STARTING EXPOSURE at: 2022-05-12 21:25:43   @ 2.6 sec    > Mean: 124.  Next: 2.6 sec
STARTING EXPOSURE at: 2022-05-12 21:26:43   @ 2.6 sec    > Mean: 159.  Next: 3.3 sec
STARTING EXPOSURE at: 2022-05-12 21:27:45   @ 3.1 sec    > Mean: 123.  Next: 3.1 sec
STARTING EXPOSURE at: 2022-05-12 21:28:45   @ 3.1 sec    > Mean: 156.  Next: 4.1 sec
STARTING EXPOSURE at: 2022-05-12 21:29:47   @ 3.9 sec    > Mean: 124.  Next: 3.9 sec
STARTING EXPOSURE at: 2022-05-12 21:30:48   @ 3.9 sec    > Mean: 153.  Next: 5.1 sec
STARTING EXPOSURE at: 2022-05-12 21:31:50   @ 4.8 sec    > Mean: 121.  Next: 4.8 sec
STARTING EXPOSURE at: 2022-05-12 21:32:50   @ 4.8 sec    > Mean: 156.  Next: 6.3 sec
STARTING EXPOSURE at: 2022-05-12 21:33:53   @ 5.9 sec    > Mean: 126.  Next: 5.9 sec
STARTING EXPOSURE at: 2022-05-12 21:34:54   @ 5.9 sec    > Mean: 159.  Next: 7.5 sec
STARTING EXPOSURE at: 2022-05-12 21:35:57   @ 7.1 sec    > Mean: 130.  Next: 7.1 sec
STARTING EXPOSURE at: 2022-05-12 21:36:57   @ 7.1 sec    > Mean: 168.  Next: 9.2 sec
STARTING EXPOSURE at: 2022-05-12 21:38:01   @ 8.6 sec    > Mean: 141.  Next: 8.2 sec
STARTING EXPOSURE at: 2022-05-12 21:39:02   @ 8.3 sec    > Mean: 130.  Next: 8.3 sec
STARTING EXPOSURE at: 2022-05-12 21:40:03   @ 8.3 sec    > Mean: 173.  Next: 10.6 sec
STARTING EXPOSURE at: 2022-05-12 21:41:07   @ 10.0 sec   > Mean: 148.  Next: 9.4 sec
STARTING EXPOSURE at: 2022-05-12 21:42:07   @ 9.6 sec    > Mean: 138.  Next: 9.2 sec
STARTING EXPOSURE at: 2022-05-12 21:43:08   @ 9.3 sec    > Mean: 135.  Next: 9.3 sec
STARTING EXPOSURE at: 2022-05-12 21:44:08   @ 9.3 sec    > Mean: 184.  Next: 11.9 sec
STARTING EXPOSURE at: 2022-05-12 21:45:13   @ 11.3 sec   > Mean: 154.  Next: 10.6 sec
STARTING EXPOSURE at: 2022-05-12 21:46:14   @ 10.7 sec   > Mean: 143.  Next: 10.2 sec
STARTING EXPOSURE at: 2022-05-12 21:47:15   @ 10.4 sec   > Mean: 141.  Next: 9.9 sec
STARTING EXPOSURE at: 2022-05-12 21:48:15   @ 10.0 sec   > Mean: 139.  Next: 9.6 sec
STARTING EXPOSURE at: 2022-05-12 21:49:16   @ 9.7 sec    > Mean: 140.  Next: 9.7 sec
STARTING EXPOSURE at: 2022-05-12 21:50:17   @ 9.7 sec    > Mean: 199.  Next: 11.9 sec
STARTING EXPOSURE at: 2022-05-12 21:51:22   @ 11.3 sec   > Mean: 167.  Next: 10.3 sec
STARTING EXPOSURE at: 2022-05-12 21:52:23   @ 10.6 sec   > Mean: 157.  Next: 9.9 sec
STARTING EXPOSURE at: 2022-05-12 21:53:24   @ 10.1 sec   > Mean: 151.  Next: 9.5 sec
STARTING EXPOSURE at: 2022-05-12 21:54:24   @ 9.6 sec    > Mean: 143.  Next: 9.2 sec
STARTING EXPOSURE at: 2022-05-12 21:55:25   @ 9.3 sec    > Mean: 127.  Next: 9.3 sec
STARTING EXPOSURE at: 2022-05-12 21:56:25   @ 9.3 sec    > Mean: 165.  Next: 12.1 sec
~~~

EricClaeys, 2022-05-16:
Here's the file...
[capture-0_8_3_3.txt](https://github.com/thomasjacquin/allsky/files/8698199/capture-0_8_3_3.txt)

ennui2342, 2022-05-18:
Hi Eric,

Here's the header you asked for:

[allsky.log.txt](https://github.com/thomasjacquin/allsky/files/8720302/allsky.log.txt)

Autogain was already off. I've turned off stretching. New capture file compiled successfully and in place for tonight, seems to be capturing okay in the day so will see how it goes...

-Mark

ennui2342, 2022-05-19:
Here's the results from last night...

https://user-images.githubusercontent.com/912310/169288211-b8199847-0985-4a70-b80a-ba3148077a15.mp4

[allsky.log.txt](https://github.com/thomasjacquin/allsky/files/8728839/allsky.log.txt)

EricClaeys, 2022-05-27:
@ennui2342 Mark,
Try the attached file.

~~~shell
sudo systemctl stop allsky
cd ~/allsky/src
mv capture.cpp capture-OLD.cpp
# move the attached file and rename to "capture.cpp"
make capture
cp capture ..
sudo truncate -s 0 /var/log/allsky.log
sudo systemctl start allsky
~~~
[capture-0_8_3_3.txt](https://github.com/thomasjacquin/allsky/files/8784722/capture-0_8_3_3.txt)

EricClaeys, 2022-06-03:
@bribon2, @Jonk2, @Apennine2, @ennui2342,
If you are still having problems with flickering when there are clouds and the sun (daytime) or moon (nighttime) is out, try changing your **USB Bandwidth** in the WebUI to 40.   Also try changing the **Auto USB**.
I was reading  through some older Issues, and lowering the USB Bandwidth fixed the flickering for another user.  I have no idea why, though.

Apennine2, 2022-06-03:
@EricClaeys,

Thanks for that suggestion. I updated to the latest software version and
set the USB Bandwidth as suggested.

-Chris

EricClaeys, 2022-06-03:
@Apennine2, please keep me informed.  I have my fingers crossed...

ennui2342, 2022-06-18:
Hi Eric,

I tried with AutoUSB off and setting USB bandwidth to 40 with the original capture binary, but didn't solve the issue. I installed your latest capture.cpp and ran it for a few minutes during a cloudy day - the logs attached. Seems to be all over the place, but maybe there's something in the log that's helpful to you.

-Mark

[allsky.log](https://github.com/thomasjacquin/allsky/files/8933391/allsky.log)

EricClaeys, 2022-06-19:
@ennui2342     Mark,   If you haven't already tried with **Version 0.8 Exposure** OFF, then please do so.
Either way, do you have a powered USB hub you can try?  Or a more powerful power brick for the Pi and/or a thicker wire going to the Pi?

ennui2342, 2022-06-19:
I have it running off a 12v 20A supply with a buck converter to 5v. I took it apart today to test the voltage and I'm getting a steady 5v. I dug around a bit to see if there is a power issue and I did find something of note. When I have 0.8 exposure on I get these messages every capture in syslog which from googling seem to be a particular thing with the 3b+:

usb 1-1.1.2: reset high-speed USB device number 4 using dwc_otg

I tried a couple of settings in my boot config: max_usb_current=1 to raise the power to 1200ma which didn't help and downgrading to USB1 with dwc_otg.speed=1 which also didn't help. However when I turned off 0.8 exposure the messages disappeared for both auto and manual USB bandwidth. However, the exposure is all over the place with lots of retries every capture, so manual USB bandwidth at 40 and 0.8 exposure off isn't the answer. But it's probably relevant that 0.8 exposure is causing the above USB resets.

One other thing to note about my setup which takes it from a clean install is that I have the pi connected to both wifi and running its own access point. I can't imagine this is causing power or USB issues - the wifi chip goes through SDIO not USB, but it is something different from a vanilla install so I thought I'd mention it.

It could be the 3b+, but I did swap that out just a couple of months ago because I had a board die, and I don't have another to test with (and with current supply chain issues I'm not likely to anytime soon).

ennui2342, 2022-06-19:
For example only with 0.8 exposure these usb message appear each capture:

> Jun 19 15:57:54 allsky allsky.sh[17614]:   > Got image @ mean 122.  Reported exposure: 295 us, auto=no
Jun 19 15:57:54 allsky allsky.sh[17614]: === next exposure=361 (multiply by 1.226) [last_exposure_us=295, minAcceptable=136, mean=122]
Jun 19 15:57:54 allsky allsky.sh[17614]:  >xxx mean was 160 and went from 10 above max of 150 to 14 below min of 136, is now at 122; should NOT set temp min to current_exposure_us of 295
Jun 19 15:57:54 allsky allsky.sh[17614]:   >> Retry 13 @ 305 us, min=295 us, max=305 us: mean (122)
Jun 19 15:57:54 allsky allsky.sh[17614]:   > Exposure set to 305 us, timeout: 5000 ms
**Jun 19 15:57:54 allsky kernel: [ 3519.522081] usb 1-1.1.2: reset high-speed USB device number 4 using dwc_otg**
Jun 19 15:57:54 allsky allsky.sh[17614]:   > Got image @ mean 160.  Reported exposure: 305 us, auto=no
Jun 19 15:57:54 allsky allsky.sh[17614]: === next exposure=259 (multiply by 0.852) [last_exposure_us=305, maxAcceptable=150, mean=160]
Jun 19 15:57:54 allsky allsky.sh[17614]:  >xxx mean was 122 and went from 14 below min of 136 to 10 above max of 150, is now at 160; OK to set temp max to current_exposure_us of 305
Jun 19 15:57:54 allsky allsky.sh[17614]:   >> Retry 14 @ 295 us, min=295 us, max=305 us: mean (160)
Jun 19 15:57:54 allsky allsky.sh[17614]:   > Exposure set to 295 us, timeout: 5000 ms
**Jun 19 15:57:55 allsky kernel: [ 3520.122111] usb 1-1.1.2: reset high-speed USB device number 4 using dwc_otg**
Jun 19 15:57:55 allsky allsky.sh[17614]:   > Got image @ mean 122.  Reported exposure: 295 us, auto=no
Jun 19 15:57:55 allsky allsky.sh[17614]: === next exposure=361 (multiply by 1.226) [last_exposure_us=295, minAcceptable=136, mean=122]
Jun 19 15:57:55 allsky allsky.sh[17614]:  >xxx mean was 160 and went from 10 above max of 150 to 14 below min of 136, is now at 122; should NOT set temp min to current_exposure_us of 295
Jun 19 15:57:55 allsky allsky.sh[17614]:   >> Retry 15 @ 305 us, min=295 us, max=305 us: mean (122)
Jun 19 15:57:55 allsky allsky.sh[17614]:   > Exposure set to 305 us, timeout: 5000 ms
**Jun 19 15:57:55 allsky kernel: [ 3520.722170] usb 1-1.1.2: reset high-speed USB device number 4 using dwc_otg**
Jun 19 15:57:56 allsky allsky.sh[17614]:   > Got image @ mean 160.  Reported exposure: 305 us, auto=no
Jun 19 15:57:56 allsky allsky.sh[17614]:   > max attempts reached - using exposure of 305 us us with mean 160
Jun 19 15:57:56 allsky allsky.sh[17614]:   >>>> Skipping this frame
Jun 19 15:57:56 allsky allsky.sh[17614]: STARTING EXPOSURE at: 2022-06-19 15:57:56   @ 305 us
Jun 19 15:57:56 allsky allsky.sh[17614]:   > Exposure set to 305 us, timeout: 5000 ms
**Jun 19 15:57:56 allsky kernel: [ 3521.392077] usb 1-1.1.2: reset high-speed USB device number 4 using dwc_otg**

Apennine2, 2022-06-19:
@EricClaeys ,

USB to 40, and auto bandwidth off also did not stop my flickering. This is with 0.8 ON. Here is a representative video:

[6/15/22](https://www.thewanderers.xyz/allsky/videos/allsky-20220615.mp4)

Here is the log from that night:

[allsky_6_15_16_22.txt](https://github.com/thomasjacquin/allsky/files/8935858/allsky_6_15_16_22.txt)

I have a 3b, powered with PoE. I don't notice the same usb resets as @ennui2342 . I notice I have debug level 3 set however, will that only show with level 4?

Thank you,

Chris

Jonk2, 2022-06-19:
I also confirm that it's not USB bandwith or power causing this, it's exposure hunting.

I have Pi4B, USB3-M.2 drive, a 178-MC cool powered directly by a 12V 5A supply. No power or bandwidth issues.

In a Windows allsky program I used to use, the method to get around changes of brightness over time (clouds etc.) was for the user to set how many exposures until the next target exposure time was reached, thus smoothing transitions.

For example, lots of small clouds passing through quickly will cause a lot of exposure changes, so I might set this setting to 3 exposures (which in effect I guess would be the same as the 'agression' setting used in this software).

Whereas on a clear moonlit night the exposure changes should be minimal, so once the optimal exposure time is met, it shouldn't fluctuate.

On the issue above about high speed resets this is found in the kernel log file at /var/log/kern.log

Here's mine from the past couple of minutes:

`Jun 19 17:26:24 allsky kernel: [24776.359352] usb 1-1.4: reset high-speed USB device number 3 using xhci_hcd
Jun 19 17:26:40 allsky kernel: [24792.629701] usb 1-1.4: reset high-speed USB device number 3 using xhci_hcd
Jun 19 17:26:42 allsky kernel: [24793.839643] usb 1-1.4: reset high-speed USB device number 3 using xhci_hcd
Jun 19 17:26:43 allsky kernel: [24795.059638] usb 1-1.4: reset high-speed USB device number 3 using xhci_hcd
Jun 19 17:26:44 allsky kernel: [24796.269668] usb 1-1.4: reset high-speed USB device number 3 using xhci_hcd
Jun 19 17:26:45 allsky kernel: [24797.479703] usb 1-1.4: reset high-speed USB device number 3 using xhci_hcd
Jun 19 17:26:46 allsky kernel: [24798.689709] usb 1-1.4: reset high-speed USB device number 3 using xhci_hcd
Jun 19 17:26:48 allsky kernel: [24799.919872] usb 1-1.4: reset high-speed USB device number 3 using xhci_hcd
Jun 19 17:26:49 allsky kernel: [24801.149917] usb 1-1.4: reset high-speed USB device number 3 using xhci_hcd
Jun 19 17:26:50 allsky kernel: [24802.369761] usb 1-1.4: reset high-speed USB device number 3 using xhci_hcd
Jun 19 17:26:51 allsky kernel: [24803.579780] usb 1-1.4: reset high-speed USB device number 3 using xhci_hcd
Jun 19 17:26:53 allsky kernel: [24804.799814] usb 1-1.4: reset high-speed USB device number 3 using xhci_hcd
Jun 19 17:26:54 allsky kernel: [24806.019817] usb 1-1.4: reset high-speed USB device number 3 using xhci_hcd
Jun 19 17:26:55 allsky kernel: [24807.239849] usb 1-1.4: reset high-speed USB device number 3 using xhci_hcd
Jun 19 17:26:56 allsky kernel: [24808.459869] usb 1-1.4: reset high-speed USB device number 3 using xhci_hcd
Jun 19 17:26:57 allsky kernel: [24809.679879] usb 1-1.4: reset high-speed USB device number 3 using xhci_hcd
Jun 19 17:26:59 allsky kernel: [24810.890061] usb 1-1.4: reset high-speed USB device number 3 using xhci_hcd
Jun 19 17:27:15 allsky kernel: [24827.200236] usb 1-1.4: reset high-speed USB device number 3 using xhci_hcd
Jun 19 17:27:16 allsky kernel: [24828.410190] usb 1-1.4: reset high-speed USB device number 3 using xhci_hcd
Jun 19 17:27:17 allsky kernel: [24829.620204] usb 1-1.4: reset high-speed USB device number 3 using xhci_hcd
Jun 19 17:27:19 allsky kernel: [24830.840223] usb 1-1.4: reset high-speed USB device number 3 using xhci_hcd
Jun 19 17:27:20 allsky kernel: [24832.070234] usb 1-1.4: reset high-speed USB device number 3 using xhci_hcd
Jun 19 17:27:21 allsky kernel: [24833.290267] usb 1-1.4: reset high-speed USB device number 3 using xhci_hcd
Jun 19 17:27:22 allsky kernel: [24834.500277] usb 1-1.4: reset high-speed USB device number 3 using xhci_hcd
Jun 19 17:27:23 allsky kernel: [24835.710288] usb 1-1.4: reset high-speed USB device number 3 using xhci_hcd
Jun 19 17:27:25 allsky kernel: [24836.930319] usb 1-1.4: reset high-speed USB device number 3 using xhci_hcd
Jun 19 17:27:26 allsky kernel: [24838.150382] usb 1-1.4: reset high-speed USB device number 3 using xhci_hcd
Jun 19 17:27:27 allsky kernel: [24839.360354] usb 1-1.4: reset high-speed USB device number 3 using xhci_hcd
Jun 19 17:27:28 allsky kernel: [24840.580380] usb 1-1.4: reset high-speed USB device number 3 using xhci_hcd
Jun 19 17:27:29 allsky kernel: [24841.790397] usb 1-1.4: reset high-speed USB device number 3 using xhci_hcd
Jun 19 17:27:31 allsky kernel: [24843.020411] usb 1-1.4: reset high-speed USB device number 3 using xhci_hcd
Jun 19 17:27:32 allsky kernel: [24844.230442] usb 1-1.4: reset high-speed USB device number 3 using xhci_hcd
Jun 19 17:27:33 allsky kernel: [24845.450451] usb 1-1.4: reset high-speed USB device number 3 using xhci_hcd
Jun 19 17:27:49 allsky kernel: [24861.730788] usb 1-1.4: reset high-speed USB device number 3 using xhci_hcd
Jun 19 17:27:51 allsky kernel: [24862.940958] usb 1-1.4: reset high-speed USB device number 3 using xhci_hcd
Jun 19 17:28:07 allsky kernel: [24879.231077] usb 1-1.4: reset high-speed USB device number 3 using xhci_hcd
Jun 19 17:28:23 allsky kernel: [24895.471258] usb 1-1.4: reset high-speed USB device number 3 using xhci_hcd
Jun 19 17:28:39 allsky kernel: [24911.751578] usb 1-1.4: reset high-speed USB device number 3 using xhci_hcd
Jun 19 17:28:56 allsky kernel: [24928.031908] usb 1-1.4: reset high-speed USB device number 3 using xhci_hcd
Jun 19 17:29:12 allsky kernel: [24944.292028] usb 1-1.4: reset high-speed USB device number 3 using xhci_hcd
Jun 19 17:29:28 allsky kernel: [24960.532291] usb 1-1.4: reset high-speed USB device number 3 using xhci_hcd`

I think this is due to the camera sdk / driver / software / library (ASICamera.h / ASICamera.a etc.). which interestingly, the current version of allsky is bundled with v1.2 whereas the latest available is V1.22.

I even put my camera on a USB2 port rather than USB3, but no difference.

`Jun 19 10:33:35 allsky allsky.sh[930]:  ZWO SDK version 1, 20`

Is there any reasno the latest is not being used?

EricClaeys, 2022-06-21:
@ennui2342   Mark,  The USB resets occur every time the video mode on the camera is turned on.  With version 0.8 ON, that's every image.  With it OFF, video mode is turned on once, when the program starts so you will only see one reset message.  That's ZWO's library doing that, and I have no idea why.  The messages are harmless from what I can tell.

@Jonk2   Jon, the **Aggression** setting in the WebUI smooths out the exposure changes similar in functionality to the Windows program.  The targeted exposure change between frames is multiplied by the Aggression to get the actual change used.  For example, if the targeted exposure change is 2000 ms and Aggression is 75% it'll actually make a 1500 ms change (2000 * .75).  The RPi HQ version of the capture program does exactly what the Windows version does, and we plan to port that auto exposure algorithm to the ZWO at some point.
We're not using the latest ZWO library since there haven't been any notable changes versus what we have - only some new cameras supported.

@bribon2, @Jonk2, @Apennine2, @ennui2342,
If you have a powered USB 3.0 hub, please try using it to see if it helps the flicker.  I realize some of you have sufficient power going to the Pi, but we've noticed some Pi's produce marginal power on the USB ports, and a powered hub has helped some people.  So did changing the USB Bandwidth.  Unfortunately those don't help everyone.

Apennine2, 2022-06-21:
@EricClaeys,

Thank you, unfortunately my equipment is sealed and caulked in its outdoor
enclosure, so I won't be able to get to it until something breaks down. In
the next few months I might start building a second AllSky, so I can
experiment at that time. My lasting observation is I did not experience the
strobing, until upgrading to 0.8.X from 0.7.X software versions. But, even
with the 0.8 Exposure method turned off now, it still strobes until
settling in at its max exposure time.

Thanks again,

Chris

EricClaeys, 2022-06-22:
@Apennine2  Chris, several other people have also reported that version 0.7 didn't strobe, but 0.8 did.  Turning "0.8 exposure" on or off solved it for some people. Others solved it by playing with the USB settings. And some by changing power supplies or adding a powered USB hub.  And some have had no luck with anything.  To a large extent this is also true with the ASI_ERROR_TIMEOUT messages.  Most people don't see them, some of the people that HAVE seen them did something that fixed it, and a few people still have problems. 

Unfortunately neither of my ZWO cameras exhibit stobing so it's hard for me to test.  If you are willing I would like to send some test versions of the capture program.  One is basically the 0.7 version that understands the changed names of some of the settings.

[9 more replies]

closed 2023-06-19 by EricClaeys