AllskyTeam/allsky #514: Permission problem with latest commits

WirthmU, 2021-09-28:
I get the following errors when I install from github running ./install.sh:

```
* Compile allsky software
Package opencv4 was not found in the pkg-config search path.
Perhaps you should add the directory containing `opencv4.pc'
to the PKG_CONFIG_PATH environment variable
No package 'opencv4' found
Package opencv4 was not found in the pkg-config search path.
Perhaps you should add the directory containing `opencv4.pc'
to the PKG_CONFIG_PATH environment variable
No package 'opencv4' found
arm-linux-gnueabihf-g++  capture.cpp -o capture -Wall -Wno-psabi -g -O2 -lpthread -march=armv7 -mthumb -D_LIN -D_DEBUG -DGLIBC_20 -Llib/armv7 -I./include -I/usr/include/opencv -lopencv_shape -lopencv_stitching -lopencv_superres -lopencv_videostab -lopencv_aruco -lopencv_bgsegm -lopencv_bioinspired -lopencv_ccalib -lopencv_datasets -lopencv_dpm -lopencv_face -lopencv_freetype -lopencv_fuzzy -lopencv_hdf -lopencv_line_descriptor -lopencv_optflow -lopencv_video -lopencv_plot -lopencv_reg -lopencv_saliency -lopencv_stereo -lopencv_structured_light -lopencv_phase_unwrapping -lopencv_rgbd -lopencv_viz -lopencv_surface_matching -lopencv_text -lopencv_ximgproc -lopencv_calib3d -lopencv_features2d -lopencv_flann -lopencv_xobjdetect -lopencv_objdetect -lopencv_ml -lopencv_xphoto -lopencv_highgui -lopencv_videoio -lopencv_imgcodecs -lopencv_photo -lopencv_imgproc -lopencv_core -lASICamera2 -I/usr/include/libusb-1.0 -lusb-1.0
/usr/bin/ld: cannot open output file capture: Permission denied
collect2: error: ld returned 1 exit status
make: *** [Makefile:73: capture] Error 1

chmod: changing permissions of 'allsky.sh': Operation not permitted
chmod: changing permissions of 'scripts/allsky_mfs.sh': Operation not permitted
chmod: changing permissions of 'scripts/copy_notification_image.sh': Operation not permitted
chmod: changing permissions of 'scripts/darkCapture.sh': Operation not permitted
chmod: changing permissions of 'scripts/darkSubtract.sh': Operation not permitted
chmod: changing permissions of 'scripts/endOfNight_additionalSteps.sh': Operation not permitted
chmod: changing permissions of 'scripts/endOfNight.sh': Operation not permitted
chmod: changing permissions of 'scripts/filename.sh': Operation not permitted
chmod: changing permissions of 'scripts/generateForDay.sh': Operation not permitted
chmod: changing permissions of 'scripts/postData.sh': Operation not permitted
chmod: changing permissions of 'scripts/removeBadImages.sh': Operation not permitted
chmod: changing permissions of 'scripts/saveImageDay.sh': Operation not permitted
chmod: changing permissions of 'scripts/saveImageNight.sh': Operation not permitted
chmod: changing permissions of 'scripts/timelapse.sh': Operation not permitted
chmod: changing permissions of 'scripts/uploadForDay.sh': Operation not permitted
* Sunwait installation
* Allow using the camera without root access

* Autostart script
sed: can't read /etc/xdg/lxsession/LXDE-pi/autostart: No such file or directory
sed: couldn't open temporary file autostart/sedosXhbN: Permission denied
sed: couldn't open temporary file autostart/sedTf5KmM: Permission denied

* Configure log rotation

* Add ALLSKY_HOME environment variable

* Copy camera settings files
cp: cannot create regular file 'settings_ZWO.json': Permission denied
cp: cannot create regular file 'settings_RPiHQ.json': Permission denied
cp: cannot create regular file 'config.sh': Permission denied
cp: cannot create regular file 'scripts/ftp-settings.sh': Permission denied

The Allsky Software is now installed. You should reboot the Raspberry Pi to finish the installation
```

linuxkidd, 2021-09-28:
Grr.... fixing..

linuxkidd, 2021-09-28:
So, there is definitely at least one bug in the `install.sh`, but it looks like the `git clone` was performed as root?

Can you check if the files are owned by `root`..  and if so, run:
`sudo chown pi:pi -R ~pi/allsky`

-- You may need to adjust the username / path to allsky if you're not doing things the default way.

Once that's done, try running the `install.sh` script again as the normal user.  You should only be left with a single permission error ( if you have `/etc/xdg/lxsession/LXDE-pi/autostart` present ).

Thanks!

PR #515 (2021-09-28): Fixed missing sudo on sed /etc/xdg/lxsession/LXDE-pi/autostart
Resolves #514 

There was one missing sudo, but it wasn't the full contributor to all the errors.  Looks like the local git repo was owned by root, but the compile / install was tried as regular user.  Regular user `install.sh` will work, but the git repo must be owned by that user.
closed 2021-09-28 by ckuethe