AllskyTeam/allsky #516: Can't compile latest commits

dbainbridge, 2021-09-28:
capture.cpp:1:28: fatal error: opencv2/core.hpp: No such file or directory

Also when trying make install I get:
Package opencv4 was not found in the pkg-config search path.

ckuethe, 2021-09-28:
What version of raspbian?

dbainbridge, 2021-09-28:
Raspbian GNU/Linux 9 (stretch)

ckuethe, 2021-09-28:
Did this ever work for you? If so, which was the last working version? Any interest in doing a [git bisect](https://www.metaltoad.com/blog/beginners-guide-git-bisect-process-elimination) to figure out when it broke?

Don't worry about opencv4 not being found. That comes from [Makefile:11](https://github.com/thomasjacquin/allsky/blob/master/Makefile#L11) where we try compute the right opencv flags.

Can you paste the complete output that failed? It looks like this: `g++  capture.cpp -o capture -Wall -Wno-psabi -g -O2 -lpthread -D_LIN -D_DEBUG -DGLIBC_20 ... -lASICamera2 -I/usr/include/libusb-1.0 -lusb-1.0`

dbainbridge, 2021-09-28:
I was able to at least build master a month or so ago when I commented on the issue about the time outs occurring.  I still had other problems such as the daytime issue being completely overexposed so I am hoping everything is fixed to a useable state again on master. 

I decided to just update raspbian to latest so can't test on stretch anymore.

If this doesn't work I'm going to go back to the last commit in June of this year since I can't keep having my camera down from all these bugs I'm running into with later versions.

linuxkidd, 2021-09-28:
@dbainbridge don't give up.. we should be able to get this fixed pretty quickly.

Can you provide the output of the following commands:

- `pkg-config --libs opencv`
- `pkg-config --cflags opencv`
- `pkg-config --cflags --libs opencv`
- `dpkg -l | grep -i opencv`

dbainbridge, 2021-09-29:
I reformatted SD card and installed Raspbian GNU/Linux 10 (buster)
and the issues went away.

ckuethe, 2021-09-29:
Glad to hear it, though we would have been happy to investigate to figure out what would need to be done to make it backwards compatible to stretch.

EricClaeys, 2021-10-01:
Closing - issue resolved when user reinstalled Raspbian.

WirthmU, 2021-10-01:
I am using Raspbian GNU/Linux 10 (buster) but still get the opencv4 error:
```
Package opencv4 was not found in the pkg-config search path.
Perhaps you should add the directory containing `opencv4.pc'
to the PKG_CONFIG_PATH environment variable
No package 'opencv4' found
```
Checking gives the following:
```
pi@allsky:~ $ pkg-config --libs opencv
-lopencv_shape -lopencv_stitching -lopencv_superres -lopencv_videostab -lopencv_aruco -lopencv_bgsegm -lopencv_bioinspired -lopencv_ccalib -lopencv_datasets -lopencv_dpm -lopencv_face -lopencv_freetype -lopencv_fuzzy -lopencv_hdf -lopencv_line_descriptor -lopencv_optflow -lopencv_video -lopencv_plot -lopencv_reg -lopencv_saliency -lopencv_stereo -lopencv_structured_light -lopencv_phase_unwrapping -lopencv_rgbd -lopencv_viz -lopencv_surface_matching -lopencv_text -lopencv_ximgproc -lopencv_calib3d -lopencv_features2d -lopencv_flann -lopencv_xobjdetect -lopencv_objdetect -lopencv_ml -lopencv_xphoto -lopencv_highgui -lopencv_videoio -lopencv_imgcodecs -lopencv_photo -lopencv_imgproc -lopencv_core
```
```
pi@allsky:~ $ pkg-config --cflags opencv
-I/usr/include/opencv
```
```
pi@allsky:~ $ pkg-config --cflags --libs opencv
-I/usr/include/opencv -lopencv_shape -lopencv_stitching -lopencv_superres -lopencv_videostab -lopencv_aruco -lopencv_bgsegm -lopencv_bioinspired -lopencv_ccalib -lopencv_datasets -lopencv_dpm -lopencv_face -lopencv_freetype -lopencv_fuzzy -lopencv_hdf -lopencv_line_descriptor -lopencv_optflow -lopencv_video -lopencv_plot -lopencv_reg -lopencv_saliency -lopencv_stereo -lopencv_structured_light -lopencv_phase_unwrapping -lopencv_rgbd -lopencv_viz -lopencv_surface_matching -lopencv_text -lopencv_ximgproc -lopencv_calib3d -lopencv_features2d -lopencv_flann -lopencv_xobjdetect -lopencv_objdetect -lopencv_ml -lopencv_xphoto -lopencv_highgui -lopencv_videoio -lopencv_imgcodecs -lopencv_photo -lopencv_imgproc -lopencv_core
```
```
pi@allsky:~ $ dpkg -l | grep -i opencv
ii  libopencv-calib3d-dev:armhf           3.2.0+dfsg-6                            armhf        development files for libopencv-calib3d3.2
ii  libopencv-calib3d3.2:armhf            3.2.0+dfsg-6                            armhf        computer vision Camera Calibration library
ii  libopencv-contrib-dev:armhf           3.2.0+dfsg-6                            armhf        development files for libopencv-contrib3.2
ii  libopencv-contrib3.2:armhf            3.2.0+dfsg-6                            armhf        computer vision contrlib library
ii  libopencv-core-dev:armhf              3.2.0+dfsg-6                            armhf        development files for libopencv-core3.2
ii  libopencv-core3.2:armhf               3.2.0+dfsg-6                            armhf        computer vision core library
ii  libopencv-dev                         3.2.0+dfsg-6                            armhf        development files for opencv
ii  libopencv-features2d-dev:armhf        3.2.0+dfsg-6                            armhf        development files for libopencv-features2d3.2
ii  libopencv-features2d3.2:armhf         3.2.0+dfsg-6                            armhf        computer vision Feature Detection and Descriptor Extraction library
ii  libopencv-flann-dev:armhf             3.2.0+dfsg-6                            armhf        development files for libopencv-flann3.2
ii  libopencv-flann3.2:armhf              3.2.0+dfsg-6                            armhf        computer vision Clustering and Search in Multi-Dimensional spaces library
ii  libopencv-highgui-dev:armhf           3.2.0+dfsg-6                            armhf        development files for libopencv-highgui3.2
ii  libopencv-highgui3.2:armhf            3.2.0+dfsg-6                            armhf        computer vision High-level GUI and Media I/O library
ii  libopencv-imgcodecs-dev:armhf         3.2.0+dfsg-6                            armhf        development files for libopencv-imgcodecs3.2
ii  libopencv-imgcodecs3.2:armhf          3.2.0+dfsg-6                            armhf        computer vision Image Codecs library
ii  libopencv-imgproc-dev:armhf           3.2.0+dfsg-6                            armhf        development files for libopencv-imgproc3.2
ii  libopencv-imgproc3.2:armhf            3.2.0+dfsg-6                            armhf        computer vision Image Processing library
ii  libopencv-ml-dev:armhf                3.2.0+dfsg-6                            armhf        development files for libopencv-ml3.2
ii  libopencv-ml3.2:armhf                 3.2.0+dfsg-6                            armhf        computer vision Machine Learning library
ii  libopencv-objdetect-dev:armhf         3.2.0+dfsg-6                            armhf        development files for libopencv-objdetect3.2
ii  libopencv-objdetect3.2:armhf          3.2.0+dfsg-6                            armhf        computer vision Object Detection library
ii  libopencv-photo-dev:armhf             3.2.0+dfsg-6                            armhf        development files for libopencv-photo3.2
ii  libopencv-photo3.2:armhf              3.2.0+dfsg-6                            armhf        computer vision computational photography library
ii  libopencv-shape-dev:armhf             3.2.0+dfsg-6                            armhf        development files for libopencv-shape3.2
ii  libopencv-shape3.2:armhf              3.2.0+dfsg-6                            armhf        computer vision shape descriptors and matchers library
[... cut ...]
ii  libopencv-viz3.2:armhf                3.2.0+dfsg-6                            armhf        computer vision 3D data visualization library
ii  libopencv3.2-java                     3.2.0+dfsg-6                            all          Java bindings for the computer vision library
ii  libopencv3.2-jni                      3.2.0+dfsg-6                            armhf        Java jni library for the computer vision library
ii  opencv-data                           3.2.0+dfsg-6                            all          development data for opencv
```

ckuethe, 2021-10-01:
Yay! Reproducible error! We can fix it

linuxkidd, 2021-10-01:
Closing this one again since it was original a problem with Debian Stretch.  Continuing to investigate via #523 

Thanks!

closed 2021-10-01 by linuxkidd