AllskyTeam/allsky #518: Parameter keogram/startrail in script generateForDay.sh

hoetzgit, 2021-09-29:
Hi, please correct the parameters for the call of keogram and startrail (-b, -e, -o ...)

ckuethe, 2021-09-29:
Can you give more details about the problem? Are you sure your installation is up to date, eg. [here](https://github.com/thomasjacquin/allsky/blob/master/scripts/endOfNight.sh#L47) and [here](https://github.com/thomasjacquin/allsky/blob/master/scripts/endOfNight.sh#L75)

```
~/allsky/scripts$ bash ./endOfNight.sh 20210928
endOfNight.sh: Generating Keogram

endOfNight.sh: Generating Startrails

Minimum: 0.109428 maximum: 0.662406 mean: 0.187987 median: 0.150412
endOfNight.sh: Generating Timelapse

timelapse.sh: Creating symlinks to generate timelapse
timelapse.sh: Created 5293 links total
[swscaler @ 0x55b8d9257bc0] deprecated pixel format used, make sure you did set range correctly
[swscaler @ 0x55b8d9b9ab80] deprecated pixel format used, make sure you did set range correctly
[swscaler @ 0x55b8d9b98380] deprecated pixel format used, make sure you did set range correctly
    Last message repeated 4 times
timelapse.sh: Deleting sequence
timelapse.sh: Timelapse was created

```

and while that was running, the following commands were invoked
```
/home/ckuethe/allsky/keogram -d /home/ckuethe/allsky/images/20210928/ -e jpg -o /home/ckuethe/allsky/images/20210928/keogram/keogram-20210928.jpg
/home/ckuethe/allsky/startrails -d /home/ckuethe/allsky/images/20210928/ -e jpg -b 0.35 -o /home/ckuethe/allsky/images/20210928/startrails/startrails-20210928.jpg
ffmpeg -y -f image2 -loglevel warning -r 25 -i /home/ckuethe/allsky/images/20210928/sequence/%04d.jpg -vcodec h264_nvenc -b:v 2000k -pix_fmt yuv420p -movflags +faststart /home/ckuethe/allsky/images/20210928/allsky-20210928.mp4
```

This produces the following output:
```
$ D="20210928" ; ls -lt ~/allsky/images/$D/{keogram,startrails}/* ~/allsky/images/$D/*mp4
-rwxrwxrwx 1 ckuethe 1006 52834165 Sep 29 12:11 /home/ckuethe/allsky/images/20210928/allsky-20210928.mp4
-rwxrwxrwx 1 ckuethe 1006   311326 Sep 29 12:09 /home/ckuethe/allsky/images/20210928/startrails/startrails-20210928.jpg
-rwxrwxrwx 1 ckuethe 1006  1575755 Sep 29 12:07 /home/ckuethe/allsky/images/20210928/keogram/keogram-20210928.jpg
```

hoetzgit, 2021-09-30:
endOfNight.sh is ok, but generateForDay.sh is without -d, -e, -o ...

ckuethe, 2021-09-30:
Ah, I see. Thanks for clarification.

EricClaeys, 2021-09-30:
The "Centralized upload" PR I submitted last night corrects this.

commit e5bb8469 (2021-10-01): generateForDay.sh

Fixes #518
closed 2021-10-01 by ckuethe