AllskyTeam/allsky #523: opencv4 Issue

WirthmU, 2021-10-01:
I am using Raspbian GNU/Linux 10 (buster) but still get the opencv4 error:
```
Package opencv4 was not found in the pkg-config search path.
Perhaps you should add the directory containing `opencv4.pc'
to the PKG_CONFIG_PATH environment variable
No package 'opencv4' found
```
Checking gives the following:
```
pi@allsky:~ $ pkg-config --libs opencv
-lopencv_shape -lopencv_stitching -lopencv_superres -lopencv_videostab -lopencv_aruco -lopencv_bgsegm -lopencv_bioinspired -lopencv_ccalib -lopencv_datasets -lopencv_dpm -lopencv_face -lopencv_freetype -lopencv_fuzzy -lopencv_hdf -lopencv_line_descriptor -lopencv_optflow -lopencv_video -lopencv_plot -lopencv_reg -lopencv_saliency -lopencv_stereo -lopencv_structured_light -lopencv_phase_unwrapping -lopencv_rgbd -lopencv_viz -lopencv_surface_matching -lopencv_text -lopencv_ximgproc -lopencv_calib3d -lopencv_features2d -lopencv_flann -lopencv_xobjdetect -lopencv_objdetect -lopencv_ml -lopencv_xphoto -lopencv_highgui -lopencv_videoio -lopencv_imgcodecs -lopencv_photo -lopencv_imgproc -lopencv_core
```
```
pi@allsky:~ $ pkg-config --cflags opencv
-I/usr/include/opencv
```
```
pi@allsky:~ $ pkg-config --cflags --libs opencv
-I/usr/include/opencv -lopencv_shape -lopencv_stitching -lopencv_superres -lopencv_videostab -lopencv_aruco -lopencv_bgsegm -lopencv_bioinspired -lopencv_ccalib -lopencv_datasets -lopencv_dpm -lopencv_face -lopencv_freetype -lopencv_fuzzy -lopencv_hdf -lopencv_line_descriptor -lopencv_optflow -lopencv_video -lopencv_plot -lopencv_reg -lopencv_saliency -lopencv_stereo -lopencv_structured_light -lopencv_phase_unwrapping -lopencv_rgbd -lopencv_viz -lopencv_surface_matching -lopencv_text -lopencv_ximgproc -lopencv_calib3d -lopencv_features2d -lopencv_flann -lopencv_xobjdetect -lopencv_objdetect -lopencv_ml -lopencv_xphoto -lopencv_highgui -lopencv_videoio -lopencv_imgcodecs -lopencv_photo -lopencv_imgproc -lopencv_core
```
```
pi@allsky:~ $ dpkg -l | grep -i opencv
ii  libopencv-calib3d-dev:armhf           3.2.0+dfsg-6                            armhf        development files for libopencv-calib3d3.2
ii  libopencv-calib3d3.2:armhf            3.2.0+dfsg-6                            armhf        computer vision Camera Calibration library
ii  libopencv-contrib-dev:armhf           3.2.0+dfsg-6                            armhf        development files for libopencv-contrib3.2
ii  libopencv-contrib3.2:armhf            3.2.0+dfsg-6                            armhf        computer vision contrlib library
ii  libopencv-core-dev:armhf              3.2.0+dfsg-6                            armhf        development files for libopencv-core3.2
ii  libopencv-core3.2:armhf               3.2.0+dfsg-6                            armhf        computer vision core library
ii  libopencv-dev                         3.2.0+dfsg-6                            armhf        development files for opencv
ii  libopencv-features2d-dev:armhf        3.2.0+dfsg-6                            armhf        development files for libopencv-features2d3.2
ii  libopencv-features2d3.2:armhf         3.2.0+dfsg-6                            armhf        computer vision Feature Detection and Descriptor Extraction library
ii  libopencv-flann-dev:armhf             3.2.0+dfsg-6                            armhf        development files for libopencv-flann3.2
ii  libopencv-flann3.2:armhf              3.2.0+dfsg-6                            armhf        computer vision Clustering and Search in Multi-Dimensional spaces library
ii  libopencv-highgui-dev:armhf           3.2.0+dfsg-6                            armhf        development files for libopencv-highgui3.2
ii  libopencv-highgui3.2:armhf            3.2.0+dfsg-6                            armhf        computer vision High-level GUI and Media I/O library
ii  libopencv-imgcodecs-dev:armhf         3.2.0+dfsg-6                            armhf        development files for libopencv-imgcodecs3.2
ii  libopencv-imgcodecs3.2:armhf          3.2.0+dfsg-6                            armhf        computer vision Image Codecs library
ii  libopencv-imgproc-dev:armhf           3.2.0+dfsg-6                            armhf        development files for libopencv-imgproc3.2
ii  libopencv-imgproc3.2:armhf            3.2.0+dfsg-6                            armhf        computer vision Image Processing library
ii  libopencv-ml-dev:armhf                3.2.0+dfsg-6                            armhf        development files for libopencv-ml3.2
ii  libopencv-ml3.2:armhf                 3.2.0+dfsg-6                            armhf        computer vision Machine Learning library
ii  libopencv-objdetect-dev:armhf         3.2.0+dfsg-6                            armhf        development files for libopencv-objdetect3.2
ii  libopencv-objdetect3.2:armhf          3.2.0+dfsg-6                            armhf        computer vision Object Detection library
ii  libopencv-photo-dev:armhf             3.2.0+dfsg-6                            armhf        development files for libopencv-photo3.2
ii  libopencv-photo3.2:armhf              3.2.0+dfsg-6                            armhf        computer vision computational photography library
ii  libopencv-shape-dev:armhf             3.2.0+dfsg-6                            armhf        development files for libopencv-shape3.2
ii  libopencv-shape3.2:armhf              3.2.0+dfsg-6                            armhf        computer vision shape descriptors and matchers library
ii  libopencv-stitching-dev:armhf         3.2.0+dfsg-6                            armhf        development files for libopencv-stitching3.2
ii  libopencv-stitching3.2:armhf          3.2.0+dfsg-6                            armhf        computer vision image stitching library
ii  libopencv-superres-dev:armhf          3.2.0+dfsg-6                            armhf        development files for libopencv-superres3.2
ii  libopencv-superres3.2:armhf           3.2.0+dfsg-6                            armhf        computer vision Super Resolution library
ii  libopencv-ts-dev:armhf                3.2.0+dfsg-6                            armhf        development files for TS library of OpenCV (Open Computer Vision)
ii  libopencv-video-dev:armhf             3.2.0+dfsg-6                            armhf        development files for libopencv-video3.2
ii  libopencv-video3.2:armhf              3.2.0+dfsg-6                            armhf        computer vision Video analysis library
ii  libopencv-videoio-dev:armhf           3.2.0+dfsg-6                            armhf        development files for libopencv-videoio3.2
ii  libopencv-videoio3.2:armhf            3.2.0+dfsg-6                            armhf        computer vision Video I/O library
ii  libopencv-videostab-dev:armhf         3.2.0+dfsg-6                            armhf        development files for libopencv-videostab3.2
ii  libopencv-videostab3.2:armhf          3.2.0+dfsg-6                            armhf        computer vision video stabilization library
ii  libopencv-viz-dev:armhf               3.2.0+dfsg-6                            armhf        development files for libopencv-viz3.2
ii  libopencv-viz3.2:armhf                3.2.0+dfsg-6                            armhf        computer vision 3D data visualization library
ii  libopencv3.2-java                     3.2.0+dfsg-6                            all          Java bindings for the computer vision library
ii  libopencv3.2-jni                      3.2.0+dfsg-6                            armhf        Java jni library for the computer vision library
ii  opencv-data                           3.2.0+dfsg-6                            all          development data for opencv
```

_Originally posted by @WirthmU in https://github.com/thomasjacquin/allsky/issues/516#issuecomment-932161262_

linuxkidd, 2021-10-01:
- Can you check the contents of the `Makefile` -- does it have the lines looking like [these in the repo version](https://github.com/thomasjacquin/allsky/blob/master/Makefile#L11-L16)  ( Lines 11 through 16 ).
- If there's any difference, please try updating your `allsky` contents so it matches what's in the repository and retry the `install.sh` script.

If they do look the same, can you provide the output of this command:

~~~
pkg-config --exists opencv && echo Yes || echo No
~~~

WirthmU, 2021-10-04:
With the latest commits the error is gone.

linuxkidd, 2021-10-04:
Great to hear!  We'll close the issue.

closed 2021-10-04 by linuxkidd