AllskyTeam/allsky #525: Current picture upload not working

WirthmU, 2021-10-01:
I have installed from the latest commit and I get the following in the log:

```
Capturing & saving image...
Oct  1 16:08:51 allsky allsky.sh[772]: Capturing & saving image done, now wait 50 seconds...
Oct  1 16:08:53 allsky allsky.sh[772]: /home/pi/allsky/scripts/upload.sh: line 93: [: : integer expression expected
Oct  1 16:08:53 allsky allsky.sh[772]: *** upload.sh: ERROR:
Oct  1 16:08:53 allsky allsky.sh[772]: FILE_TO_UPLOAD='image-resize.jpg'
Oct  1 16:08:53 allsky allsky.sh[772]: REMOTE_DIR='/site/allsky/'
Oct  1 16:08:53 allsky allsky.sh[772]: TEMP_NAME='SI-18378'
Oct  1 16:08:53 allsky allsky.sh[772]: DESTINATION_FILE='image-resize.jpg'
Oct  1 16:08:53 allsky allsky.sh[772]: put: image-resize.jpg: Access failed: 553 SI-18378: Permission denied.
Oct  1 16:08:53 allsky allsky.sh[772]: mv: Access failed: 553 SI-18378: No such file or directory.
Oct  1 16:08:53 allsky allsky.sh[772]: Commands used (run via: lftp -f /home/pi/allsky/tmp/lftp_cmds.txt):
Oct  1 16:08:53 allsky allsky.sh[772]: open --user "user" --password "password" ftp://ftp-adress
Oct  1 16:08:53 allsky allsky.sh[772]: set net:max-retries 2
Oct  1 16:08:53 allsky allsky.sh[772]: set net:timeout 20
Oct  1 16:08:53 allsky allsky.sh[772]: rm -f SI-18378
Oct  1 16:08:53 allsky allsky.sh[772]: put "image-resize.jpg" -o SI-18378
Oct  1 16:08:53 allsky allsky.sh[772]: rm -f "image-resize.jpg"
Oct  1 16:08:53 allsky allsky.sh[772]: mv SI-18378 "/site/allsky//image-resize.jpg"
Oct  1 16:08:53 allsky allsky.sh[772]: bye
```

EricClaeys, 2021-10-01:
@WirthmU , we will fix the error on the 3rd line of the log.
For the error in uploading the file using lftp, try running the commands manually.  Look in /home/pi/allsky/tmp/lftp_cmds.txt for the list of commands.  Did you change the acutal password, user name, and ftp address on this line?
`open --user "user" --password "password" ftp://ftp-adress`
If not, then you'll need to update ftp_settings.sh.

WirthmU, 2021-10-01:
Yes - user/password/ftp are not the actual ones.

EricClaeys, 2021-10-01:
Reopened since the failed lftp has not been resolved.
@WirthmU please download the updated upload.sh.  It should fix the line 93 problem.
Let me know what happens when you try running the commands manually.

EricClaeys, 2021-10-14:
@WirthmU, did the updated `upload.sh` fix the issue?

PR #527 (2021-10-01): upload.sh: Set default ALLSKY_DEBUG_LEVEL
Fixes #525 

This is a hot one - uploads are failing due to it.