AllskyTeam/allsky #526: Question on install.

maserowik, 2021-10-01:
Package opencv4 was not found in the pkg-config search path.
Perhaps you should add the directory containing `opencv4.pc'
to the PKG_CONFIG_PATH environment variable
No package 'opencv4' found
Package opencv4 was not found in the pkg-config search path.
Perhaps you should add the directory containing `opencv4.pc'
to the PKG_CONFIG_PATH environment variable
No package 'opencv4' found
arm-linux-gnueabihf-g++  capture.cpp -o capture -Wall -Wno-psabi -g -O2 -lpthread -march=armv7 -mthumb -D_LIN -D_DEBUG -DGLIBC_20 -Llib/armv7 -I./include -I/usr/include/opencv -lopencv_shape -lopencv_stitching -lopencv_superres -lopencv_videostab -lopencv_aruco -lopencv_bgsegm -lopencv_bioinspired -lopencv_ccalib -lopencv_datasets -lopencv_dpm -lopencv_face -lopencv_freetype -lopencv_fuzzy -lopencv_hdf -lopencv_line_descriptor -lopencv_optflow -lopencv_video -lopencv_plot -lopencv_reg -lopencv_saliency -lopencv_stereo -lopencv_structured_light -lopencv_phase_unwrapping -lopencv_rgbd -lopencv_viz -lopencv_surface_matching -lopencv_text -lopencv_ximgproc -lopencv_calib3d -lopencv_features2d -lopencv_flann -lopencv_xobjdetect -lopencv_objdetect -lopencv_ml -lopencv_xphoto -lopencv_highgui -lopencv_videoio -lopencv_imgcodecs -lopencv_photo -lopencv_imgproc -lopencv_core -lASICamera2 -I/usr/include/libusb-1.0 -lusb-1.0
Package opencv4 was not found in the pkg-config search path.
Perhaps you should add the directory containing `opencv4.pc'
to the PKG_CONFIG_PATH environment variable
No package 'opencv4' found
arm-linux-gnueabihf-g++  capture_RPiHQ.cpp -o capture_RPiHQ -Wall -Wno-psabi -g -O2 -lpthread -march=armv7 -mthumb -D_LIN -D_DEBUG -DGLIBC_20 -Llib/armv7 -I./include -I/usr/include/opencv -lopencv_shape -lopencv_stitching -lopencv_superres -lopencv_videostab -lopencv_aruco -lopencv_bgsegm -lopencv_bioinspired -lopencv_ccalib -lopencv_datasets -lopencv_dpm -lopencv_face -lopencv_freetype -lopencv_fuzzy -lopencv_hdf -lopencv_line_descriptor -lopencv_optflow -lopencv_video -lopencv_plot -lopencv_reg -lopencv_saliency -lopencv_stereo -lopencv_structured_light -lopencv_phase_unwrapping -lopencv_rgbd -lopencv_viz -lopencv_surface_matching -lopencv_text -lopencv_ximgproc -lopencv_calib3d -lopencv_features2d -lopencv_flann -lopencv_xobjdetect -lopencv_objdetect -lopencv_ml -lopencv_xphoto -lopencv_highgui -lopencv_videoio -lopencv_imgcodecs -lopencv_photo -lopencv_imgproc -lopencv_core
Package opencv4 was not found in the pkg-config search path.
Perhaps you should add the directory containing `opencv4.pc'
to the PKG_CONFIG_PATH environment variable
No package 'opencv4' found
arm-linux-gnueabihf-g++  startrails.cpp -o startrails -Wall -Wno-psabi -g -O2 -lpthread -march=armv7 -mthumb -D_LIN -D_DEBUG -DGLIBC_20 -Llib/armv7 -I./include -I/usr/include/opencv -lopencv_shape -lopencv_stitching -lopencv_superres -lopencv_videostab -lopencv_aruco -lopencv_bgsegm -lopencv_bioinspired -lopencv_ccalib -lopencv_datasets -lopencv_dpm -lopencv_face -lopencv_freetype -lopencv_fuzzy -lopencv_hdf -lopencv_line_descriptor -lopencv_optflow -lopencv_video -lopencv_plot -lopencv_reg -lopencv_saliency -lopencv_stereo -lopencv_structured_light -lopencv_phase_unwrapping -lopencv_rgbd -lopencv_viz -lopencv_surface_matching -lopencv_text -lopencv_ximgproc -lopencv_calib3d -lopencv_features2d -lopencv_flann -lopencv_xobjdetect -lopencv_objdetect -lopencv_ml -lopencv_xphoto -lopencv_highgui -lopencv_videoio -lopencv_imgcodecs -lopencv_photo -lopencv_imgproc -lopencv_core
Package opencv4 was not found in the pkg-config search path.
Perhaps you should add the directory containing `opencv4.pc'
to the PKG_CONFIG_PATH environment variable
No package 'opencv4' found
arm-linux-gnueabihf-g++  keogram.cpp -o keogram -Wall -Wno-psabi -g -O2 -lpthread -march=armv7 -mthumb -D_LIN -D_DEBUG -DGLIBC_20 -Llib/armv7 -I./include -I/usr/include/opencv -lopencv_shape -lopencv_stitching -lopencv_superres -lopencv_videostab -lopencv_aruco -lopencv_bgsegm -lopencv_bioinspired -lopencv_ccalib -lopencv_datasets -lopencv_dpm -lopencv_face -lopencv_freetype -lopencv_fuzzy -lopencv_hdf -lopencv_line_descriptor -lopencv_optflow -lopencv_video -lopencv_plot -lopencv_reg -lopencv_saliency -lopencv_stereo -lopencv_structured_light -lopencv_phase_unwrapping -lopencv_rgbd -lopencv_viz -lopencv_surface_matching -lopencv_text -lopencv_ximgproc -lopencv_calib3d -lopencv_features2d -lopencv_flann -lopencv_xobjdetect -lopencv_objdetect -lopencv_ml -lopencv_xphoto -lopencv_highgui -lopencv_videoio -lopencv_imgcodecs -lopencv_photo -lopencv_imgproc -lopencv_core
git submodule init
git submodule update
make -C sunwait-src
make[1]: Entering directory '/home/pi/allsky/sunwait-src'
gcc -c -Wall  sunwait.cpp -o sunwait.o
sunwait.cpp: In function ‘int main(int, char**)’:
sunwait.cpp:664:62: warning: value computed is not used [-Wunused-value]
     if (arg[0] == '-' && arg[1] != '\0' && !isdigit(arg[1])) *arg++;
                                                              ^~~~~~
gcc -c -Wall  sunriset.cpp -o sunriset.o
sunriset.cpp: In function ‘void sun_RA_dec(double, double*, double*, double*)’:
sunriset.cpp:151:18: warning: variable ‘zs’ set but not used [-Wunused-but-set-variable]
   double xs, ys, zs;
                  ^~
gcc -c -Wall  print.cpp -o print.o
gcc sunwait.o sunriset.o print.o sunwait.h sunriset.h print.h -o sunwait -lm -lstdc++
make[1]: Leaving directory '/home/pi/allsky/sunwait-src'
cp sunwait-src/sunwait .

* Sunwait installation
* Allow using the camera without root access

* Autostart script

* Configure log rotation

* Add ALLSKY_HOME environment variable

* Copy camera settings files

The Allsky Software is now installed. You should reboot the Raspberry Pi to finish the installation

Do you want to reboot now? [y/n] y

maserowik, 2021-10-01:
so after i install the latest version i am seeing this in the log file for allsky.

Oct  1 10:35:00 Pi-AllSky allsky.sh[2512]: /home/pi/allsky/scripts/upload.sh: line 93: [: : integer expression expected

something to worry about?

EricClaeys, 2021-10-01:
@maserowik I have a fix in for the "line 93" error.  If your uploads are working then you can ignore the message.

@ckuethe Chris, please take a look at the compiler error.

linuxkidd, 2021-10-01:
@maserowik 
- The `opencv4` warnings don't appear to be critical since the compile is succeeding otherwise.
- Can you check the contents of the `Makefile` -- does it have the lines looking like [these in the repo version](https://github.com/thomasjacquin/allsky/blob/master/Makefile#L11-L16)  ( Lines 11 through 16 ).
- If there's any difference, please try updating your `allsky` contents so it matches what's in the repository and retry the `install.sh` script.

**A bit of explanation:**
- OpenCV4 is only part of Raspberry Pi OS based on Debian Bullseye, not Buster
- Debian Buster is the default for new installs.
- We can see that the `pkg-config` is working as expected for the opencv version on your Pi because the compile commands include all the `opencv` declarations.
- The code on line 11 is designed to look for `opencv` first, and if found, use it.  If it's not found, it checks for `opencv4`.
- I suspect the `Makefile` in your `allsky` directory doesn't look like what I linked above -- an earlier version would produce the OpenCV4 errors because it tried to populate both versions without checking for their existence.

maserowik, 2021-10-01:
@linuxkidd 
Agree that it is currently not causing a issue with my current install.

I checked my file and it look exactly as what is in the repo.
here is what i have in lines 11- 16

OPENCV = $(shell pkg-config --exists opencv && pkg-config --cflags --libs opencv || pkg-config --exists opencv4 && pkg-config --cflags --libs opencv4)

ifeq (,$(OPENCV))
  $(warning Did not find any OpenCV Libraries, you may need to install libopencv-dev.)
  $(error Missing dependencies)
endif

on the repo

OPENCV = $(shell pkg-config --exists opencv && pkg-config --cflags --libs opencv || pkg-config --exists opencv4 && pkg-config --cflags --libs opencv4)

ifeq (,$(OPENCV))
  $(warning Did not find any OpenCV Libraries, you may need to install libopencv-dev.)
  $(error Missing dependencies)
endif

pi@Pi-AllSky:~/allsky $ cat /etc/os-release
PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"
VERSION_CODENAME=buster
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"

linuxkidd, 2021-10-01:
Thanks, @maserowik 
Can you provide the output of these commands:

~~~
pkg-config --exists opencv && echo Yes || echo No
pkg-config --exists opencv4 && echo Yes4 || echo No4
pkg-config --exists opencv && echo bYes || pkg-config --exists opencv4 && echo bYes4
~~~

Thanks!

maserowik, 2021-10-01:
@linuxkidd 
pi@Pi-AllSky:~/allsky $ pkg-config --exists opencv && echo Yes || echo No
Yes
pi@Pi-AllSky:~/allsky $ pkg-config --exists opencv4 && echo Yes4 || echo No4
No4
pi@Pi-AllSky:~/allsky $ pkg-config --exists opencv && echo bYes || pkg-config --exists opencv4 && echo bYes4
bYes
bYes4
pi@Pi-AllSky:~/allsky $

linuxkidd, 2021-10-01:
Ah, very interesting... the opencv4 bit is getting triggered even though opencv is present.  I'll fix up this logic...

maserowik, 2021-10-01:
pi@Pi-AllSky:~/allsky $ pkg-config --exists opencv && echo bYes || ( pkg-config --exists opencv4 && echo bYes4 )
bYes

linuxkidd, 2021-10-01:
Ok... thx @maserowik  -- would you be comfortable doing .. one more test.. :)

Edit your `Makefile`, change line 11 to this:
~~~
OPENCV = $(shell pkg-config --exists opencv && pkg-config --cflags --libs opencv || ( pkg-config --exists opencv4 && pkg-config --cflags --libs opencv4))
~~~

Save that, then run `make capture` and let us know if there are any odd warnings / errors.  :)

maserowik, 2021-10-01:
@linuxkidd 

pi@Pi-AllSky:~/allsky $ make capture
make: 'capture' is up to date.

i remember reading about this earlier need to reset the head??

maserowik, 2021-10-01:
pi@Pi-AllSky:~/allsky $ git reset --hard HEAD
HEAD is now at cc1b8fc Merge pull request #522 from thomasjacquin/capture-segmentation-fault
pi@Pi-AllSky:~/allsky $ make capture
Package opencv4 was not found in the pkg-config search path.
Perhaps you should add the directory containing `opencv4.pc'
to the PKG_CONFIG_PATH environment variable
No package 'opencv4' found
Package opencv4 was not found in the pkg-config search path.
Perhaps you should add the directory containing `opencv4.pc'
to the PKG_CONFIG_PATH environment variable
No package 'opencv4' found
arm-linux-gnueabihf-g++  capture.cpp -o capture -Wall -Wno-psabi -g -O2 -lpthread -march=armv7 -mthumb -D_LIN -D_DEBUG -DGLIBC_20 -Llib/armv7 -I./include -I/usr/include/opencv -lopencv_shape -lopencv_stitching -lopencv_superres -lopencv_videostab -lopencv_aruco -lopencv_bgsegm -lopencv_bioinspired -lopencv_ccalib -lopencv_datasets -lopencv_dpm -lopencv_face -lopencv_freetype -lopencv_fuzzy -lopencv_hdf -lopencv_line_descriptor -lopencv_optflow -lopencv_video -lopencv_plot -lopencv_reg -lopencv_saliency -lopencv_stereo -lopencv_structured_light -lopencv_phase_unwrapping -lopencv_rgbd -lopencv_viz -lopencv_surface_matching -lopencv_text -lopencv_ximgproc -lopencv_calib3d -lopencv_features2d -lopencv_flann -lopencv_xobjdetect -lopencv_objdetect -lopencv_ml -lopencv_xphoto -lopencv_highgui -lopencv_videoio -lopencv_imgcodecs -lopencv_photo -lopencv_imgproc -lopencv_core -lASICamera2 -I/usr/include/libusb-1.0 -lusb-1.0

linuxkidd, 2021-10-01:
Ah, ya, I haven't pushed that change to the repo .. was hope'n you could confirm it works properly on buster before I send it.

So, the steps would be:
1. Edit `Makefile` with your editor of choice, to make line 11 read:
~~~
OPENCV = $(shell pkg-config --exists opencv && pkg-config --cflags --libs opencv || ( pkg-config --exists opencv4 && pkg-config --cflags --libs opencv4))
~~~
2. Issue the following commands:
~~~
make clean
make all
~~~

That should re-compile the code.. and if the `pkg-config` messages are gone, we're golden.

maserowik, 2021-10-01:
pi@Pi-AllSky:~/allsky $ make clean
Package opencv4 was not found in the pkg-config search path.
Perhaps you should add the directory containing `opencv4.pc'
to the PKG_CONFIG_PATH environment variable
No package 'opencv4' found
rm -f capture capture_RPiHQ startrails keogram
pi@Pi-AllSky:~/allsky $ make all
Package opencv4 was not found in the pkg-config search path.
Perhaps you should add the directory containing `opencv4.pc'
to the PKG_CONFIG_PATH environment variable
No package 'opencv4' found
Package opencv4 was not found in the pkg-config search path.
Perhaps you should add the directory containing `opencv4.pc'
to the PKG_CONFIG_PATH environment variable
No package 'opencv4' found
arm-linux-gnueabihf-g++  capture.cpp -o capture -Wall -Wno-psabi -g -O2 -lpthread -march=armv7 -mthumb -D_LIN -D_DEBUG -DGLIBC_20 -Llib/armv7 -I./include -I/usr/include/opencv -lopencv_shape -lopencv_stitching -lopencv_superres -lopencv_videostab -lopencv_aruco -lopencv_bgsegm -lopencv_bioinspired -lopencv_ccalib -lopencv_datasets -lopencv_dpm -lopencv_face -lopencv_freetype -lopencv_fuzzy -lopencv_hdf -lopencv_line_descriptor -lopencv_optflow -lopencv_video -lopencv_plot -lopencv_reg -lopencv_saliency -lopencv_stereo -lopencv_structured_light -lopencv_phase_unwrapping -lopencv_rgbd -lopencv_viz -lopencv_surface_matching -lopencv_text -lopencv_ximgproc -lopencv_calib3d -lopencv_features2d -lopencv_flann -lopencv_xobjdetect -lopencv_objdetect -lopencv_ml -lopencv_xphoto -lopencv_highgui -lopencv_videoio -lopencv_imgcodecs -lopencv_photo -lopencv_imgproc -lopencv_core -lASICamera2 -I/usr/include/libusb-1.0 -lusb-1.0
Package opencv4 was not found in the pkg-config search path.
Perhaps you should add the directory containing `opencv4.pc'
to the PKG_CONFIG_PATH environment variable
No package 'opencv4' found
arm-linux-gnueabihf-g++  capture_RPiHQ.cpp -o capture_RPiHQ -Wall -Wno-psabi -g -O2 -lpthread -march=armv7 -mthumb -D_LIN -D_DEBUG -DGLIBC_20 -Llib/armv7 -I./include -I/usr/include/opencv -lopencv_shape -lopencv_stitching -lopencv_superres -lopencv_videostab -lopencv_aruco -lopencv_bgsegm -lopencv_bioinspired -lopencv_ccalib -lopencv_datasets -lopencv_dpm -lopencv_face -lopencv_freetype -lopencv_fuzzy -lopencv_hdf -lopencv_line_descriptor -lopencv_optflow -lopencv_video -lopencv_plot -lopencv_reg -lopencv_saliency -lopencv_stereo -lopencv_structured_light -lopencv_phase_unwrapping -lopencv_rgbd -lopencv_viz -lopencv_surface_matching -lopencv_text -lopencv_ximgproc -lopencv_calib3d -lopencv_features2d -lopencv_flann -lopencv_xobjdetect -lopencv_objdetect -lopencv_ml -lopencv_xphoto -lopencv_highgui -lopencv_videoio -lopencv_imgcodecs -lopencv_photo -lopencv_imgproc -lopencv_core
Package opencv4 was not found in the pkg-config search path.
Perhaps you should add the directory containing `opencv4.pc'
to the PKG_CONFIG_PATH environment variable
No package 'opencv4' found
arm-linux-gnueabihf-g++  startrails.cpp -o startrails -Wall -Wno-psabi -g -O2 -lpthread -march=armv7 -mthumb -D_LIN -D_DEBUG -DGLIBC_20 -Llib/armv7 -I./include -I/usr/include/opencv -lopencv_shape -lopencv_stitching -lopencv_superres -lopencv_videostab -lopencv_aruco -lopencv_bgsegm -lopencv_bioinspired -lopencv_ccalib -lopencv_datasets -lopencv_dpm -lopencv_face -lopencv_freetype -lopencv_fuzzy -lopencv_hdf -lopencv_line_descriptor -lopencv_optflow -lopencv_video -lopencv_plot -lopencv_reg -lopencv_saliency -lopencv_stereo -lopencv_structured_light -lopencv_phase_unwrapping -lopencv_rgbd -lopencv_viz -lopencv_surface_matching -lopencv_text -lopencv_ximgproc -lopencv_calib3d -lopencv_features2d -lopencv_flann -lopencv_xobjdetect -lopencv_objdetect -lopencv_ml -lopencv_xphoto -lopencv_highgui -lopencv_videoio -lopencv_imgcodecs -lopencv_photo -lopencv_imgproc -lopencv_core
Package opencv4 was not found in the pkg-config search path.
Perhaps you should add the directory containing `opencv4.pc'
to the PKG_CONFIG_PATH environment variable
No package 'opencv4' found
arm-linux-gnueabihf-g++  keogram.cpp -o keogram -Wall -Wno-psabi -g -O2 -lpthread -march=armv7 -mthumb -D_LIN -D_DEBUG -DGLIBC_20 -Llib/armv7 -I./include -I/usr/include/opencv -lopencv_shape -lopencv_stitching -lopencv_superres -lopencv_videostab -lopencv_aruco -lopencv_bgsegm -lopencv_bioinspired -lopencv_ccalib -lopencv_datasets -lopencv_dpm -lopencv_face -lopencv_freetype -lopencv_fuzzy -lopencv_hdf -lopencv_line_descriptor -lopencv_optflow -lopencv_video -lopencv_plot -lopencv_reg -lopencv_saliency -lopencv_stereo -lopencv_structured_light -lopencv_phase_unwrapping -lopencv_rgbd -lopencv_viz -lopencv_surface_matching -lopencv_text -lopencv_ximgproc -lopencv_calib3d -lopencv_features2d -lopencv_flann -lopencv_xobjdetect -lopencv_objdetect -lopencv_ml -lopencv_xphoto -lopencv_highgui -lopencv_videoio -lopencv_imgcodecs -lopencv_photo -lopencv_imgproc -lopencv_core
git submodule init
git submodule update
make -C sunwait-src
make[1]: Entering directory '/home/pi/allsky/sunwait-src'
gcc -c -Wall  sunwait.cpp -o sunwait.o
sunwait.cpp: In function ‘int main(int, char**)’:
sunwait.cpp:664:62: warning: value computed is not used [-Wunused-value]
     if (arg[0] == '-' && arg[1] != '\0' && !isdigit(arg[1])) *arg++;
                                                              ^~~~~~
gcc -c -Wall  sunriset.cpp -o sunriset.o
sunriset.cpp: In function ‘void sun_RA_dec(double, double*, double*, double*)’:
sunriset.cpp:151:18: warning: variable ‘zs’ set but not used [-Wunused-but-set-variable]
   double xs, ys, zs;
                  ^~
gcc -c -Wall  print.cpp -o print.o
gcc sunwait.o sunriset.o print.o sunwait.h sunriset.h print.h -o sunwait -lm -lstdc++
make[1]: Leaving directory '/home/pi/allsky/sunwait-src'
cp sunwait-src/sunwait .
pi@Pi-AllSky:~/allsky $

linuxkidd, 2021-10-01:
I've submitted a PR that should fix this.  Waiting for another dev to review / merge.

Thanks!

maserowik, 2021-10-01:
i am now confused.  i see the comment it being fixed but the log still show No package 'opencv4' found.

linuxkidd, 2021-10-01:
Can you copy/paste the modified `Makefile` contents after your edit?

Thanks!

maserowik, 2021-10-01:
@linuxkidd 

Not sure what I did incorrect the first time but I did it again and BINGO the opencv4 error went away.  
Thanks for the assistance.
 
Want me to close this?

linuxkidd, 2021-10-01:
Thanks @maserowik for the confirmation, great to hear it worked!  I'll reopen for this issue until the PR gets merged -- it'll auto-close then.

closed 2021-10-01 by ckuethe

ckuethe, 2021-10-01:
Pro-tip, if you're testing a diff/PR (especially for build-related fixes you can pull the diff from the PR directly.

So a non-destructive way to test this would be something like:

```
cd /tmp
git clone https://github.com/thomasjacquin/allsky allsky-test
cd allsky-test
wget https://patch-diff.githubusercontent.com/raw/thomasjacquin/allsky/pull/528.diff
patch < 528.diff
make
```

Substitute your favorite temporary location and diff number as appropriate.

PR #528 (2021-10-01): Fixed opencv pkg-config complaints
Resolves #526
commit 78c28d4c (2021-10-01): Fixed `pkg-config` complaints.  Resolves #526