AllskyTeam/allsky #531: can't generate timelapse manually  with generateForDay.sh

aleveilleVE2DTL, 2021-10-02:
Working OK with keogram and startrail.

Went further when creating a directory /sequence owned by pi.
see log below.

First ticket on GitHub! I really want that project works. Very good work everyone.

 pi@allSkyCam10:~/allsky $ ./scripts/generateForDay.sh -k 20211002
===== Generating Keogram
Completed

================
If you want to upload the file(s) you just created, execute 'uploadForDay.sh 20211002
================
pi@allSkyCam10:~/allsky $ ./scripts/generateForDay.sh -s 20211002
===== Generating Startrails, threshold=0.1
Minimum: 0.400927 maximum: 0.420447 mean: 0.410087 median: 0.406201
No images below threshold, writing the minimum image only
Completed

================
If you want to upload the file(s) you just created, execute 'uploadForDay.sh 20211002
================
pi@allSkyCam10:~/allsky $ ./scripts/generateForDay.sh -t 20211002
===== Generating Timelapse
ln: failed to create symbolic link '/sequence/0001.jpg': No such file or directory
ln: failed to create symbolic link '/sequence/0002.jpg': No such file or directory
ln: failed to create symbolic link '/sequence/0003.jpg': No such file or directory
ln: failed to create symbolic link '/sequence/0004.jpg': No such file or directory
ln: failed to create symbolic link '/sequence/0005.jpg': No such file or directory
ln: failed to create symbolic link '/sequence/0006.jpg': No such file or directory
ln: failed to create symbolic link '/sequence/0007.jpg': No such file or directory
ln: failed to create symbolic link '/sequence/0008.jpg': No such file or directory
ln: failed to create symbolic link '/sequence/0009.jpg': No such file or directory
ln: failed to create symbolic link '/sequence/0010.jpg': No such file or directory
ln: failed to create symbolic link '/sequence/0011.jpg': No such file or directory
ln: failed to create symbolic link '/sequence/0012.jpg': No such file or directory
ln: failed to create symbolic link '/sequence/0013.jpg': No such file or directory
timelapse.sh: Processing 13 images...
[image2 @ 0xce8260] Could find no file with path '/home/pi/allsky/images/20211002/sequencce/%04d.jpg' and index in the range 0-4
/home/pi/allsky/images/20211002/sequencce/%04d.jpg: No such file or directory

*** timelapse.sh: ERROR: ffmpeg failed with RET=1
Links in '/home/pi/allsky/images/20211002/sequencce' left for debugging.
Remove them when the problem is fixed.

================
If you want to upload the file(s) you just created, execute 'uploadForDay.sh 20211002
================
pi@allSkyCam10:~/allsky $ sudo mkdir /sequence
pi@allSkyCam10:~/allsky $ sudo chown pi:pi  /sequence
pi@allSkyCam10:~/allsky $ ./scripts/generateForDay.sh -t 20211002
===== Generating Timelapse
timelapse.sh: Processing 13 images...
[image2 @ 0x13a3260] Could find no file with path '/home/pi/allsky/images/20211002/sequencce/%04d.jpg' and index in the range 0-4
/home/pi/allsky/images/20211002/sequencce/%04d.jpg: No such file or directory

*** timelapse.sh: ERROR: ffmpeg failed with RET=1
Links in '/home/pi/allsky/images/20211002/sequencce' left for debugging.
Remove them when the problem is fixed.

================
If you want to upload the file(s) you just created, execute 'uploadForDay.sh 20211002
================
pi@allSkyCam10:~/allsky $

aleveilleVE2DTL, 2021-10-02:
[cantCreateTimelapse.log](https://github.com/thomasjacquin/allsky/files/7271867/cantCreateTimelapse.log)

better with a container...

ckuethe, 2021-10-02:
It looks like there's a typo, which I didn't spot sooner. Edit scripts/timelapse.sh - around line 55 you'll see

`SEQUENCE_DIR="${DATE_DIR}/sequencce"`

Change that to

`SEQUENCE_DIR="${DATE_DIR}/sequence"`

and try it again.

aleveilleVE2DTL, 2021-10-02:
still trying to create directory "sequence" under /    root directory
[log2.log](https://github.com/thomasjacquin/allsky/files/7271968/log2.log)
.

ckuethe, 2021-10-02:
Cool, I'm glad at least the sequence part is fixed. Thanks for the log, that's helpful. I'm curious as to why timelapse is using the wrong directory.

Can you edit scripts/timelapse.sh and add `set -x` on line 2? That'll make the timelapse script run in a tracing mode so we can see exactly what it's doing.

aleveilleVE2DTL, 2021-10-02:
Voila
[log3.log](https://github.com/thomasjacquin/allsky/files/7272006/log3.log)
!

ckuethe, 2021-10-02:
I'm gonna reopen this because I didn't fully fix the issue yet. But progress.

aleveilleVE2DTL, 2021-10-02:
yes, I will be your tester... Tnx all.

ckuethe, 2021-10-02:
I think I see the next problem. After line 70, there are these two lines in an awk script...

```
printf "ln -s %s '$DIR'/sequence/%04d.'$EXTENSION'\n", $0, a;
printf "ln -s %s '$DIR'/sequence/%04d.'$EXTENSION'\n", $0, a >> "'${TMP}'";
```

I think that should read 
```
printf "ln -s %s '${SEQUENCE_DIR}'/sequence/%04d.'$EXTENSION'\n", $0, a;
printf "ln -s %s '${SEQUENCE_DIR}'/sequence/%04d.'$EXTENSION'\n", $0, a >> "'${TMP}'";
```

Let me know if that helps

aleveilleVE2DTL, 2021-10-02:
change it:
    ls -rt "${DATE_DIR}"/*.${EXTENSION} |
        gawk 'BEGIN{ a=1 }
                {
                #       printf "ln -s %s '$DIR'/sequence/%04d.'$EXTENSION'\n", $0, a;
                #       printf "ln -s %s '$DIR'/sequence/%04d.'$EXTENSION'\n", $0, a >> "'${TMP}'";
                                printf "ln -s %s '${SEQUENCE_DIR}'/sequence/%04d.'$EXTENSION'\n", $0, a;
                                printf "ln -s %s '${SEQUENCE_DIR}'/sequence/%04d.'$EXTENSION'\n", $0, a >> "'${TMP}'";
                        a++;
                }' |

and got this trace 
[log4.log](https://github.com/thomasjacquin/allsky/files/7272139/log4.log)

ckuethe, 2021-10-02:
Great. You can remove `/sequence` from both of those printf commands and that should help. I'm not near my allsky machine at the moment, so I can't easily test this first.

EricClaeys, 2021-10-02:
Sorry for introducing these bugs.  I had it working on my Pi but didn't copy the code to Git correctly. 
The $DIR should be $SEQUENCE_DIR.
These lines should be inside the "gawk":
```
printf "ln -s %s '${SEQUENCE_DIR}'/%04d.'${EXTENSION}'\n", $0, a;
printf "ln -s %s '${SEQUENCE_DIR}'/%04d.'${EXTENSION}'\n", $0, a >> "'${TMP}'";
```

aleveilleVE2DTL, 2021-10-02:
changed it:
  ls -rt "${DATE_DIR}"/*.${EXTENSION} |
        gawk 'BEGIN{ a=1 }
                {
                #       printf "ln -s %s '$DIR'/sequence/%04d.'$EXTENSION'\n", $0, a;
                #       printf "ln -s %s '$DIR'/sequence/%04d.'$EXTENSION'\n", $0, a >> "'${TMP}'";
                                printf "ln -s %s '${SEQUENCE_DIR}'/%04d.'$EXTENSION'\n", $0, a;
                                printf "ln -s %s '${SEQUENCE_DIR}'/%04d.'$EXTENSION'\n", $0, a >> "'${TMP}'";
                        a++;
                }' |
        bash

and it is working fine, TNX!

             }'
++ wc -l
+ NUM_FILES=29
+ '[' 29 -eq 0 ']'
+ echo 'timelapse.sh: Processing 29 images...'
timelapse.sh: Processing 29 images...
+ SCALE=
+ TIMELAPSEWIDTH=0
+ TIMELAPSEHEIGHT=0
+ '[' 0 '!=' 0 ']'
+ OUTPUT_FILE=/home/pi/allsky/images/20211002/allsky-20211002.mp4
+ ffmpeg -y -f image2 -loglevel warning -r 25 -i /home/pi/allsky/images/20211002/sequence/%04d.jpg -vcodec libx264 -b:v 2000k -pix_fmt yuv420p -movflags +faststart /home/pi/allsky/images/20211002/allsky-20211002.mp4
[swscaler @ 0x1749cf0] deprecated pixel format used, make sure you did set range correctly
+ RET=0
+ '[' 0 -ne 0 ']'
+ '[' -z '' ']'
+ rm -rf /sequence
+ echo -en 'timelapse.sh: \033[0;32mTimelapse was created\033[0m\n'
timelapse.sh: Timelapse was created
+ exit 0
Completed

================
If you want to upload the file(s) you just created, execute 'uploadForDay.sh 20211002
================
pi@allSkyCam10:~/allsky $

closed 2021-10-02 by ckuethe

ckuethe, 2021-10-03:
You're welcome. Thanks for the useful logs that helped us to troubleshoot.

PR #534 (2021-10-02): timelapse.sh: fix bug in $DIR
Fixes #531
commit 8c7421d9 (2021-10-02): timelapse.sh: fix bug in $DIR

Fixes #531