AllskyTeam/allsky #537: startrails endless looping on Ubuntu server 20.04 Pi 4B

JosWennmacker, 2021-10-03:
Please change line 75 in startrails.cpp from:

75 --> while ((c = getopt(argc, argv, `"hvsb:d:e:o:")) != -1)  {

to    

75 --> while (((c = getopt(argc, argv, "hvsb:d:e:o:")) != -1) && ( c != 255)) {

c was declared char. When a char is used as int it is unsigned on the RPi, so it will never be -1, but 255 in stead.

Sorry for the :o: but I do not know how to escape the strange mark up. Obviously it should be :(no space)o(nospace):

ckuethe, 2021-10-03:
Your source is out of date. I already fixed that by changing the type of `c` to `int`. Let me know if the issue persists after updating. 

You can mark code as code by wrapping it in triple backquotes '```'

JosWennmacker, 2021-10-03:
Thanks. New source installed.

closed 2021-10-03 by linuxkidd