AllskyTeam/allsky #539: New install ASI120

Dentonknifeworks, 2021-10-03:
Not geting Startrails or Killogram

error at start of script
Timelapse works fine.

endOfNight.sh: Generating Startrails

OpenCV Error: Sizes of input arguments do not match (The operation is neither 'array op array' (where arrays have the same size and type), nor 'array op scalar', nor 'scalar op array') in binary_op, file /build/opencv-L65chJ/opencv-3.2.0+dfsg/modules/core/src/arithm.cpp, line 225
terminate called after throwing an instance of 'cv::Exception'
  what():  /build/opencv-L65chJ/opencv-3.2.0+dfsg/modules/core/src/arithm.cpp:225: error: (-209) The operation is neither 'array op array' (where arrays have the same size and type), nor 'array op scalar', nor 'scalar op array' in function binary_op

./endOfNight.sh: line 92: 22458 Aborted                 ${ALLSKY_HOME}/startrails -d $LAST_NIGHT_DIR/ -e $EXTENSION -b $BRIGHTNESS_THRESHOLD -o $OUTPUT
endOfNight.sh: Generating Timelapse

ckuethe, 2021-10-03:
I'll look into this later today but I think that if you specify height and width in your config file then keogram and startrails can filter out the images with the wrong sizes. If not... I'll add that.

ckuethe, 2021-10-04:
I'm guessing you have Notification Images enabled.

Anyway, this will kind of mitigate the issue if you specify your desired image size in your settings file. Then keogram and startrails will reject any image that's not the size produce by your camera.

EricClaeys, 2021-10-04:
@ckuethe Chris, can startrails.cpp and keogram.cpp print a nice error message when the size of the file isn't correct, rather than the cryptic `OpenCV Error `above?  Including the name of the image would be good.  If the size is incorrect or the image corrupt or whatever caused the `OpenCV Error,` can those programs ignore the bad image and continue?

I'm not positive, but I think when those programs are reading a bunch of mono images and find a color one, even if the same size, they barf.  Or reading color and finding mono.  Those cases should also produce a useful error message with the name of the offending file(s).  This can happen when a Notification Image is incorrectly there, or the user switches cameras during the night, from mono to color or vice versa (I know that almost never happens, except with me...).

ckuethe, 2021-10-04:
Nah, that's too hard. ;)

https://github.com/thomasjacquin/allsky/blob/master/src/keogram.cpp#L277
https://github.com/thomasjacquin/allsky/blob/master/src/startrails.cpp#L169
https://github.com/thomasjacquin/allsky/pull/511/files#diff-eae79f3c0d3b9e332f5c84c05542799604adb5541a6ac729772e36eefa5a5852R322

You can get a diagnostic message about incorrectly sized images if you run with `-v` but my assumption was "shut up and make pretty pictures already", so I just silently skip unexpected image sizes.

I didn't add similar diagnostic messaging for unexpected color spaces when I skip those offending images. I could try convert their channels, but they're likely already dropped by the image size check. It's pretty easy to try convert number of channels.

ckuethe, 2021-10-04:
I was experimenting with exception handlers, so maybe I can try/catch opencv errors and print a better message without crashing. In my testing I tend to only find a few bad images a night (if any), this is not noticeable in startrails and just shows up as a 1px black vertical line in the keogram, which I'm OK with because it preserves timing.

Dentonknifeworks, 2021-10-04:
I trying the image size setting in the config. I will report back and yes I have notification images on.
I just checked and there is one of them in the images dir.

EricClaeys, 2021-10-04:
@Dentonknifeworks there's a bug somewhere that allows notification images to sometimes appear in the images/DATE folder.  I <think> I have it fixed on my Pi but haven't uploaded the code yet.
What does the notification image say?  That might help me fix the bug.
Notification images WILL appear in thumbnails to give users an idea what time things happened.

@ckuethe I suggest displaying messages about "bad" images - to me they indicate (possible) errors the user may want to check into.
FYI, my newest saveImage*.sh scripts optionally call removeBadImage.sh after every picture is created rather than do them all in endOfNight.sh, since it only adds a few seconds in saveImage*.sh but ten or more minutes in endOfNight.sh.

EricClaeys, 2021-10-04:
@ckuethe Chris, did the code display errors on non-matching sizes (no `-v` needed) when @Dentonknifeworks reported the error, or did you add that afterwards?  If prior to the error, I assume the `OpenCV Error` message was from non-matching colors?  Or something else?

Dentonknifeworks, 2021-10-04:
@ericClaeys. "Allsky software is not running." is the image. That looks to be the issue. Last night the time-lapse, Kelo and startrails worked with no issues and no status images in the image directory.

ckuethe, 2021-10-04:
@EricClaeys  If you don't pass in a size filter then no checks are done and OpenCV is free to crash. Using the recently updated `endOfNight.sh` you can set your image size which will then allow the notification images to be rejected. There's no value in printing out the fact that some garbage image on disk was skipped, at least not by default. I'm going for "no news" is good news.

I'll add channel correction to keogram and startrails, that's easy. Then I'll make it so `-v` only prints a message if it has to correct or skip a file, and `-vv` if you want to see everything during processing.

EricClaeys, 2021-10-04:
@Dentonknifeworks, please make sure you have the latest scripts/copy_notification_image.sh where line 71 is
`"${ALLSKY_SCRIPTS}/upload.sh" --silent "${IMAGE_TO_USE}" "${IMGDIR}" "${FULL_FILENAME}" "NI"`
If you have `${IMG_DIR} `instead of `${IMGDIR}` then you need to upgrade.

@ckuethe, my style is to always produce useful error messages.  The OpenCV Error messages are totally useless to users.  I think you said you'd look into doing a try/catch to print better messages?  I also prefer to let users know when "bad" input is encountered.  If you make the `-v` / `-vv` change then perhaps a good compromise would be to add a variable in config.sh to allow users to decide the debug level they want, like "show me bad files" (-v) or "show me everything" (-vv).  Or we could use ALLSKY_DEBUG_LEVEL of 4 for `-vv` and a lower number for `-v` (not sure if it should be 1, 2, or 3).

@linuxkidd, your thoughts?

linuxkidd, 2021-10-05:
I agree with trapping errors and providing useful feedback.  If the error message is obtuse, then the ease of correcting the issue is much higher.

I think there's no reason to add `-v` / `-vv`... that's what debug level is for.  Higher number == more `v`s.

ckuethe, 2021-10-05:
Right that's what -v does. If you look at the getopt handler you'll see that it counts the number of -v arguments. From there I decide what to print based on the value of the config.verbose element

ckuethe, 2021-10-05:
Printing a recoverable/recovered error by default is next to useless. There's nothing for the user to do - we've already handled the problem.

EricClaeys, 2021-10-05:
so we don't need another config.sh variable, but can use the value of ALLSKY_DEBUG_LEVEL to determine how many `-v`s to use.  Right now, 4 is the highest ALLSKY_DEBUG_LEVEL and would normally only be used for debugging or testing.  Should startrails.cpp and keogram.cpp use -vvvv to print everything?  How many `-v`s to print info on "bad" images?  2 or 3?

ckuethe, 2021-10-05:
To me:

* 0 `-v` = shut up and make pretty pictures or GTFO
* 1 `-v` = OK, you can tell me about anomalies even if they're recoverable
* 2+ `-v` = full debugging.
I can probably also use `ALLSKY_DEBUG_LEVEL` if it's not set by the arguments.

Dentonknifeworks, 2021-10-05:
@ericclaeys.  I download and installed fresh copy of Allsky on 9*30*2021. My copy_nofification_image.sh contans 
#!/bin/bash

ME="$(basename "$BASH_ARGV0")"	# Include script name in output so it's easier to find in the log file

NOTIFICATIONFILE="$1"	# filename, minus the extension, since the extension may vary
if [ "$1" = "" ] ; then
	echo "*** $ME: ERROR: no file specified" >&2
	exit 1
fi

source $ALLSKY_HOME/config.sh
source $ALLSKY_HOME/scripts/filename.sh
source $ALLSKY_HOME/scripts/ftp-settings.sh

cd $ALLSKY_HOME

NOTIFICATIONFILE="notification_images/$NOTIFICATIONFILE.$EXTENSION"
if [ ! -e "$NOTIFICATIONFILE" ] ; then
	echo "*** $ME: ERROR: File '$NOTIFICATIONFILE' does not exist or is empty!" >&2
	exit 1
fi

IMAGE_TO_USE="$FULL_FILENAME"
cp "$NOTIFICATIONFILE" "$IMAGE_TO_USE"	# don't overwrite notification image

# Resize the image if required
if [ $IMG_RESIZE = "true" ]; then
        convert "$IMAGE_TO_USE" -resize "$IMG_WIDTH"x"$IMG_HEIGHT" "$IMAGE_TO_USE"
	RET=$?
	if [ $RET -ne 0 ] ; then
		echo "*** $ME: ERROR: IMG_RESIZE failed with RET=$RET"
		exit 1
	fi
fi

# IMG_DIR and IMG_PREFIX are in config.sh
# If the user specified an IMG_PREFIX, copy the file to that name so the websites can display it.
if [ "${IMG_PREFIX}" != "" ]; then
	cp "$IMAGE_TO_USE" "${IMG_PREFIX}${FILENAME}.${EXTENSION}"
fi

# If 24 hour saving is desired, save the image in today's thumbnail directory
# so the user can see when things changed.
# Don't save in main image directory because we don't want the notification image in timelapses.
if [ "$CAPTURE_24HR" = "true" ] ; then
	CURRENT=$(date +'%Y%m%d')
	mkdir -p images/$CURRENT
	THUMBNAILS_DIR=images/$CURRENT/thumbnails
	mkdir -p $THUMBNAILS_DIR

	SAVED_FILE="$FILENAME-$(date +'%Y%m%d%H%M%S').$EXTENSION"
	# Create a thumbnail of the image for faster load in web GUI
	convert "$IMAGE_TO_USE" -resize "${THUMBNAIL_SIZE_X}x${THUMBNAIL_SIZE_Y}" "$THUMBNAILS_DIR/$SAVED_FILE"
	RET=$?
	if [ $RET -ne 0 ] ; then
		echo "*** $ME: WARNING: THUMBNAIL resize failed with RET=$RET; continuing."
    	fi
fi

# If upload is true, optionally create a smaller version of the image and upload it
if [ "$UPLOAD_IMG" = "true" ] ; then
	if [ "$RESIZE_UPLOADS" = "true" ]; then
		echo -e "$ME: Resizing $NOTIFICATIONFILE for uploading"
		# Create a smaller version for upload
		convert "$IMAGE_TO_USE" -resize "$RESIZE_UPLOADS_SIZE" -gravity East -chop 2x0 "$IMAGE_TO_USE"
		RET=$?
		if [ $RET -ne 0 ] ; then
			echo "*** $ME: ERROR: RESIZE_UPLOADS failed with RET=$RET"
			exit 1
    		fi
	fi

	TS=$(ls -l --time-style='+%H:%M:%S' $IMAGE_TO_USE | awk '{print $6}')
	echo -e "$ME: Uploading $(basename $NOTIFICATIONFILE) with timestamp: $TS\n"
	if [ $PROTOCOL = "S3" ] ; then
                $AWS_CLI_DIR/aws s3 cp "$IMAGE_TO_USE" s3://$S3_BUCKET$IMGDIR --acl $S3_ACL &
        elif [ $PROTOCOL = "local" ] ; then
		cp "$IMAGE_TO_USE" "$IMGDIR" &
	else
		TEMP_NAME="ni-$RANDOM"
		# "ni" = notification image.  Use unique temporary name.
		lftp "$PROTOCOL://$USER:$PASSWORD@$HOST:$IMGDIR" -e "set net:max-retries 2; set net:timeout 20; put "$IMAGE_TO_USE" -o $TEMP_NAME; rm -f "$IMAGE_TO_USE"; mv $TEMP_NAME "$IMAGE_TO_USE"; bye" &

        fi
fi

Line 71 contains FI.    Am I to understand that I need to upgrade the full install?

Thank
Brad

ckuethe, 2021-10-05:
I just edited your reply to wrap the code in `~~~` like this:

```
~~~
#!/bin/bash
...
~~~
```

Please try to do that for other pastes, otherwise github may incorrectly interpret some of the output as formatting rather than information.

ckuethe, 2021-10-05:
If you don't mind doing a full install, we'd love your help in testing.

But you can quickly patch it by just retrieving the latest version of endofnight with `wget https://raw.githubusercontent.com/thomasjacquin/allsky/master/scripts/endOfNight.sh`

EricClaeys, 2021-10-05:
@Dentonknifeworks, I strongly suggest a full install.  Changes in `endOfNight.sh` require the new `upload.sh` script which is now what calls lftp, so you don't want to get just one file.

Prior to full install, save your ~/allsky/images directory and restore it after the install.
Also, save your configuration files so you can restore the settings - I suggest NOT simply copying the old files back since we change the name of a few variables.  The configuration files you want to save are:

- ~/allsky/config.sh
- ~/allsky/scripts/ftp-settings.sh
- If you are NOT using the GUI, also settings_ZWO.json.
- If you ARE using the GUI, don't worry about settings_ZWO.json.

EricClaeys, 2021-10-05:
@ckuethe, I respectfully disagree that "Printing a recoverable/recovered error by default is next to useless".  There is often an underlying cause for those errors that the user (or we) may want to check into.  If the user can't do anything about them (as is the case with many Windows Event Viewer errors), then I agree they shouldn't be printed.
But I think we have a good middle ground by letting the user decide by setting the debug level.

EricClaeys, 2021-10-07:
@Dentonknifeworks it's probably a good idea to wait for a more stable release.  We are making a lot of changes in a short period of time so we can get to a stable release.
Glad you got things working.