AllskyTeam/allsky #540: endOfNight.sh error line 71

maserowik, 2021-10-03:
Trying to figure out why my uploads are not working.  when i run the endOfNight.sh I am seeing a few errors show up once the keograms are created.  I have attached a screen scrape of the errors.

<pi@Pi-AllSky:~/allsky/scripts $ ./endOfNight.sh
endOfNight.sh: Posting next twilight time to let server know when to resume liveview

postData.sh: Posting Next Twilight Time
***** upload.sh: ERROR:**
FILE_TO_UPLOAD='/home/pi/allsky/tmp/data.json'
REMOTE_DIR='current'
TEMP_NAME='PD-29686'
DESTINATION_FILE='data.json'

**mv: Access failed: 550 Rename current/data.json: No such file or directory (PD-29686)**

Commands used (run via: lftp -f /home/pi/allsky/tmp/lftp_cmds.txt):
open --user "          " --password "         " ftp://         
set net:max-retries 2
set net:timeout 20
rm -f PD-29686
put "/home/pi/allsky/tmp/data.json" -o PD-29686
rm -f **"data.json"**
mv PD-29686 "current/data.json"
bye

endOfNight.sh: ===== Generating Keogram
upload.sh: FTP'ing /home/pi/allsky/images/20211002/keogram/keogram-20211002.jpg to /domains/n3bsq.com/public_html/allsky/keograms//keogram-20211002.jpg
INFO: '/home/pi/allsky/tmp/lftp_cmds.txt' contains the lftp commands used to upload the file.

**./endOfNight.sh: line 71: syntax error near unexpected token `fi'
./endOfNight.sh: line 71: `fi'**
pi@Pi-AllSky:~/allsky/scripts $ pi@Pi-AllSky:~/allsky/scripts $ ./endOfNight.sh
-bash: pi@Pi-AllSky:~/allsky/scripts: No such file or directory
pi@Pi-AllSky:~/allsky/scripts $ endOfNight.sh: Posting next twilight time to let server know when to resume liveview
-bash: endOfNight.sh:: command not found
put "/home/pi/allsky/tmp/data.json" -o PD-29686
rm -f "data.json"
mv PD-29686 "current/data.json"
bye

here are lines 66 - 75
 66                         echo "${ME}: Copying ${UPLOAD_FILE} to ${WEB_KEOGRAM_DIR}"
 67                         cp ${UPLOAD_FILE} "${WEB_KEOGRAM_DIR}"
 68                 fi
 69         fi
 70         echo
 71 fi
 72
 73 # Generate startrails from collected images.

thinking the fi on line 71 is invalid?

Mike

linuxkidd, 2021-10-03:
I've spotted the issue... there's a missing conditional check at the top of that code block... I'll have the fix posted momentarily.

linuxkidd, 2021-10-03:
If you'd like to fix this in your local code and re-test, edit the `endOfNight.sh` file, making line 48 and 49 look like this:

~~~
# Generate keogram from collected images
if [[ ${KEOGRAM} == "true" ]]; then
~~~

The `Generate keogram` is already present.. you'll just add the `if ...` line after it.

Thanks!

PR #544 (2021-10-03): Fixed missing KEOGRAM conditional
resolves #540
closed 2021-10-03 by ckuethe