AllskyTeam/allsky #543: can't find core.hpp

Clutch57, 2021-10-03:
I'm trying to compile allsky on a Raspberry 3 with Raspian version stretch.
The file that it is looking for is in /usr/include/opencv2/core  
What do I need to do?

Thanks,
Clutch

uname -a
Linux piski 4.19.66-v7+ #1253 SMP Thu Aug 15 11:49:46 BST 2019 armv7l GNU/Linux

* Compile allsky software

make: Entering directory '/home/pi/allsky/src'
2021-10-03 11:54:18 Building capture program...
capture.cpp:1:28: fatal error: opencv2/core.hpp: No such file or directory

linuxkidd, 2021-10-03:
Hi @Clutch57

Can you provide the output from the following:
~~~
pkg-config --exists opencv && echo Yes || echo No
pkg-config --cflags opencv
pkg-config --libs opencv
dpkg -l | grep opencv
~~~

Please wrap the console copy/paste output with ~~~ on the line above and below so it formats properly.  (I've edited your initial comment to add this as an example.

Thanks!

Clutch57, 2021-10-03:
~~~
pi@piski:~/allsky $ pkg-config --exists opencv && echo Yes || echo No
Yes

pi@piski:~/allsky $ pkg-config --cflags opencv
-I/usr/include/opencv

pi@piski:~/allsky $ pkg-config --libs opencv
/usr/lib/arm-linux-gnueabihf/libopencv_calib3d.so -lopencv_calib3d /usr/lib/arm-linux-gnueabihf/libopencv_contrib.so -lopencv_contrib /usr/lib/arm-linux-gnueabihf/libopencv_core.so -lopencv_core /usr/lib/arm-linux-gnueabihf/libopencv_features2d.so -lopencv_features2d /usr/lib/arm-linux-gnueabihf/libopencv_flann.so -lopencv_flann /usr/lib/arm-linux-gnueabihf/libopencv_gpu.so -lopencv_gpu /usr/lib/arm-linux-gnueabihf/libopencv_highgui.so -lopencv_highgui /usr/lib/arm-linux-gnueabihf/libopencv_imgproc.so -lopencv_imgproc /usr/lib/arm-linux-gnueabihf/libopencv_legacy.so -lopencv_legacy /usr/lib/arm-linux-gnueabihf/libopencv_ml.so -lopencv_ml /usr/lib/arm-linux-gnueabihf/libopencv_objdetect.so -lopencv_objdetect /usr/lib/arm-linux-gnueabihf/libopencv_ocl.so -lopencv_ocl /usr/lib/arm-linux-gnueabihf/libopencv_photo.so -lopencv_photo /usr/lib/arm-linux-gnueabihf/libopencv_stitching.so -lopencv_stitching /usr/lib/arm-linux-gnueabihf/libopencv_superres.so -lopencv_superres /usr/lib/arm-linux-gnueabihf/libopencv_ts.so -lopencv_ts /usr/lib/arm-linux-gnueabihf/libopencv_video.so -lopencv_video /usr/lib/arm-linux-gnueabihf/libopencv_videostab.so -lopencv_videostab

pi@piski:~/allsky $ dpkg -l | grep opencv
ii  libopencv-calib3d-dev:armhf           2.4.9.1+dfsg1-2                     armhf        development files for libopencv-calib3d
ii  libopencv-calib3d2.4v5:armhf          2.4.9.1+dfsg1-2                     armhf        computer vision Camera Calibration library
ii  libopencv-contrib-dev:armhf           2.4.9.1+dfsg1-2                     armhf        development files for libopencv-contrib
ii  libopencv-contrib2.4v5:armhf          2.4.9.1+dfsg1-2                     armhf        computer vision contrib library
ii  libopencv-core-dev:armhf              2.4.9.1+dfsg1-2                     armhf        development files for libopencv-core
ii  libopencv-core2.4v5:armhf             2.4.9.1+dfsg1-2                     armhf        computer vision core library
ii  libopencv-dev                         2.4.9.1+dfsg1-2                     armhf        development files for opencv
ii  libopencv-features2d-dev:armhf        2.4.9.1+dfsg1-2                     armhf        development files for libopencv-features2d
ii  libopencv-features2d2.4v5:armhf       2.4.9.1+dfsg1-2                     armhf        computer vision Feature Detection and Descriptor Extraction library
ii  libopencv-flann-dev:armhf             2.4.9.1+dfsg1-2                     armhf        development files for libopencv-flann
ii  libopencv-flann2.4v5:armhf            2.4.9.1+dfsg1-2                     armhf        computer vision Clustering and Search in Multi-Dimensional spaces library
ii  libopencv-gpu-dev:armhf               2.4.9.1+dfsg1-2                     armhf        development files for libopencv-gpu2.4v5
ii  libopencv-gpu2.4v5:armhf              2.4.9.1+dfsg1-2                     armhf        computer vision GPU library
ii  libopencv-highgui-dev:armhf           2.4.9.1+dfsg1-2                     armhf        development files for libopencv-highgui
ii  libopencv-highgui2.4-deb0:armhf       2.4.9.1+dfsg1-2                     armhf        computer vision High-level GUI and Media I/O library
ii  libopencv-imgproc-dev:armhf           2.4.9.1+dfsg1-2                     armhf        development files for libopencv-imgproc
ii  libopencv-imgproc2.4v5:armhf          2.4.9.1+dfsg1-2                     armhf        computer vision Image Processing library
ii  libopencv-legacy-dev:armhf            2.4.9.1+dfsg1-2                     armhf        development files for libopencv-legacy
ii  libopencv-legacy2.4v5:armhf           2.4.9.1+dfsg1-2                     armhf        computer vision legacy library
ii  libopencv-ml-dev:armhf                2.4.9.1+dfsg1-2                     armhf        development files for libopencv-ml
ii  libopencv-ml2.4v5:armhf               2.4.9.1+dfsg1-2                     armhf        computer vision Machine Learning library
ii  libopencv-objdetect-dev:armhf         2.4.9.1+dfsg1-2                     armhf        development files for libopencv-objdetect
ii  libopencv-objdetect2.4v5:armhf        2.4.9.1+dfsg1-2                     armhf        computer vision Object Detection library
ii  libopencv-ocl-dev:armhf               2.4.9.1+dfsg1-2                     armhf        development files for libopencv-ocl2.4v5
ii  libopencv-ocl2.4v5:armhf              2.4.9.1+dfsg1-2                     armhf        computer vision OpenCL support library
ii  libopencv-photo-dev:armhf             2.4.9.1+dfsg1-2                     armhf        development files for libopencv-photo2.4v5
ii  libopencv-photo2.4v5:armhf            2.4.9.1+dfsg1-2                     armhf        computer vision computational photography library
ii  libopencv-stitching-dev:armhf         2.4.9.1+dfsg1-2                     armhf        development files for libopencv-stitching2.4v5
ii  libopencv-stitching2.4v5:armhf        2.4.9.1+dfsg1-2                     armhf        computer vision image stitching library
ii  libopencv-superres-dev:armhf          2.4.9.1+dfsg1-2                     armhf        development files for libopencv-superres2.4v5
[... cut ...]
ii  libopencv-videostab2.4v5:armhf        2.4.9.1+dfsg1-2                     armhf        computer vision video stabilization library
ii  libopencv2.4-java                     2.4.9.1+dfsg1-2                     all          Java bindings for the computer vision library
ii  libopencv2.4-jni                      2.4.9.1+dfsg1-2                     armhf        Java jni library for the computer vision library
ii  opencv-data                           2.4.9.1+dfsg1-2                     all          development data for opencv

~~~

linuxkidd, 2021-10-03:
Ah... it appears the OpenCV include structure has changed from the 2.x to 3.x/4.x versions.

Is there something else that prevents you from upgrading to Buster?  ( can be done in-place without re-flashing / installing everything from scratch if you'd like ).

Otherwise, I'll have to look into conditional checks and optional include statements for OpenCV 2.x.

Clutch57, 2021-10-03:
I'll upgrade it to buster and see what happens.  There isn't anything on this RPi that matters to me.

Clutch57, 2021-10-03:
So much for an in place upgrade.

Clutch

~~~
pi@piski:~/allsky $ ./install.sh

**********************************************
*** Welcome to the Allsky Camera installer ***
**********************************************

* Dependencies installation

make: Entering directory '/home/pi/allsky/src'
2021-10-03 13:53:52 Installing build dependencies...
Hit:1 http://raspbian.raspberrypi.org/raspbian buster InRelease
Hit:2 http://archive.raspberrypi.org/debian buster InRelease
Reading package lists... Done                      
Building dependency tree       
Reading state information... Done
714 packages can be upgraded. Run 'apt list --upgradable' to see them.
Reading package lists... Done
Building dependency tree       
Reading state information... Done
imagemagick is already the newest version (8:6.9.10.23+dfsg-2.1+deb10u1).
libusb-1.0-0-dev is already the newest version (2:1.0.22-2).
libusb-dev is already the newest version (2:0.1.12-32).
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 ffmpeg : Depends: libavdevice58 (= 7:4.1.6-1~deb10u1+rpt2) but it is not going to be installed
 libopencv-dev : Depends: libopencv-contrib-dev (= 3.2.0+dfsg-6) but it is not going to be installed
                 Depends: libopencv-viz-dev (= 3.2.0+dfsg-6) but it is not going to be installed
                 Depends: libopencv3.2-java (= 3.2.0+dfsg-6) but it is not going to be installed
                 Depends: libopencv-contrib3.2 but it is not going to be installed
E: Unable to correct problems, you have held broken packages.
Makefile:28: recipe for target 'deps' failed
make: *** [deps] Error 100
make: Leaving directory '/home/pi/allsky/src'
Installing dependencies failed!

~~~

linuxkidd, 2021-10-03:
Hi @Clutch57 
  Sorry, I meant in-place upgrade of the Raspberry Pi OS, not allsky by itself.  Here's an article that shows how to accomplish it:

https://www.makeuseof.com/tag/raspberry-pi-update-raspbian-os/
( Check the section labeled `Update Raspbian to the Latest Version (Stretch to Buster)` )

Let us know how it turns out.

Clutch57, 2021-10-04:
I got it to compile.  It is taking pictures of my ceiling right now.
Thank you,
Clutch

closed 2021-10-04 by Clutch57