AllskyTeam/allsky #554: *** Waiting for you to fix this.  Restart when done fixing. ***

WirthmU, 2021-10-04:
The latest commit gives no pictures on a RPiHQ camera. The log shows the following:

```
Oct  4 16:40:56 allsky allsky.sh[822]:      ***** Starting AllSky *****
Oct  4 16:40:56 allsky allsky.sh[822]: Making sure allsky.sh is not already running...
Oct  4 16:40:56 allsky allsky.sh[822]: Trying to automatically choose between ZWO and RPI camera
Oct  4 16:40:56 allsky allsky.sh[822]: CAMERA: RPiHQ
Oct  4 16:40:56 allsky allsky.sh[822]: CAMERA_SETTINGS: /etc/raspap/settings_RPiHQ.json
Oct  4 16:40:56 allsky allsky.sh[822]: cp: cannot create regular file '/home/pi/allsky/tmp/notification-image-resize.jpg': No such file or directory
Oct  4 16:40:56 allsky allsky.sh[822]: convert-im6.q16: unable to open image `/home/pi/allsky/tmp/notification-image-resize.jpg': No such file or directory @ error/blob.c/OpenBlob/2874.
Oct  4 16:40:56 allsky allsky.sh[822]: convert-im6.q16: no images defined `/home/pi/allsky/tmp/notification-image-resize.jpg' @ error/convert.c/ConvertImageCommand/3258.
Oct  4 16:40:56 allsky allsky.sh[822]: *** copy_notification_image.sh: ERROR: IMG_RESIZE failed
Oct  4 16:40:56 allsky allsky.sh[822]: Starting allsky camera...
Oct  4 16:40:56 allsky allsky.sh[822]: /home/pi/allsky/allsky.sh: line 148: /home/pi/allsky/capture_RPiHQ: No such file or directory
Oct  4 16:40:56 allsky allsky.sh[822]: capture_RPiHQ exited with retcode=127
Oct  4 16:40:56 allsky allsky.sh[822]: cp: cannot create regular file '/home/pi/allsky/tmp/notification-image-resize.jpg': No such file or directory
Oct  4 16:40:56 allsky allsky.sh[822]: convert-im6.q16: unable to open image `/home/pi/allsky/tmp/notification-image-resize.jpg': No such file or directory @ error/blob.c/OpenBlob/2874.
Oct  4 16:40:56 allsky allsky.sh[822]: convert-im6.q16: no images defined `/home/pi/allsky/tmp/notification-image-resize.jpg' @ error/convert.c/ConvertImageCommand/3258.
Oct  4 16:40:56 allsky allsky.sh[822]: *** copy_notification_image.sh: ERROR: IMG_RESIZE failed
Oct  4 16:40:56 allsky allsky.sh[822]: *** Waiting for you to fix this.  Restart when done fixing. ***

Any help to this?
```

linuxkidd, 2021-10-04:
Hi @WirthmU -- thanks for catching this.  There's a pending PR #553 that resolves this.

In the interim, you can get your local version working by logging in as the `pi` user, then run:
`mkdir allsky/tmp`

Thanks, and apologies for the inconvenience.

EricClaeys, 2021-10-04:
@linuxkidd Michael, would it make sense to have variables.sh check if each directory exists, and either make it (e.g., ALLSKY_TMP) or print an error message and exit (e.g., ALLSKY_SCRIPTS)?
Doing this (which is only done once per invocation of allsky.sh) would catch and possibly errors like the one above.
Comments?

linuxkidd, 2021-10-04:
Ya.. I had strongly considered `variables.sh` checking that the paths exists as a potential fix.

It may be better to check inside `allsky.sh` instead... since `variables.sh` (and all the other config files) get referenced extensively by `scripts/*.sh`.

EricClaeys, 2021-10-04:
@linuxkidd Michael, variables.sh is source'd into all the scripts, but it only executes its code once per invocation of allsky.sh via this code:
```
if [ "${ALLSKY_VARIABLE_SET}" = "" ]; then
	set -a	# automatically export all variables

	ALLSKY_VARIABLE_SET="true"	# so we only do the following once

       # other commands ...
fi
```

So if we add checks for all the directories it won't add execution time except in allsky.sh, which is when we want to do the checks anyhow.

@ckuethe Your input?

linuxkidd, 2021-10-04:
I could be wrong.. but I think that each instantiation of a script ( as executed from `capture` for example ) is a new, separate bash instance.  I don't think the environment variables that were present when `capture` started are used.

-- Actually.. I know there not used because I've had errors in my `variables.sh` or `config.sh`, and once corrected, the scripts returned to normal operation without restarting `allsky` systemd service.

So, it will prevent `variables.sh` from running twice within the same bash shell, but not across script runs from within `capture` or `captures_RPiHQ`.

EricClaeys, 2021-10-04:
@linuxkidd Michael, my intent was for variables.sh to execute the commands when allsky.sh first source'd it in, and all the variables (including ALLSKY_VARIABLES_SET) would automatically be exported via the `set -a`.  The capture commands would hence get those variables and pass them to the scripts they call, so when those scripts ran, variables.sh in them would not re-execute the commands.
It should be an easy test - put an "echo $0" in the code of variables.sh and restart allsky and look in the log file after the first image is processed.  Hopefully there will only be one echo statement.  If there are two, something didn't pass ALLSKY_VARIABLES_SET to its child.
I'm 99% sure there's a way to make this work - it may need something in the capture programs to force them to pass their environment to their children.
We could also try a printf of ALLSKY_VARIABLES_SET from the capture program using the function getenv() that will get the value.  Per the system() man page, the callers environment should be passed to the child.
It's also possible there's a bug in variables.sh...

linuxkidd, 2021-10-04:
Maybe --- the issue is that variables.sh isn't called -before- `capture` starts in `allsky.sh`.  So those Environment variables aren't available to pass to the child processes.

linuxkidd, 2021-10-04:
Haha.. indeed it is the issue.

- I modified my `variables.sh` to output some text.
- Monitored logs, and saw that output text each time an image capture completed.
- I modified `allsky.sh` to source `variables.sh` right before attempting to start the capture* program
- Restarted `allsky` via systemd
- Only saw my `variables.sh` script text 1 time.
 
I'll submit this as a PR and that should get us down to 1 time run through of `variables.sh`... then we can discuss at auto-creating directories if they're missing, or just warning about it.

EricClaeys, 2021-10-04:
Crap.  Another case where I had it working in my test version of Allsky but didn't copy the `source variables.sh` when I updated allsky.

closed 2021-10-05 by linuxkidd