AllskyTeam/allsky #558: Install script fail

Telfish2, 2021-10-04:
I am trying to install and when trying to run the install script in terminal is causes an error "This script must not be run as root".

Any ideas how to overcome this?

maserowik, 2021-10-04:
drop the sudo just ./install

linuxkidd, 2021-10-04:
To expand a bit on what @maserowik mentioned:

1. Ensure you're logged in as `pi`
2. Run `git clone https://github.com/thomasjacquin/allsky.git`
3. Change into the 'allsky' directory: `cd allsky`
4. Run the install script:  `./install.sh`

If you've already performed the clone as root and don't want to re-'clone' the repository, you can correct the ownership with:
`sudo chown -R pi:pi ~pi/allsky`

Let us know if you have any other issues getting it working. :)

Telfish2, 2021-10-05:
Thanks I managed to do the install but am getting this error on trying to
start the service. I am running the Rpi HQ camera. I have created the
capture_RPiHQ directory but still get the error.

Thanks for the help

Terry

● allsky.service - All Sky Camera
   Loaded: loaded (/etc/systemd/system/allsky.service; enabled; vendor
preset: enabled)
   Active: inactive (dead) since Mon 2021-10-04 21:07:12 EDT; 15s ago
  Process: 1617 ExecStart=/home/pi/allsky/allsky.sh (code=killed,
signal=TERM)
 Main PID: 1617 (code=killed, signal=TERM)

Oct 04 21:07:05 allsky allsky.sh[1617]: /home/pi/allsky/allsky.sh: line
148: /home/pi/allsky/capture_RPiHQ: No such file or directory
Oct 04 21:07:05 allsky allsky.sh[1617]: capture_RPiHQ exited with
retcode=127
Oct 04 21:07:05 allsky allsky.sh[1617]: *** Waiting for you to fix this.
Restart when done fixing. ***
Oct 04 21:07:05 allsky sudo[1691]:       pi : TTY=unknown ;
PWD=/home/pi/allsky ; USER=root ; COMMAND=/usr/sbin/service allsky stop
Oct 04 21:07:05 allsky sudo[1691]: pam_unix(sudo:session): session opened
for user root by (uid=0)
Oct 04 21:07:12 allsky systemd[1]: Stopping All Sky Camera...
Oct 04 21:07:12 allsky sudo[1691]: pam_unix(sudo:session): session closed
for user root
Oct 04 21:07:12 allsky systemd[1]: allsky.service: Main process exited,
code=killed, status=15/TERM
Oct 04 21:07:12 allsky systemd[1]: allsky.service: Succeeded.
Oct 04 21:07:12 allsky systemd[1]: Stopped All Sky Camera.
~

linuxkidd, 2021-10-05:
My apologies, @Telfish2  ... it seems I missed copying the RPiHQ capture program out of the src folder.  I'm submitting a fix for it now, but to get your allsky capturing, you can run:
~~~
cp ~pi/allsky/src/capture_RPiHQ ~pi/allsky/
~~~

Thanks, and sorry for the miss

closed 2021-10-05 by ckuethe

Telfish2, 2021-10-05:
> Closed #558 <https://github.com/thomasjacquin/allsky/issues/558> via #564

Got it, thanks!

Telfish2, 2021-10-05:
Sorry to be a pain, copied the file across and on running I now get this
error.

● allsky.service - All Sky Camera
   Loaded: loaded (/etc/systemd/system/allsky.service; enabled; vendor
preset: enabled)
   Active: activating (auto-restart) (Result: exit-code) since Tue
2021-10-05 08:27:58 EDT; 2s ago
  Process: 5418 ExecStart=/home/pi/allsky/allsky.sh (code=exited,
status=1/FAILURE)
 Main PID: 5418 (code=exited, status=1/FAILURE)
~

Any ideas?

Telfish2, 2021-10-05:
All working now thanks. deleted and reinstalled.

PR #564 (2021-10-05): Copy capture_RPiHQ to $ALLSKY_HOME
resolves #558