AllskyTeam/allsky #559: wlan0 instead of eth0

maserowik, 2021-10-04:
@EricClaeys 
If I recall correctly you pointed me to the place to modify the ethernet dashboard.  I know is is purely cosmetic but is there a way the default can have this changed in the repo?

<
cd /var/www/html/includes
sudo nano -l dashboard_eth0.php
line 65 change wlan0 to eth0
>

linuxkidd, 2021-10-04:
Hi @maserowik ... the file you're looking for is:

`/var/www/htm/includes/dashboard_eth0.php`

You'll see it around [line number 65](https://github.com/thomasjacquin/allsky-portal/blob/master/includes/dashboard_eth0.php#L65)

Just change `wlan0` on that line to `eth0`.

linuxkidd, 2021-10-04:
Hi @maserowik  -- I submitted a PR to fix this, and it was merged.  Thanks for reminding us about that one! :+1: 

I'll go ahead and close the ticket.

closed 2021-10-04 by linuxkidd

EricClaeys, 2021-10-04:
@maserowik I could have sworn I fixed this after you first pointed it out.  Maybe I was lazy and just told you were to fix it.  Either way, thanks to @linuxkidd for actually fixing it.

@linuxkidd The LAN and WLAN pages on the GUI display the information correctly but (for me and another user) the "start" and "stop" buttons don't work.  If I manually execute `sudo ifdown eth0` or `sudo ifup eth0` (or the same for wlan0 they don't work either.  Do they work on your Pi?

linuxkidd, 2021-10-04:
Hmm... I'll have to get a different test setup going to play with that...  I've opened [a new issue against allsky-portal](https://github.com/thomasjacquin/allsky-portal/issues/34) -- I'll investigate/update there.