AllskyTeam/allsky #560: Install script failed?

robmorin, 2021-10-04:
I installed this last week,  it worked fine, I started over on a new install today , not working

Ideas?

pi@raspberrypi:~/allsky$ ./install.sh 

**********************************************
*** Welcome to the Allsky Camera installer ***
**********************************************

* Dependencies installation

make: Entering directory '/home/pi/allsky/src'
Makefile:42: *** Did not find any OpenCV Libraries, try 'make deps'.  Stop.
make: Leaving directory '/home/pi/allsky/src'
Installing dependencies failed!

Thanks

robmorin, 2021-10-04:
For anyone else having same issue,  installing libopencv-dev did the trick

apt install libopencv-dev
:)

robmorin, 2021-10-04:
Resolved after install libopencv-dev

closed 2021-10-04 by robmorin

linuxkidd, 2021-10-05:
I realize you've got this all taken care of now.. but I'm curious the flow you took that landed you into these errors ( and those from your other ticket(s)).

Did you run the `install.sh` script in the top level directory, as your regular user?  ( presumably the `pi` user ).
If so, can you provide the console output ( if it's still around ) from that run...

I ask, because the installer should have added those dependencies for you ( libopencv-dev ) and handled all of the permissions, etc.  So if that all failed, we'd like to know and fix it for future users.

adymac101, 2021-10-06:
I've run into the exact same problem just today. Installed a new OS onto Pi 4 (Release date: May 7th 2021, Kernel version: 5.10). Followed the instructions to the letter for the allsky install while logged on as the pi user, received an error that 'package libusb-1.0 was not found' during the dependencies check and the installation aborts. Ran 'sudo apt-get install libusb-1.0' and then attempted install.sh again. This time got the error 'did not find any opencv libraries'. Did an install libopencv-dev and tried again and everything worked. Hope that helps others.

Curtis2012, 2021-10-06:
I just had the same problem. My first install of allskycam went smoothly following the published instructions. Then the SD card got corrupted. No big deal...just burn another one and re-install right...wrong. With a freshly burned Raspian 32 bit image on a 3A+ the install failed. Resolved by "install libopencv-dev".

Output from install.sh below:

...
pi@allsky:~/allsky $ ./install.sh

**********************************************
*** Welcome to the Allsky Camera installer ***
**********************************************

* Dependencies installation

make: Entering directory '/home/pi/allsky/src'
Makefile:42: *** Did not find any OpenCV Libraries, try 'make deps'.  Stop.
make: Leaving directory '/home/pi/allsky/src'
Installing dependencies failed!

...

linuxkidd, 2021-10-07:
I have a reproducer for this now.. and am working on it.