AllskyTeam/allsky #567: Files appear in the root of my hosting site

maserowik, 2021-10-05:
Ok I though this was fixed but today the file appeared in the root of my hosting site.

<Oct  5 06:52:30 Pi-AllSky allsky.sh[2314]: === Starting daytime capture ===
Oct  5 06:52:30 Pi-AllSky allsky.sh[2314]: ==========
Oct  5 06:52:30 Pi-AllSky allsky.sh[2314]: Saving images with delay of 150000 ms (150 sec)
Oct  5 06:52:30 Pi-AllSky allsky.sh[2314]: Turning off daytime auto-exposure to use histogram exposure.
Oct  5 06:52:30 Pi-AllSky allsky.sh[2314]: Using last night exposure of 929898 us (929.90 ms)
Oct  5 06:52:30 Pi-AllSky allsky.sh[2314]:   xxxx Adjusting DAY gain by 125 on next picture to 125; will be gain change # 1 of 6.
Oct  5 06:52:30 Pi-AllSky allsky.sh[2314]: STARTING EXPOSURE at: 2021-10-05 06:52:30
Oct  5 06:52:30 Pi-AllSky allsky.sh[2314]:   > Exposure set to 929898 us (929.90 ms), timeout: 6859 ms
Oct  5 06:52:30 Pi-AllSky allsky.sh[2314]: endOfNight.sh: Posting next twilight time to let server know when to resume liveview
Oct  5 06:52:30 Pi-AllSky allsky.sh[2314]: postData.sh: Posting Next Twilight Time
Oct  5 06:52:31 Pi-AllSky allsky.sh[2314]:   > Got image @ exposure: 929898 us (929.90 ms)
Oct  5 06:52:31 Pi-AllSky allsky.sh[2314]: *** upload.sh: ERROR:
Oct  5 06:52:31 Pi-AllSky allsky.sh[2314]: FILE_TO_UPLOAD='/home/pi/allsky/tmp/data.json'
Oct  5 06:52:31 Pi-AllSky allsky.sh[2314]: REMOTE_DIR='current'
Oct  5 06:52:31 Pi-AllSky allsky.sh[2314]: TEMP_NAME='PD-27021'
Oct  5 06:52:31 Pi-AllSky allsky.sh[2314]: DESTINATION_FILE='data.json'
Oct  5 06:52:31 Pi-AllSky allsky.sh[2314]: mv: Access failed: 550 Rename current/data.json: No such file or directory (PD-27021)
Oct  5 06:52:31 Pi-AllSky allsky.sh[2314]: Commands used (run via: lftp -f /home/pi/allsky/tmp/lftp_cmds.txt):
Oct  5 06:52:31 Pi-AllSky allsky.sh[2314]: open --user "u919658081" --password "C@llN3bsq" ftp://n3bsq.com
Oct  5 06:52:31 Pi-AllSky allsky.sh[2314]: set net:max-retries 2
Oct  5 06:52:31 Pi-AllSky allsky.sh[2314]: set net:timeout 20
Oct  5 06:52:31 Pi-AllSky allsky.sh[2314]: rm -f PD-27021
Oct  5 06:52:31 Pi-AllSky allsky.sh[2314]: put "/home/pi/allsky/tmp/data.json" -o PD-27021
Oct  5 06:52:31 Pi-AllSky allsky.sh[2314]: rm -f "data.json"
Oct  5 06:52:31 Pi-AllSky allsky.sh[2314]: mv PD-27021 "current/data.json"
Oct  5 06:52:31 Pi-AllSky allsky.sh[2314]: bye

here is the file on the server
[image]

he is what is located in the file
[image]

I believe this file should be in the allsky or allsky/image folder
[image]
notice the json.data is not updated.

Thanks
Mike

linuxkidd, 2021-10-05:
@maserowik  While I've removed it from your comment now,  you should change your password on that server immediately ( and anywhere else you use that same password ).   There are bots that scour the internet and GitHub specifically for accidental credential leaks.

maserowik, 2021-10-05:
Wow, I thought I scrubbed it.  
thanks for the catch.

EricClaeys, 2021-10-06:
@maserowik I found and fixed that bug a day ago.  You can get the latest scripts/postData.sh file, or on the last line, change the `IMG_DIR `to `IMGDIR`.
Sorry about that.

maserowik, 2021-10-06:
@EricClaeys 
I will test as soon as i can get my allsky back on line.  I mucked it up good!!!!!!

Mike

maserowik, 2021-10-09:
issue went away

closed 2021-10-09 by maserowik