AllskyTeam/allsky #568: no notification pictures showed

Chr1sw1e, 2021-10-05:
Hi,

https://github.com/thomasjacquin/allsky/blob/45f9fe8a5b10570af6057e12331544bbd2525267/scripts/copy_notification_image.sh#L51

Maybe wrong path ? It moves the notification image to image.jpg but in liveview only liveview-image.jpg is displayed

i tried mv -f "${IMAGE_TO_USE}" "${ALLSKY_HOME}/${IMG_PREFIX}${FULL_FILENAME}" now it worked

Is it right or do i have a wrong configuration ?

EricClaeys, 2021-10-06:
@Chr1sw1e, thanks for pointing out the bug.  Please try removing the ${IMG_PREFIX} you added and add this line above the `mv -f `line:
`FULL_FILENAME="${IMG_PREFIX}${FULL_FILENAME}"`
There are other lines in the file that I believe also need the `${IMG_PREFIX}` added, and the line above handles all of them.

Let me know how it goes.

EricClaeys, 2021-10-06:
Excellent!  There's a new version of copy_notification_image.sh with the fix.  You can download it but don't need to.

closed 2021-10-06 by EricClaeys