AllskyTeam/allsky #570: Image upload error + no keogram / timelapse / star trails produced today only

Jonk2, 2021-10-05:
I've tried to keep up with the many new releases / files and this could be a case of delete it all and start again!

To avoid confusion, I have 2 x RPI4bs and 2 x ASi178MCs (outdoors is standard, indoors test one is the cooled version).

Outside known as Allsky PI, inside test one Allsky PI 2.

My log file on the inside test Allsky PI 2 says it's v0.8.1 and for the past couple of days, the image upload (i.e. 'live-view' image) to my NAS fails.

Here's the log from the latest attempt (personal details obscured with ******):

`Oct  5 20:54:56 allsky allsky.sh[3453]: STARTING EXPOSURE at: 2021-10-05 20:54:56
Oct  5 20:54:56 allsky allsky.sh[3453]:   > Extra Text File Age Disabled So Displaying Anyway
Oct  5 20:54:56 allsky allsky.sh[3453]:   > Saving image 'image.jpg' that started at 2021-10-05 20:54:56  (0 us)
Oct  5 20:54:56 allsky allsky.sh[3453]:   > Sleeping: 29988 ms
Oct  5 20:54:58 allsky allsky.sh[3453]: *** upload.sh: ERROR:
Oct  5 20:54:58 allsky allsky.sh[3453]: FILE_TO_UPLOAD='image.jpg'
Oct  5 20:54:58 allsky allsky.sh[3453]: REMOTE_DIR='/Allsky/Allsky PI 2/images'
Oct  5 20:54:58 allsky allsky.sh[3453]: TEMP_NAME='SI-26766'
Oct  5 20:54:58 allsky allsky.sh[3453]: DESTINATION_FILE='image.jpg'
Oct  5 20:54:58 allsky allsky.sh[3453]: put: image.jpg: Access failed: 550 SI-26766: Forbidden filename
Oct  5 20:54:58 allsky allsky.sh[3453]: mv: Access failed: 550 SI-26766: No such file or directory
Oct  5 20:54:58 allsky allsky.sh[3453]: Commands used (run via: lftp -f /home/pi/allsky/tmp/lftp_cmds.txt):
Oct  5 20:54:58 allsky allsky.sh[3453]: open --user "*******" --password "******" ftp://************
Oct  5 20:54:58 allsky allsky.sh[3453]: set net:max-retries 2
Oct  5 20:54:58 allsky allsky.sh[3453]: set net:timeout 20
Oct  5 20:54:58 allsky allsky.sh[3453]: rm -f SI-26766
Oct  5 20:54:58 allsky allsky.sh[3453]: put "image.jpg" -o SI-26766
Oct  5 20:54:58 allsky allsky.sh[3453]: rm -f "image.jpg"
Oct  5 20:54:58 allsky allsky.sh[3453]: mv SI-26766 "/Allsky/Allsky PI 2/images/image.jpg"
Oct  5 20:54:58 allsky allsky.sh[3453]: bye`

This is on my test PI indoors. There doesn't appear to be any 'saveImageNight.sh' entries. The FTP settings are correct and match the PI outdoors which is also running v0.8.1 (from the log), albeit a different folder (Allsky, not Allsky 2). 

The PI outdoors seems to be working fine, and I've had no issue with timelapse, startrails and keogram generation on the PI outdoors, albeit cloudy and wet ones!

`Oct  5 21:02:53 allsky allsky.sh[828]: STARTING EXPOSURE at: 2021-10-05 21:02:53
Oct  5 21:02:54 allsky allsky.sh[828]: saveImageNight.sh: Uploading 'image.jpg' with timestamp: 21:02:53
Oct  5 21:03:23 allsky allsky.sh[828]:   > Saving image 'image.jpg' that started at 2021-10-05 21:02:53  (0 us)
Oct  5 21:03:23 allsky allsky.sh[828]:   > Sleeping from auto exposure: 0 ms (0 sec)
Oct  5 21:03:23 allsky allsky.sh[828]: STARTING EXPOSURE at: 2021-10-05 21:03:23
Oct  5 21:03:24 allsky allsky.sh[828]: saveImageNight.sh: Uploading 'image.jpg' with timestamp: 21:03:24
Oct  5 21:03:54 allsky allsky.sh[828]:   > Saving image 'image.jpg' that started at 2021-10-05 21:03:23  (0 us)
Oct  5 21:03:54 allsky allsky.sh[828]:   > Sleeping from auto exposure: 0 ms (0 sec)
Oct  5 21:03:54 allsky allsky.sh[828]: STARTING EXPOSURE at: 2021-10-05 21:03:54
Oct  5 21:03:55 allsky allsky.sh[828]: saveImageNight.sh: Uploading 'image.jpg' with timestamp: 21:03:54
Oct  5 21:04:24 allsky allsky.sh[828]:   > Saving image 'image.jpg' that started at 2021-10-05 21:03:54  (0 us)
Oct  5 21:04:24 allsky allsky.sh[828]:   > Sleeping from auto exposure: 0 ms (0 sec)
Oct  5 21:04:24 allsky allsky.sh[828]: STARTING EXPOSURE at: 2021-10-05 21:04:24
Oct  5 21:04:25 allsky allsky.sh[828]: saveImageNight.sh: Uploading 'image.jpg' with timestamp: 21:04:24`

All of the files are saving fin on the PI in home/pi//allsky/inages, and thre are keograms etc since the last update (3rd of October) but this morning's (5th of October) failed with the following in the log:

Keogram

`Oct  5 07:15:37 allsky allsky.sh[3564]: endOfNight.sh: ===== Generating Keogram
Oct  5 07:15:42 allsky allsky.sh[3564]:   > Extra Text File Age Disabled So Displaying Anyway
Oct  5 07:15:42 allsky allsky.sh[3564]:   > Saving image 'image.jpg' that started at 2021-10-05 07:15:36  (0 us)
Oct  5 07:15:42 allsky allsky.sh[3564]:   > Sleeping from histogram exposure: 30000 ms
Oct  5 07:15:43 allsky allsky.sh[3564]: *** upload.sh: ERROR:
Oct  5 07:15:43 allsky allsky.sh[3564]: FILE_TO_UPLOAD='image.jpg'
Oct  5 07:15:43 allsky allsky.sh[3564]: REMOTE_DIR='/Allsky/Allsky PI 2/images'
Oct  5 07:15:43 allsky allsky.sh[3564]: TEMP_NAME='SI-16085'
Oct  5 07:15:43 allsky allsky.sh[3564]: DESTINATION_FILE='image.jpg'
Oct  5 07:15:43 allsky allsky.sh[3564]: put: image.jpg: Access failed: 550 SI-16085: Forbidden filename
Oct  5 07:15:43 allsky allsky.sh[3564]: mv: Access failed: 550 SI-16085: No such file or directory
Oct  5 07:15:43 allsky allsky.sh[3564]: Commands used (run via: lftp -f /home/pi/allsky/tmp/lftp_cmds.txt):
Oct  5 07:15:43 allsky allsky.sh[3564]: open --user "******" --password "******" ftp://******
Oct  5 07:15:43 allsky allsky.sh[3564]: set net:max-retries 2
Oct  5 07:15:43 allsky allsky.sh[3564]: set net:timeout 20
Oct  5 07:15:43 allsky allsky.sh[3564]: rm -f SI-16085
[... 30 lines cut ...]
Oct  5 07:16:11 allsky allsky.sh[3564]: [swscaler @ 0x791c50] deprecated pixel format used, make sure you did set range correctly
|
|
|
Oct  5 07:30:13 allsky allsky.sh[3564]: timelapse.sh: Timelapse was created
Oct  5 07:30:14 allsky allsky.sh[3564]: *** upload.sh: ERROR:
Oct  5 07:30:14 allsky allsky.sh[3564]: FILE_TO_UPLOAD='/home/pi/allsky/images/20211004/allsky-20211004.mp4'
Oct  5 07:30:14 allsky allsky.sh[3564]: REMOTE_DIR=''
Oct  5 07:30:14 allsky allsky.sh[3564]: TEMP_NAME='TL-26567'
Oct  5 07:30:14 allsky allsky.sh[3564]: DESTINATION_FILE='allsky-20211004.mp4'
Oct  5 07:30:14 allsky allsky.sh[3564]: put: /home/pi/allsky/images/20211004/allsky-20211004.mp4: Access failed: 550 TL-26567: Forbidden filename
Oct  5 07:30:14 allsky allsky.sh[3564]: mv: Access failed: 550 TL-26567: No such file or directory
Oct  5 07:30:14 allsky allsky.sh[3564]: Commands used (run via: lftp -f /home/pi/allsky/tmp/lftp_cmds.txt):
Oct  5 07:30:14 allsky allsky.sh[3564]: open --user "******" --password "******" ftp://******
Oct  5 07:30:14 allsky allsky.sh[3564]: set net:max-retries 2
Oct  5 07:30:14 allsky allsky.sh[3564]: set net:timeout 20
Oct  5 07:30:14 allsky allsky.sh[3564]: rm -f TL-26567
Oct  5 07:30:14 allsky allsky.sh[3564]: put "/home/pi/allsky/images/20211004/allsky-20211004.mp4" -o TL-26567
Oct  5 07:30:14 allsky allsky.sh[3564]: rm -f "allsky-20211004.mp4"
Oct  5 07:30:14 allsky allsky.sh[3564]: mv TL-26567 "allsky-20211004.mp4"
Oct  5 07:30:14 allsky allsky.sh[3564]: bye`

With my limited understanding of this, the common denominator appears to be the NAS. I don't think it's a NAS access issue, but I suppose I should check that too. Nothing else has an issue accessing the location. It'd the 'Access failed: 550 TL-26567: Forbidden filename' that says it's not an access issue, but then also says 'mv: Access failed: 550 TL-26567: No such file or directory' which would suggest it can't see the location or has been refused.

If this is totally weird and I need to delete allsky and start again with a fresh clone, that's fine, but it might be more sinister. As I said, it could be a case of a botched installation i.e. mix / match of versions, but as both versions are shown as 0.8.1 in the logs, I have to assume they're the same and I haven't done something silly.

Jonk2, 2021-10-05:
Ok, is it a simple thing to do, as the instructions say to delete the allsky folder and clone a new one? What I do is rename the allsky folder, clone a new one and installed as fresh, manually putting my settings back in. Maybe that's not a good thing to do.

The NAS is accessible for both Allsky PI and Allsky PI 2 from my PCs, phones etc no issue. I do have mfa set up on all my NASes but that doesn't affect file access, only web admin stuff. There's nothing in the NAS access logs showing a connection refusal.

It was since I did this on the 3rd of October (rename and clone / install the new version), the latest image hasn't updated on the NAS since, none of the timelapses etc have been uploaded either but they have been created and are stored on the PI. Only this morning did they fail.

I'm thinking that I should clean up everything allsky related and start again, this is why I have a test PI! Unless there's an easy way to overwrite the files from the latest on Github, which I'm not sure how to do?

Jonk2, 2021-10-05:
You may be right...?

[image]

ckuethe, 2021-10-05:
> You may be right...?
> 
> [image]

Clearly you've changed something, as we all have.
* sunwait-src changes because sunwait was compiled there
* image.jpg changes because that's the captured image (we should probably gitignore that)
* and you're probably testing things in allsky.sh

ckuethe, 2021-10-05:
> Ok I'm learning...
> 
> I did a 'git pull' and it's showing me this:
> 
> [image]
> 
> Obviously something's changed but I'm not sure how to interpret these results.
> 
> I'll do some more reading / testing, but please feel free to close this 'issue' if noone else needs to contribute.
> 
> I'll probably end up deleting and starting afresh, to avoid wasting anyone's time. Unless I break something!

Yep lots of changes there. You can do `git log` to see what commits you just pulled, if you're curious.

You'll want to rebuild the the binaries - `cd src && make` - but I don't think you'll need to completely rip everything out

linuxkidd, 2021-10-06:
Actually, now that you're fully up to date with the latest code... you can run the following from the `allsky` folder:
~~~
make clean all
sudo make install
~~~

You'll need to transfer your settings from `config.sh` and `ftp-settings.sh` info the same named files inside `allsky/config` once that is done.  Once those are populated, you can backup / remove the same named files from the `allsky` directory.

linuxkidd, 2021-10-06:
Oh, and then restart allsky:
~~~
sudo systemctl restart allsky
~~~

EricClaeys, 2021-10-06:
@Jonk2 Can you do me a favor? I don't have access to my Pi but want to test something.

In scripts/upload.sh, after the echo set net:timeout 20 line, add this:
echo "echo ls xxxx || echo xxxx does not exist" on one line and
echo "echo test 3 && echo test 4" on another.
Then once the next image is uploaded, look in ~/allsky/tmp/upload_log.txt. If it has "xxxx" in it, copy the file to your reply.

You can then remove those two lines.

I appreciate it.  This will help me determine how to exit after the first error.

Jonk2, 2021-10-06:
Also, sorry but installing the latest (as of 30 minutes ago) my log shows this:

> 
> Oct  6 20:27:54 allsky allsky.sh[5860]: #033[32m ******************************************
> Oct  6 20:27:54 allsky allsky.sh[5860]: #033[32m *** Allsky Camera Software v0.8.1 | 2021 ***
> Oct  6 20:27:54 allsky allsky.sh[5860]: #033[32m ******************************************

Is this correct? How can there be so many changes over the past few days but the version has remained the same? Can you add a build number or something because to idiots like me, I don't know whether I'm up to date or what version I have, and this makes it very difficult to sort out issues before asking for help.

Jonk2, 2021-10-06:
No need to apologise, it must be frustrating trying to solve everyone's issues, we'll all have different problems and I suspect not many of us are programmers!

I'll do as you suggest and report back. Thanks.

Jonk2, 2021-10-06:
Ok here's the terminal result... does it look correct?

> pi@allsky:~ $ cd allsky/gui
> pi@allsky:~/allsky/gui $ sudo ./install.sh
> sudo: unable to resolve host allsky: Name or service not known
> 
> ****************************************************************
> *** Welcome to the Allsky Administration Portal installation ***
> ****************************************************************
> 
> * Installation of the webserver
> 
> Hit:1 http://giteduberger.fr rpimonitor/ InRelease
> Hit:2 http://raspbian.raspberrypi.org/raspbian buster InRelease                 
> Hit:3 http://archive.raspberrypi.org/debian buster InRelease                    
> Reading package lists... Done
> Reading package lists... Done
> Building dependency tree       
> Reading state information... Done
> avahi-daemon is already the newest version (0.7-4+deb10u1).
> hostapd is already the newest version (2:2.7+git20190128+0c1e29f-6+deb10u3).
> lighttpd is already the newest version (1.4.53-4+deb10u1).
> php-cgi is already the newest version (2:7.3+69).
> php-gd is already the newest version (2:7.3+69).
> dnsmasq is already the newest version (2.80-1+rpt1+deb10u1).
> The following packages were automatically installed and are no longer required:
>   gir1.2-gmenu-3.0 gnome-menus libgdata-common libgdata22 libgnome-menu-3-0
>   libgoa-1.0-0b libgoa-1.0-common libimobiledevice6 libjansson4 libldb1
>   liboauth0 libplist3 librtimulib-dev librtimulib-utils librtimulib7 libtevent0
>   libusbmuxd4 python3-rtimulib python3-sense-hat
> Use 'sudo apt autoremove' to remove them.
> 0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
> already enabled
> Run "service lighttpd force-reload" to enable changes
> 
> * Configuring lighttpd
> 
> * Changing hostname to allsky
> 
> * Setting avahi-daemon configuration
> 
> * Adding the right permissions to the web server
> 
> * Retrieving github files to build admin portal
> Cloning into '/var/www/html'...
> remote: Enumerating objects: 2052, done.
> remote: Counting objects: 100% (260/260), done.
> remote: Compressing objects: 100% (176/176), done.
> remote: Total 2052 (delta 145), reused 164 (delta 83), pack-reused 1792
> Receiving objects: 100% (2052/2052), 3.30 MiB | 9.92 MiB/s, done.
> Resolving deltas: 100% (773/773), done.
> 
> * Modify config.sh
> 
> 
> The Allsky Portal is now installed
> You can now reboot the Raspberry Pi and connect to it from your laptop, computer, phone, tablet at this address: http://allsky.local or http://******
> 
> Do you want to reboot now? [y/n] 
>

Jonk2, 2021-10-07:
Nope, changing the /var/www/html/allsky folder structure to 777 didn't make a difference.

linuxkidd, 2021-10-08:
Sorry, I"m late to this party...
```
sudo: unable to resolve host allsky: Name or service not known
```
^ this error is likely caused by changing the Pi's hostname ( in `/etc/hostname` ), but not adding an entry in `/etc/hosts` -- typically, you'd replace the 'raspberrypi' name in the hosts file with the new hostname.  Or, you can just add a new line like:
```
127.0.1.1  newhostname
```
-------------------
Having two PI's with the same hostname on the same network can result in unpredictable things from your other computers perspective.  I'd suggest re-running the `gui/install.sh` ... specifying a different hostname for one of the PI's.

Example:
```
gui/install.sh allsky2
```

There's a mechanism built into that `gui/install.sh` script to handle this.

Jonk2, 2021-10-08:
@linuxkidd I thought so, but didn't realise you could set a custom name like that - could you add it to the install script as a prompt to the user? That would be easy for us - right at the end before reboot say "now rename your PI to <text box> or continue with the default (allsky)"....is that easy to do? I don't use my PIs just for allsky, so this makes sense to me.

@EricClaeys The install.sh you provided on that link did not work either directly, so I copied the text into a new notepad++ file (on windows and my notepad++ is set to unix for new files) and saved - it overwrote and has a slightly different filesize.
Anyway, I've run it and here's the result....rebooting now.

[image]

Jonk2, 2021-10-08:
> Is the "Allsky software is starting" message no appearing in the GUI, or on the website on a remote server? Either way, is there anything in /var/log/allsky.log right after your start AllSky?

Sorry forgot this - there's nothing in the log about the notification images and they're not being shown or saved.

maserowik, 2021-10-08:
@Jonk2 
I had the same thing you are having.  I redid whole git pull and now it shows up.

Mike

Jonk2, 2021-10-08:
Hi Mike, I've seen that post, and I'm kind of in the same position. I'm not a Linux user really, but can follow instructions well enough.
Whilst I apopreciate what this team is doing for this software, it's really difficult when things don't work - I'm hoping they get this right and can give us mere mortals a simple way to save settings, update, reinstate settings without having to do anything else.
And when it breaks, easily roll back to a working version.
Dare I say it, like most windows software works!
BTW, reinstalling the gui only doesn't reinstate my ftpsettings on the menu after a git pull, so I'm going to wait for now until these issues are ironed out and working.
I'm happy to test new things, but I couldn't tell you what versions of each file I have, as the log tells me 0.8.1, but they're talking of 0.8.5 now?

maserowik, 2021-10-08:
I would reinstall the main allsky app then reinstall the portal

Jonk2, 2021-10-08:
Yep tried that, no change - the ftp-settings.sh doesn't show in the menu.

Jonk2, 2021-10-08:
The file exists in /home/pi/allsky/config so I've manually edited that with the correct ftp settings for my NAS, but it doesn't appear to have worked.

EricClaeys, 2021-10-15:
@maphilli14 The logs would tell you what is happening.  Look for "endOfNight.sh" after it starts daytime capture.  There should be entries for starting keogram, startrails, and timelapse if you have all three set up to be produced.  You shouldn't need to manually run endOfNight.sh.

Jonk2, 2021-10-16:
> @Jonk2 Jon, have your issues been resolved with the newest relesse?

I've installed on the test PI whichever version was latest as of 15-10-21 (last night for me).

Most things are working as far as I can tell, including FTP to my NAS although setting to SFTP does not work.

I cannot test the exposure ramping on the PI indoors as the light doesn't change like it does outside.

There is an issue that I had this morning, and that was the keogram failed - here's the log:

> Oct 16 07:33:39 allsky2 allsky.sh[9919]: === Starting daytime capture ===
> Oct 16 07:33:39 allsky2 allsky.sh[9919]: ==========
> Oct 16 07:33:39 allsky2 allsky.sh[9919]: Saving images with delay of 5000 ms (5 sec)
> Oct 16 07:33:39 allsky2 allsky.sh[9919]: STARTING EXPOSURE at: 2021-10-16 07:33:39
> Oct 16 07:33:39 allsky2 allsky.sh[9919]: endOfNight.sh: ===== Generating Keogram
> Oct 16 07:33:41 allsky2 allsky.sh[9919]: OpenCV Error: Assertion failed (!fixedSize() || ((Mat*)obj)->size.operator()() == Size(_cols, _rows)) in create, file /build/opencv-L65chJ/opencv-3.2.0+dfsg/modules/core/src/matrix.cpp, line 2287
> Oct 16 07:33:41 allsky2 allsky.sh[9919]: terminate called after throwing an instance of 'cv::Exception'
> Oct 16 07:33:41 allsky2 allsky.sh[9919]:   what():  /build/opencv-L65chJ/opencv-3.2.0+dfsg/modules/core/src/matrix.cpp:2287: error: (-215) !fixedSize() || ((Mat*)obj)->size.operator()() == Size(_cols, _rows) in function create
> Oct 16 07:33:41 allsky2 allsky.sh[9919]: scripts/endOfNight.sh: line 79:  4157 Aborted                 "${ALLSKY_HOME}/keogram" ${SIZE_FILTER} -d "${DATE_DIR}/" -e ${EXTENSION} -o "${UPLOAD_FILE}" ${KEOGRAM_EXTRA_PARAMETERS}
> Oct 16 07:33:41 allsky2 allsky.sh[9919]: endOfNight.sh: ===== Generating Startrails

Startrail creation works.

The timelapse creation works, but has an issue, in that the resolution and aspect ratio was not correct even though the config.sh resize was set to "false" as per config.sh

The images are 3096x2080 and the resulting timelapse video was 2028x1520, when it should have been 3096x2080.

```
#!/bin/bash

# NOTES:
#  - Variable names must start at the beginning of a line.
#  - There should be nothing after a variable's value, i.e., no comments on the same line.
#  - Values can optionally be quoted with double quotes, NOT single quotes.
#  - If a variable occurs multiple times, only the last value is used.

# Choose between "ZWO", "RPiHQ", or "auto".  "auto" prefers ZWO if present.
CAMERA="ZWO"

########## Images
# Set to "true" to upload the current image to your website.
IMG_UPLOAD="true"

# If UPLOAD_IMG is "true", upload images every IMG_UPLOAD_FREQUENCY frames, e.g., every 5 frames.
# 0 uploades every frame.
IMG_UPLOAD_FREQUENCY=0

# The websites look in IMG_DIR for the current image.
# When using the Allsky website from the "allsky-website" packge, "allsky" is an
# alias for "/var/www/html/allsky/"  and   "current" is an alias for "/home/pi/allsky".
# If you use "current" and have the Allsky website, prepend "/current/" to
# the "imageName" in /var/www/html/allsky/config.js, for example:  imageName: "/current/image.jpg",
# Using "current" avoids copying the image to the website.
IMG_DIR="current"

# Set an optional prefix on the image file name, before "$FILENAME.$EXTENSION".
# This must match what the local web site thinks the name is (see /var/www/html/allsky/config.js).
IMG_PREFIX="latest-"

# Resize images before cropping and saving.
# Adjust IMG_WIDTH and IMG_HEIGHT according to your camera's sensor ratio.
IMG_RESIZE="false"
IMG_WIDTH=2028
IMG_HEIGHT=1520

# Crop images, used to improve them when using a fisheye lens.
CROP_IMAGE="false"
CROP_WIDTH=640
CROP_HEIGHT=480
CROP_OFFSET_X=0
[... 119 lines cut ...]
# !!! The next few lines are for backwards compatibility and will be removed in the future - do not change !!!
UPLOAD_IMG="${IMG_UPLOAD}"	# UPLOAD_IMG is th eold name for IMG_UPLOAD
if [ "${DAYTIME_CAPTURE}" = "true" ]; then
	DAYTIME=1		# DAYTIME is the old name for DAYTIME_CAPTURE
else
	DAYTIME=0
fi
CAPTURE_24HR=${DAYTIME_SAVE}	# CAPTURE_24HR is the old name for DAYTIME_SAVE

if [[ ${CAMERA} == "auto" ]]; then
	# Restore currently saved autodiscovered camera type, if any.
	source "${ALLSKY_CONFIG}/autocam.sh"	# May reset ${CAMERA}
fi
if [[ ${CAMERA} != "auto" ]]; then
	CAMERA_SETTINGS="${CAMERA_SETTINGS_DIR}/settings_${CAMERA}.json"
 
	# So scripts can conditionally ouput messages; DO NOT CHANGE NEXT LINES.
	ALLSKY_DEBUG_LEVEL=$(jq -r '.debuglevel' "${CAMERA_SETTINGS}")
fi
ALLSKY_DEBUG_LEVEL=${ALLSKY_DEBUG_LEVEL:-0}
```

Timelapse video details:

[image]

ckuethe, 2021-10-16:
In general, when reporting issues can you please tell us which git revision you're working with (`git log --oneline | head -1`)? We're in different timezones so it's not helpful to say "whichever version was latest as of 15-10-21 (last night for me)". Git log tells us unambiguously exactly which commits you've got.

Try run it by hand... something like this:

```
# fetch the latest source
git pull

 # look at the git history; make sure you have a recent fix "db38226 add mutex header"
git log --oneline | head

# recompile the binaries
make

# test
src/keogram -vv -d images/20211015 -e jpg -o /tmp/k.jpg --num-threads 1 --nice-level 0 2>&1 | tee /tmp/kl.txt
```

That will run the freshly compiled keogram with a single thread and verbose output, copying the output to `/tmp/kl.txt` That should at least tell you what was going on when keogram crashes. If it doesn't crash, them maybe we already fixed the bug.

If it does crash... congratulations, you get to learn how to use gdb (eg. https://www.cs.cmu.edu/~gilpin/tutorial/)

```
$ sudo apt install gdb
$ gdb src/keogram
(gdb) r -vv -d images/20211015 -e jpg -o /tmp/k.jpg --num-threads 1 --nice-level 0
(gdb) bt full
```

Then post the contents of the backtrace.

ckuethe, 2021-10-16:
You'll definitely want to grab the latest commits from git. I added some additional tracing to endofnight so that the exact failed command is logged rather than `Aborted "${ALLSKY_HOME}/keogram" ${SIZE_FILTER} -d "${DATE_DIR}/" -e ${EXTENSION} -o "${UPLOAD_FILE}" ${KEOGRAM_EXTRA_PARAMETERS}` which doesn't tell me if any parameters might be causing problems

maphilli14, 2021-10-19:
My errors are as follows was there any expectation that the latest git contains more debug to the root of the errors?

```

pi@allsky:~ $ cat /var/log/allsky.log | grep endOfNight
Oct 17 06:53:08 allsky allsky.sh[1299]: endOfNight.sh: ===== Generating Keogram
Oct 17 06:53:47 allsky allsky.sh[1299]: scripts/endOfNight.sh: line 79: 12686 Aborted                 "${ALLSKY_HOME}/keogram" ${SIZE_FILTER} -d "${DATE_DIR}/" -e ${EXTENSION} -o "${UPLOAD_FILE}" ${KEOGRAM_EXTRA_PARAMETERS}
Oct 17 06:53:47 allsky allsky.sh[1299]: endOfNight.sh: ===== Generating Startrails
Oct 17 06:54:12 allsky allsky.sh[1299]: scripts/endOfNight.sh: line 105: 12774 Aborted                 "${ALLSKY_HOME}/startrails" ${SIZE_FILTER} -d "${DATE_DIR}" -e ${EXTENSION} -b ${BRIGHTNESS_THRESHOLD} -o "${UPLOAD_FILE}"
Oct 17 06:54:12 allsky allsky.sh[1299]: endOfNight.sh: ===== Generating Timelapse
Oct 18 06:53:58 allsky allsky.sh[1567]: endOfNight.sh: ===== Generating Keogram
Oct 18 06:55:06 allsky allsky.sh[1567]: scripts/endOfNight.sh: line 79: 11628 Aborted                 "${ALLSKY_HOME}/keogram" ${SIZE_FILTER} -d "${DATE_DIR}/" -e ${EXTENSION} -o "${UPLOAD_FILE}" ${KEOGRAM_EXTRA_PARAMETERS}
Oct 18 06:55:06 allsky allsky.sh[1567]: endOfNight.sh: ===== Generating Startrails
Oct 18 06:55:38 allsky allsky.sh[1567]: scripts/endOfNight.sh: line 105: 11787 Aborted                 "${ALLSKY_HOME}/startrails" ${SIZE_FILTER} -d "${DATE_DIR}" -e ${EXTENSION} -b ${BRIGHTNESS_THRESHOLD} -o "${UPLOAD_FILE}"
Oct 18 06:55:38 allsky allsky.sh[1567]: endOfNight.sh: ===== Generating Timelapse
Oct 19 06:55:04 allsky allsky.sh[1567]: endOfNight.sh: ===== Generating Keogram
Oct 19 07:00:01 allsky allsky.sh[1567]: scripts/endOfNight.sh: line 79: 12725 Aborted                 "${ALLSKY_HOME}/keogram" ${SIZE_FILTER} -d "${DATE_DIR}/" -e ${EXTENSION} -o "${UPLOAD_FILE}" ${KEOGRAM_EXTRA_PARAMETERS}
Oct 19 07:00:01 allsky allsky.sh[1567]: endOfNight.sh: ===== Generating Startrails
Oct 19 07:00:57 allsky allsky.sh[1567]: scripts/endOfNight.sh: line 105: 13104 Aborted                 "${ALLSKY_HOME}/startrails" ${SIZE_FILTER} -d "${DATE_DIR}" -e ${EXTENSION} -b ${BRIGHTNESS_THRESHOLD} -o "${UPLOAD_FILE}"
Oct 19 07:00:57 allsky allsky.sh[1567]: endOfNight.sh: ===== Generating Timelapse
Oct 19 08:21:56 allsky allsky.sh[1567]: scripts/endOfNight.sh: line 136: /home/pi/allsky/scripts/endOfNight_additionalSteps.sh: No such file or directory
pi@allsky:~ $

```

EricClaeys, 2021-10-20:
@Jonk2, Jon, grab the latest scripts/endOfNight.sh.  It will expand the variables so you can see EXACTLY what arguments were passed to the keogram and startrails programs.  We should then be able to reproduce the problem and fix it.

Also, do you have the latest src/startrails.cpp and src/keogram.cpp?  If not, please grab them and put them in the allsky/src directory.  Then run:
~~~
cd src
make all
sudo make install
~~~

My guess is that your endOfNight.sh and .cpp files are out of sync.

ckuethe, 2021-10-20:
That's perfectly normal git output which summarizes lines/bytes added and removed.

Note that I said `src/keogram ...` - that'll run the program out of the source directory without installing. We're pretty confident that the state of the tools is currently better than whatever you've currently got installed, so you may as well `make install`.

Also, when grepping for endofnight or timelapse in the logs, please use `grep -C5 ...` which will add 5 additional lines of context around each matching line. There are often additional clues nearby.

EricClaeys, 2021-11-19:
@Jonk2, I believe the latest keogram file works and will resolve the problem you were having.
If this Issue is resolved, would you please close it?  Thanks.

Jonk2, 2021-11-21:
Ok, I'm closing this as the keogram is produced and ftp is working ok at the moment. 
I'm still unsure why every time a new piece of code is added you can't up the build number? Maybe a dicussion for another thread.

maphilli14, 2021-12-09:
I have a fresh pull from a few days ago and mine is doing this now too...

```

variables.sh: ===== Generating Keogram
terminate called after throwing an instance of 'std::out_of_range'
  what():  basic_string::substr: __pos (which is 140) > this->size() (which is 0)
bash: line 1: 14347 Aborted                 '/home/pi/allsky/keogram' -d '/home/pi/allsky/images/20211208' -e jpg -o '/home/pi/allsky/images/20211208/keogram/keogram-20211208.jpg' --rotate 0 --font-size 1.0 --font-line 1 --font-color '255 255 255'
Command Failed: '/home/pi/allsky/keogram'  -d '/home/pi/allsky/images/20211208' -e jpg -o '/home/pi/allsky/images/20211208/keogram/keogram-20211208.jpg' --rotate 0 --font-size 1.0 --font-line 1 --font-color '255 255 255'
variables.sh: ===== Generating Startrails
terminate called after throwing an instance of 'std::out_of_range'
  what():  basic_string::substr: __pos (which is 140) > this->size() (which is 0)
bash: line 1: 14375 Aborted                 '/home/pi/allsky/startrails' -d '/home/pi/allsky/images/20211208' -e jpg -b 0.1 -o '/home/pi/allsky/images/20211208/startrails/startrails-20211208.jpg'
Command Failed: '/home/pi/allsky/startrails'  -d '/home/pi/allsky/images/20211208' -e jpg -b 0.1 -o '/home/pi/allsky/images/20211208/startrails/startrails-20211208.jpg'
variables.sh: ===== Generating Timelapse
^C
```

EricClaeys, 2021-12-09:
@maphilli14 there's likely a corrupted file in /home/pi/allsky/images/20211208, or the image size or number of colors changed.  The keogram program should have caught it, but obviously not.
I would like to see the corrupt file(s) so if you could post them I would appreciate it.
Create a directory called images/test and copy all the images from 20211208 there.  Then run scripts/removeBadImages.sh 20211208.
It will delete the corrupt images and output a line for each one.  Then run scripts/runForDay.sh -k -s
To create the keogram and startrails.
If that works, post the corrupt files ftom the images/temp directory, then remove that directory.

maphilli14, 2021-12-10:
I lost that dataset when I freshened my install again due to some boot issues.  I had a keogram and startrail but a short/corrupt, but I don't seem to have the runForDay.sh script in the folder, it's all brand new clone from yestereday....
```

pi@allsky:~ $ ls -la allsky/scripts/
total 88
drwxr-xr-x  2 pi pi 4096 Dec  9 15:25 .
drwxr-xr-x 13 pi pi 4096 Dec 10 10:59 ..
-rwxr-xr-x  1 pi pi 2603 Dec  9 15:25 allsky_mfs.sh
-rwxr-xr-x  1 pi pi 3309 Dec  9 15:25 copy_notification_image.sh
-rwxr-xr-x  1 pi pi 2155 Dec  9 15:25 darkCapture.sh
-rwxr-xr-x  1 pi pi 3665 Dec  9 15:25 darkSubtract.sh
-rwxr-xr-x  1 pi pi  555 Dec  9 15:25 endOfNight_additionalSteps.repo
-rwxr-xr-x  1 pi pi 3056 Dec  9 15:25 endOfNight.sh
-rwxr-xr-x  1 pi pi  173 Dec  9 15:25 filename.sh
-rwxr-xr-x  1 pi pi 7240 Dec  9 15:25 generateForDay.sh
-rw-r--r--  1 pi pi  925 Dec  9 15:25 Makefile
-rwxr-xr-x  1 pi pi  960 Dec  9 15:25 postData.sh
-rwxr-xr-x  1 pi pi 6011 Dec  9 15:25 removeBadImages.sh
-rwxr-xr-x  1 pi pi 3123 Dec  9 15:25 saveImageDay.sh
-rwxr-xr-x  1 pi pi 3923 Dec  9 15:25 saveImageNight.sh
-rwxr-xr-x  1 pi pi 4975 Dec  9 15:25 timelapse.sh
-rwxr-xr-x  1 pi pi  507 Dec  9 15:25 uploadForDay.sh
-rwxr-xr-x  1 pi pi 5765 Dec  9 15:25 upload.sh
pi@allsky:~ $

```

EricClaeys, 2021-12-10:
@maphilli14 sorry, it's called `generateForDay.sh`.
Eric

maphilli14, 2021-12-10:
Hah!  NP, why did this happen to remove bad images?

`
pi@allsky:~/allsky $ /home/pi/allsky/scripts/removeBadImages.sh  /home/pi/allsky/images/20211209
/home/pi/allsky/scripts/removeBadImages.sh: line 93: /usr/bin/which: Argument list too long
/home/pi/allsky/scripts/removeBadImages.sh: line 131: /usr/bin/egrep: Argument list too long
/home/pi/allsky/scripts/removeBadImages.sh: line 139: /usr/bin/bc: Argument list too long
/home/pi/allsky/scripts/removeBadImages.sh: line 143: /usr/bin/bc: Argument list too long
/home/pi/allsky/scripts/removeBadImages.sh: line 144: [: -eq: unary operator expected
/home/pi/allsky/scripts/removeBadImages.sh: line 153: /usr/bin/bc: Argument list too long
/home/pi/allsky/scripts/removeBadImages.sh: line 154: [: -eq: unary operator expected
/home/pi/allsky/scripts/removeBadImages.sh: line 131: /usr/bin/egrep: Argument list too long
/home/pi/allsky/scripts/removeBadImages.sh: line 139: /usr/bin/bc: Argument list too long
/home/pi/allsky/scripts/removeBadImages.sh: line 143: /usr/bin/bc: Argument list too long
/home/pi/allsky/scripts/removeBadImages.sh: line 144: [: -eq: unary operator expected
/home/pi/allsky/scripts/removeBadImages.sh: line 153: /usr/bin/bc: Argument list too long
/home/pi/allsky/scripts/removeBadImages.sh: line 154: [: -eq: unary operator expected
/home/pi/allsky/scripts/removeBadImages.sh: line 131: /usr/bin/egrep: Argument list too long
/home/pi/allsky/scripts/removeBadImages.sh: line 139: /usr/bin/bc: Argument list too long`

That was CTRL+C'ed as it seemed just not doing what I wanted.  It re ran fine with that proper command and is very  handy!    Just didn't seem to fix the length issue.  I'll try to rerun removebadimages and try again...

[image]

EricClaeys, 2021-12-11:
@maphilli14 how many images do you have in the 20211209 directory?  I would guess many thousand and the code is hitting a limit of the number of files allowed on a command line.

Until I'm able to fix it you can move, for example, half the files to a new directory in allsky/images, e.g., 20211209b, and run removeBadImages.sh on each directory.  Then when done, move all the files from 20211209b back into 20211209.

I only need a couple corrupt files to debug the startrails and keogram issue you had.

Eric

maphilli14, 2021-12-11:
Do I have too sensitive a camera?  I'm using the 290 not the 174 we've been testing in BugID = 650
```

pi@allsky:~/allsky/images $ ls
20211209  20211210  CURRENTANI.gif  liveview-image.jpg
pi@allsky:~/allsky/images $ du -h
44M     ./20211210/thumbnails
5.1M    ./20211210/keogram
432K    ./20211210/startrails
8.6G    ./20211210
51M     ./20211209/thumbnails
5.7M    ./20211209/keogram
568K    ./20211209/startrails
11G     ./20211209
19G     .
pi@allsky:~/allsky/images $  ls -1 20211209/*.jpg | wc -l
13110
pi@allsky:~/allsky/images $  ls -1 20211210/*.jpg | wc -l
11407
pi@allsky:~/allsky/images $
```

EricClaeys, 2021-12-12:
@maphilli14 I don't think there is a camera that's too sensitive.  Sensitivity is good.
You just have a whole lot of pictures in one directory and it's hitting a limit in bash.  I won't be able to fix anything for a couple weeks.  Can you try what I suggested in my last post?

Going forward you could try increasing your delay between pictures, especially during the day.  That will decrease the number of pictures and hopefully get you below the limit.

maphilli14, 2021-12-13:
What's the actual limit?  I split in 2 and it still doesn't run :(

```

pi@allsky:~/allsky/images/20211212 $ mv image-20211212* 1
pi@allsky:~/allsky/images/20211212 $ mv image-20211213* 2
pi@allsky:~/allsky/images/20211212 $ ls -1 1 | wc -l
5224
pi@allsky:~/allsky/images/20211212 $ ls -1 2 | wc -l
5767
pi@allsky:~/allsky/images/20211212 $ /home/pi/allsky/scripts/removeBadImages.sh  /home/pi/allsky/images/20211212/1
/home/pi/allsky/scripts/removeBadImages.sh: line 93: /usr/bin/which: Argument list too long
/home/pi/allsky/scripts/removeBadImages.sh: line 131: /usr/bin/egrep: Argument list too long
/home/pi/allsky/scripts/removeBadImages.sh: line 139: /usr/bin/bc: Argument list too long
/home/pi/allsky/scripts/removeBadImages.sh: line 143: /usr/bin/bc: Argument list too long
/home/pi/allsky/scripts/removeBadImages.sh: line 144: [: -eq: unary operator expected
/home/pi/allsky/scripts/removeBadImages.sh: line 153: /usr/bin/bc: Argument list too long
/home/pi/allsky/scripts/removeBadImages.sh: line 154: [: -eq: unary operator expected
/home/pi/allsky/scripts/removeBadImages.sh: line 131: /usr/bin/egrep: Argument list too long
/home/pi/allsky/scripts/removeBadImages.sh: line 139: /usr/bin/bc: Argument list too long
/home/pi/allsky/scripts/removeBadImages.sh: line 143: /usr/bin/bc: Argument list too long
/home/pi/allsky/scripts/removeBadImages.sh: line 144: [: -eq: unary operator expected
/home/pi/allsky/scripts/removeBadImages.sh: line 153: /usr/bin/bc: Argument list too long
/home/pi/allsky/scripts/removeBadImages.sh: line 154: [: -eq: unary operator expected
/home/pi/allsky/scripts/removeBadImages.sh: line 131: /usr/bin/egrep: Argument list too long
/home/pi/allsky/scripts/removeBadImages.sh: line 139: /usr/bin/bc: Argument list too long
/home/pi/allsky/scripts/removeBadImages.sh: line 143: /usr/bin/bc: Argument list too long
/home/pi/allsky/scripts/removeBadImages.sh: line 144: [: -eq: unary operator expected
/home/pi/allsky/scripts/removeBadImages.sh: line 153: /usr/bin/bc: Argument list too long
/home/pi/allsky/scripts/removeBadImages.sh: line 154: [: -eq: unary operator expected
/home/pi/allsky/scripts/removeBadImages.sh: line 131: /usr/bin/egrep: Argument list too long
/home/pi/allsky/scripts/removeBadImages.sh: line 139: /usr/bin/bc: Argument list too long
/home/pi/allsky/scripts/removeBadImages.sh: line 143: /usr/bin/bc: Argument list too long
/home/pi/allsky/scripts/removeBadImages.sh: line 144: [: -eq: unary operator expected
/home/pi/allsky/scripts/removeBadImages.sh: line 153: /usr/bin/bc: Argument list too long
/home/pi/allsky/scripts/removeBadImages.sh: line 154: [: -eq: unary operator expected
/home/pi/allsky/scripts/removeBadImages.sh: line 131: /usr/bin/egrep: Argument list too long
/home/pi/allsky/scripts/removeBadImages.sh: line 139: /usr/bin/bc: Argument list too long
/home/pi/allsky/scripts/removeBadImages.sh: line 143: /usr/bin/bc: Argument list too long
```

maphilli14, 2021-12-13:
I only record at night and it's using the default delay between photos and a 4s sub exp :(

maphilli14, 2021-12-13:
I'm pruning them by hand to test.... 
[image]

linuxkidd, 2021-12-13:
Hi @maphilli14 ...  You can run the following command to get the argument list limit:
```
getconf ARG_MAX
```

Note that the value returned is the character length of the command and is enforced by the operating system ( not the shell ).  Since it is a character count limit, the path for each file is included in the limit.   With that in mind, you may be able to get the script to run with the following:
```
cd /home/pi/allsky/scripts
./removeBadImages.sh ../images/20211212/1
```

This alone will reduce your character count by 14 for -each- image.

maphilli14, 2021-12-13:
I split it into quarters, or closely... but .... Damn, it's really punchy!
[image]

GOOD

BAD
[image]

maphilli14, 2021-12-13:
actually those total file counts are reversed, the 2528 was too long and gave errs but the 2505 processed fine and removed 23 additional bad files above and beyond the small file size i pruned by hand.  Why am I the only one with so many files?  I backed off the default inter 'frame' capture from 10ms to 100ms to abate the bad file issues.... honestly I was hoping that at 4s subs it would be less churn on the disk i/o and better results... keep tweaking I guess...

maphilli14, 2021-12-14:
Changing the delay from 10 to 100 didn't result in a successful timelapse for the recent end of night...

this still confuses me...

[image]

linuxkidd, 2021-12-14:
Hi @maphilli14 
  I think others don't experience this because they have a much longer exposure time and daytime delay.  My 'night' collections ( with 0 degree horizon offset ) only have 1494 exposures.

My settings:
- Night max exposure: 34000
- Night delay: 10

My other system only generates up to 804 exposures:
- Night max exposure: 60000
- Night delay: 10

There can be some code changes in the script to accommodate larger image sets ( creating a loop that processes say.. only 500 images at a time ).

If you need assistance tweaking your exposure / delay settings, or the image count you're achieving is necessary, please [open a new issue](https://github.com/thomasjacquin/allsky/issues/new/choose) so we can work the problem separately since it's not related to the original purpose of this ticket (also, I'm sure Jonk2 isn't thrilled with getting loads of updates on an issue they consider closed ).

Thanks!

closed 2021-12-14 by linuxkidd