AllskyTeam/allsky #572: source variables.sh in gui and website install.sh scripts

EricClaeys, 2021-10-05:
Doing a `source $ALLSKY_HOME/variables.sh` in gui/install.sh and website.install.sh (and other scripts?) would allow us to remove hard-coded paths, especially if we also set `ALLSKY_GUI_DIR=/var/www/html `in variables.sh.  This will make it easier to move/rename those directories in the future, and would be more consistent with the other scripts.

EricClaeys, 2021-10-05:
Chris, what do you mean by "the main config"?  Not sure I understand your comment (other than the "good idea" part...).

ckuethe, 2021-10-05:
Basically, whatever format of configuration we're using - a single json or json and shell snippets - variables.sh should read that config and set up the environment appropriately.

EricClaeys, 2021-10-06:
I think `source variable.sh` is always followed by `source config.sh` and often by `source ftp-settings.sh`, so having variables.sh include the functionality of the other two files as well as settings_*.json would make sense.  Especially since variables.sh only sets the variables once per invocation of allsky.sh (or whatever program is invoked from the command line).

When we have capture*.cpp read a configuration file and re-read it on SIGHUP, we would need to make sure that setting changes are re-read by variables.sh, possibly via a command-line argument to the shell scripts capture*.cpp calls.

EricClaeys, 2021-10-08:
Completed and merged into code.

closed 2021-10-08 by EricClaeys