AllskyTeam/allsky #576: Wiki

maserowik, 2021-10-06:
@EricClaeys 
Not to be a PITA but I was wondering if you had a chance to upload the WIP wiki yet .  see issue 458.

Mike

EricClaeys, 2021-10-06:
@maserowik, please be a PITA.  I sometimes need the push.  I'm doing it now.  I don't have a lot yet but something is better than nothing.

EricClaeys, 2021-10-06:
@maserowik, I added what I have and will continue adding as I get more information.
I'll let you decide when to close this issue - now because I added what I have, or later after the Wiki is more complete (it will never be done since I expect to always be updating it).

EricClaeys, 2021-10-06:
@maserowik Can you do me a favor?  I don't have access to my Pi but want to test something.

In scripts/upload.sh, after the `echo set net:timeout 20` line, add this:
`echo "echo ls xxxx || echo xxxx does not exist"` on one line and
`echo "echo test 3 && echo test 4"` on another.
Then once the next image is uploaded, look in ~/allsky/tmp/upload_log.txt.  If it has "xxxx" in it, copy the file to your reply.

You can then remove those two lines.

I appreciate it.

maserowik, 2021-10-06:
> @maserowik Can you do me a favor? I don't have access to my Pi but want to test something.
>
> In scripts/upload.sh, after the `echo set net:timeout 20` line, add this: `echo "echo ls xxxx || echo xxxx [...]

105                 echo set net:timeout 20
106
107                 echo "echo ls xxxx || echo xxxx does not exist"
108                 echo "echo test 3 && echo test 4"
109

<
pi@Pi-AllSky:~/allsky/tmp $ cat upload_log.txt
ls xxxx
test 3
test 4
pi@Pi-AllSky:~/allsky/tmp $
>

maserowik, 2021-10-06:
1 ls: ls xxxx: Fatal error: max-retries exceeded
 2 xxxx does not exist
 3 ls: ls xxxx: Fatal error: max-retries exceeded
 4 xxxx still does not exist
 5

[image]

EricClaeys, 2021-10-06:
Other than the weird Fatal error message, that's the output I expected.

Another test pretty please.
Undo what you did in the prior test.
Place a "#" in front of the "put" line, then add this line:
`echo "put '${FILE_TO_UPLOAD}XXXXX' -o '${TEMP_NAME}' || echo 'put failed!' && exit"`

You should see some error from lftp, then "put failed!".  It will NOT copy the file since it exits after the error.

When done, remove the line you added and deleted the "#" you added.
Thanks again.

EricClaeys, 2021-10-06:
Probably by pasting the text in, the highlight it and press the `<>` button.  I think there's something in the text (probably the tilda) that the web page is interpreting to mean strikethrough, just like a `#` means the text is a header.

That last message is good.  I now know how to get lftp to exit as soon as it encounters an error, rather than continuing and encountering another error.
Thanks for your help.  I'll submit an update.  It won't fix the inability to upload files, but will make it easier to see what the first error is.

closed 2021-10-14 by ckuethe