AllskyTeam/allsky #580: Focus and refresh.

Telfish2, 2021-10-06:
So tonight is the first night out with the Allsky. It's taking images but I am finding it really difficult to focus as the images are only refreshing about ever minute which is slower than the day. the interval is set to the default one second so I was hoping for a pretty continuous steam of images so I could see the effects of my adjustments in pretty much real time.

I am thinking there is a way to achieve this. All settings are at default right now.

Terry

ckuethe, 2021-10-06:
Check your exposure and delay settings. My config has this:

```
...
   "nightautoexposure":"0",
   "nightautogain":"0",
   "nightdelay":"0.001",
   "nightexposure":"5000",
   "nightgain":"45",
...
```

Which delays 1 microsecond and then takes another 5000 millisecond exposure.

ckuethe, 2021-10-06:
Try getting a close focus during the day, like of getting an approximate alignment of a finderscope so that you don't have as much to do at night. Maybe you still have a little bit of tweaking to do to dial in the focus on stars but it should be close.