AllskyTeam/allsky #586: Keograms

maserowik, 2021-10-06:
This may have been answered before but I am not able to find it.  
on my hosting site I have the keograms uploaded fine.

[image]

but when I click on the image I get a shrunken image
[image]

is there a setting I need to adjust?
Mike

maserowik, 2021-10-06:
I have really mucked up my copy of allsky.  I did the new version and now I have something mucked up good.

EricClaeys, 2021-10-06:
@ckuethe Did you add the "addRow" and "-finishline" variables?  I seem to remember them having something to do with skinny images.