AllskyTeam/allsky #587: Running two version

maserowik, 2021-10-06:
I see there have been quite a few upgraded/enhancements to this software/application.  Is there a way to have two version installed on the same PI and I can select what version I want to run?

EricClaeys, 2021-10-06:
@maserowik It's probably possible but may not be easy yet.  You can try the instructions at [https://github.com/thomasjacquin/allsky/wiki/Installation-Tips](url) but for one of the installs, change the `git clone` to get a prior version.  The reason it may not be easy is that we recently changed the location of a lot of files, for example the *.cpp files are in allsky/src and the config files are in allsky/config.  And we now have Makefile's that do a lot of the installation.

If you want to try to manually do it, you could put the older release in ~/allsky using the prior locations, then install the newest version somewhere else and edit the "variables.sh" to hard-code ALLSKY_HOME to the new location.  One gotcha is that /etc/profile.d/allsky.sh hard-codes ALLSKY_HOME to /home/pi/allsky.
Then set ALLSKY_IMAGES in variables.sh to point to ~/allsky/images.  That way both versions will use the same /images folder so the web sites can still find the images.

This is all in theory - I haven't tested it.

maserowik, 2021-10-06:
Sound like it would be more problem than gains.