AllskyTeam/allsky #590: Installation Fails

KH6JAG, 2021-10-06:
Dependancies Installation fails. I've tried to install using a fresh Buster image on Rpi 4, 3 and Zero.  
Installation worked OK two weeks ago following the same process.

ckuethe, 2021-10-06:
http://www.catb.org/~esr/faqs/smart-questions.html#beprecise

adymac101, 2021-10-06:
See last 2 comments here https://github.com/thomasjacquin/allsky/issues/560

KH6JAG, 2021-10-06:
Sorry for the lack of info - I've been frustrated all day trying to do both a fresh installation and and update with a directory removal and installation.  The result always fails as shown below.

pi@raspberrypi:~ $ sudo apt-get install git
Reading package lists... Done
Building dependency tree       
Reading state information... Done
git is already the newest version (1:2.20.1-2+deb10u3).
The following package was automatically installed and is no longer required:
  python-colorzero
Use 'sudo apt autoremove' to remove it.
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
pi@raspberrypi:~ $ git clone --recursive https://github.com/thomasjacquin/allsky.git
Cloning into 'allsky'...
remote: Enumerating objects: 2723, done.
remote: Counting objects: 100% (1160/1160), done.
remote: Compressing objects: 100% (644/644), done.
remote: Total 2723 (delta 732), reused 828 (delta 491), pack-reused 1563
Receiving objects: 100% (2723/2723), 57.42 MiB | 3.16 MiB/s, done.
Resolving deltas: 100% (1637/1637), done.
Submodule 'src/sunwait-src' (https://github.com/risacher/sunwait) registered for path 'src/sunwait-src'
Cloning into '/home/pi/allsky/src/sunwait-src'...
remote: Enumerating objects: 92, done.        
remote: Counting objects: 100% (43/43), done.        
remote: Compressing objects: 100% (40/40), done.        
remote: Total 92 (delta 18), reused 21 (delta 3), pack-reused 49        
Submodule path 'src/sunwait-src': checked out '102cb417ecbb7a3757ba9ee4b94d6db3225124c4'
pi@raspberrypi:~ $ cd allsky
pi@raspberrypi:~/allsky $ ./install.sh

**********************************************
*** Welcome to the Allsky Camera installer ***
**********************************************

* Dependencies installation

make[1]: Entering directory '/home/pi/allsky/src'
Package libusb-1.0 was not found in the pkg-config search path.
Perhaps you should add the directory containing `libusb-1.0.pc'
to the PKG_CONFIG_PATH environment variable
No package 'libusb-1.0' found
Makefile:46: *** Did not find USB Libraries, try 'make deps'.  Stop.
make[1]: Leaving directory '/home/pi/allsky/src'
make: *** [Makefile:30: deps] Error 2
Installing dependencies failed!
pi@raspberrypi:~/allsky $

ckuethe, 2021-10-06:
That's better. Note that the error says `Makefile:46: *** Did not find USB Libraries, try 'make deps'. Stop.`

So `make deps` should fix that for you, but we can probably update the makefile to try autofix it too.

KH6JAG, 2021-10-06:
The lines through the text resulted after I pasted in the text into the comment space

ckuethe, 2021-10-06:
If you put `~~~` by itself on a line before your paste, and again after the paste, it'll format as a code block. I edited your reply to add those...

ckuethe, 2021-10-06:
Try this: edit src/Makefile, insert a tab at the beginning of lines 36-42, so it looks like this:

```
deps:
	ifneq ($(shell id -u), 0)
		@echo This must be ran with root permissions.
		@echo Please run \'sudo make deps\'
	else
		@echo `date +%F\ %R:%S` Installing build dependencies...
		@apt update && apt -y install libopencv-dev libusb-dev libusb-1.0-0-dev ffmpeg gawk lftp jq imagemagick
	endif
```

then try `sudo make deps` again. Do not copy and paste the section above, you have to edit the file and insert a tab (not spaces) at the beginning of those lines.

KH6JAG, 2021-10-06:
Just to be sure I did the edit correctly all I added was the tabs to make the text format indentical to your suggestion

ckuethe, 2021-10-06:
Yes, at this stage cameras have nothing to do with it.

I'm not sure your source tree is correct; the following worked for me (just to check that `make deps` worked correctly:

```
mkdir /tmp/foobar
cd /tmp/foobar
git clone https://github.com/thomasjacquin/allsky
cd allsky/src
make deps
sudo make deps
```

What happens when you do that?

KH6JAG, 2021-10-06:
This is the result - I will be away fro the PC for awhile as I need to head to the obsevatory.

~~~
pi@raspberrypi:~ $ sudo mkdir /tmp/foobar
mkdir: cannot create directory ‘/tmp/foobar’: File exists
pi@raspberrypi:~ $ cd /tmp/foobar
pi@raspberrypi:/tmp/foobar $ git clone https://github.com/thomasjacquin/allsky
pi@raspberrypi:/tmp/foobar $ git clone https://github.com/thomasjacquin/allsky.git
fatal: could not create work tree dir 'allsky': Permission denied
pi@raspberrypi:/tmp/foobar $ sudo git clone https://github.com/thomasjacquin/allsky.git
Cloning into 'allsky'...
remote: Enumerating objects: 2731, done.
remote: Counting objects: 100% (1168/1168), done.
remote: Compressing objects: 100% (653/653), done.
remote: Total 2731 (delta 736), reused 825 (delta 490), pack-reused 1563
Receiving objects: 100% (2731/2731), 57.43 MiB | 3.35 MiB/s, done.
Resolving deltas: 100% (1641/1641), done.
pi@raspberrypi:/tmp/foobar $ cd allsky
pi@raspberrypi:/tmp/foobar/allsky $ cd src
pi@raspberrypi:/tmp/foobar/allsky/src $ make deps
Package libusb-1.0 was not found in the pkg-config search path.
Perhaps you should add the directory containing `libusb-1.0.pc'
to the PKG_CONFIG_PATH environment variable
No package 'libusb-1.0' found
Makefile:46: *** Did not find USB Libraries, try 'make deps'.  Stop.
pi@raspberrypi:/tmp/foobar/allsky/src $ sudo make deps
Package libusb-1.0 was not found in the pkg-config search path.
Perhaps you should add the directory containing `libusb-1.0.pc'
to the PKG_CONFIG_PATH environment variable
No package 'libusb-1.0' found
Makefile:46: *** Did not find USB Libraries, try 'make deps'.  Stop.
pi@raspberrypi:/tmp/foobar/allsky/src $ 

~~~

linuxkidd, 2021-10-07:
FYI - I have a reproducer for this and am working to figure out the issue.  If you'd like to move forward without waiting for a fix, please run the following command.
```
sudo apt update
sudo apt -y install libopencv-dev libusb-dev libusb-1.0-0-dev ffmpeg gawk lftp jq imagemagick
```

After that, you can re-run the `./install.sh` script and it -should- finish the installation.

Let us know how it turns out, and thanks for your patience!

closed 2021-10-07 by ckuethe

KH6JAG, 2021-10-07:
Thanks for the help but I'm still having difficulties.

I tried the changes proposed above but the installation failed.

I then started with a fresh Buster image and did a new install of the allsky and the GUI

Program started but no image available using RPiHQ.

I checked the config.sh file and found the following
# Choose between "ZWO", "", or "auto".  "auto" prefers ZWO if present.
CAMERA=" "  I changed to CAMERA=”auto” saved and rebooted

Still no image

A connected a ASI-120MC-S rebooted still no image

Changed config.sh to CAMERA=”ZWO” saved and rebooted
still no image

Changed to CAMERA=”RPiHQ” saved and rebooted
Still no image

The systems hangs with the ASI120 installed but not with the RPiHQ.  On one reboot wifi would not connect with ASI120 installed.  Removed ASI120 – rebooted – wifi now works.  Installed ASI120 – still have wifi.  I have not been able to repeat the loss of wifi but the system keeps stalling with the ASI120 installed
RPi4 4BG – 32GB SD card