AllskyTeam/allsky #623: lftp_cmd.txt

maserowik, 2021-10-10:
Trying to get a better understanding of the log files.  I am looking at the 
lftp_cmd.txt file located in /allsky/tmp
```
set ssl:check-hostname false
open --user           ' --password '            ' 'ftp: '
set net:max-retries 2
set net:timeout 20
rm -f 'SaveImageNight-25798'
put 'image.jpg' -o 'SaveImageNight-25798' || (echo 'put of image.jpg failed!' && exit)
rm -f 'image.jpg'
mv 'SaveImageNight-25798' '/domains/n3bsq.com/public_html/allsky/image/image.jpg' || ( echo 'mv of SaveImageNight-25798 to /domains/n3bsq.com/public_html/allsky/image/image.jpg failed!' )
exit
```

the line that starts with the put ...  and mv ... is the echo a condition of it not working? or is it telling me the moved failed. 
I check the server and I have a valid image on there.

Thanks
Mike

linuxkidd, 2021-10-10:
Hi @maserowik  
  I think @EricClaeys mentioned this in another issue, but I can't remember if it was you he was talking to.. so I'll repeat it here.

If you are logged into your Allsky pi as the `pi` user, and then enter the following:
```
cd ~pi/allsky
lftp
```
Once there, you can copy/paste each of the commands from the lftp_cmd.txt file.  For the 'put' and 'mv' commands, stop right before the `||` and hit enter.

Let us know the outcome of this ( you can also copy+paste the terminal output if there's something interesting ).

Thanks!

EricClaeys, 2021-10-10:
@maserowik Mike,
Glad you asked.  I just noticed something I need to fix.

the "put..." command tries to copy the "image.jpg" file from your Pi to a file called `SaveImageNight-25798` on your server, in the top-level directory.  I will change that to copy to the REMOTE_DIR you have specified in ftp-settings.sh (`/domains/n3bsq.com/public_html/allsky/image` in your case).
If the "put" fails, it executes the "echo" and then exit, so it doesn't try the next commands will will also fail and produce confusing error messages.  I have change in to fix the "rm -f 'image.jpg'" which should also be in REMOTE_DIR.  The `||` symbol is "or" - it checks if the thing before it succeeded, if so, it skips the part after `||`, otherwise it does the right side.  It's a different way of saying "IF the part on the left failed THEN do the part on the right".