AllskyTeam/allsky #631: New Git pull 10/11/2021 @ 10 AM EST issues #3

maserowik, 2021-10-10:
looking at the /var/log/lighttpd/error.log  are there error something to worry about

~~~
2021-10-10 10:20:05: (mod_fastcgi.c.421) FastCGI-stderr: PHP Notice:  Undefined offset: 1 in /var/www/html/includes/dashboard.php on line 21
2021-10-10 10:20:05: (mod_fastcgi.c.421) FastCGI-stderr: PHP Notice:  Undefined offset: 1 in /var/www/html/includes/dashboard.php on line 23
2021-10-10 10:20:05: (mod_fastcgi.c.421) FastCGI-stderr: PHP Notice:  Undefined offset: 0 in /var/www/html/includes/dashboard.php on line 29
2021-10-10 10:20:05: (mod_fastcgi.c.421) FastCGI-stderr: PHP Notice:  Undefined offset: 1 in /var/www/html/includes/dashboard.php on line 30
2021-10-10 10:20:05: (mod_fastcgi.c.421) FastCGI-stderr: PHP Notice:  Undefined offset: 1 in /var/www/html/includes/dashboard.php on line 32
2021-10-10 10:20:05: (mod_fastcgi.c.421) FastCGI-stderr: PHP Notice:  Undefined offset: 1 in /var/www/html/includes/dashboard.php on line 34
2021-10-10 10:20:05: (mod_fastcgi.c.421) FastCGI-stderr: PHP Notice:  Undefined offset: 1 in /var/www/html/includes/dashboard.php on line 36
2021-10-10 10:20:05: (mod_fastcgi.c.421) FastCGI-stderr: PHP Notice:  Undefined offset: 1 in /var/www/html/includes/dashboard.php on line 38
2021-10-10 10:20:05: (mod_fastcgi.c.421) FastCGI-stderr: PHP Notice:  Undefined offset: 1 in /var/www/html/includes/dashboard.php on line 40
2021-10-10 10:20:05: (mod_fastcgi.c.421) FastCGI-stderr: PHP Notice:  Undefined offset: 1 in /var/www/html/includes/dashboard.php on line 42
2021-10-10 10:20:05: (mod_fastcgi.c.421) FastCGI-stderr: PHP Notice:  Undefined offset: 1 in /var/www/html/includes/dashboard.php on line 44
2021-10-10 10:20:05: (mod_fastcgi.c.421) FastCGI-stderr: PHP Notice:  Undefined variable: wlan0up in /var/www/html/includes/dashboard.php on line 120
2021-10-10 10:20:08: (mod_fastcgi.c.421) FastCGI-stderr: PHP Warning:  include_once(includes/videos.php): failed to open stream: No such file or directory in /var/www/html/index.php on line 88
2021-10-10 10:20:08: (mod_fastcgi.c.421) FastCGI-stderr: PHP Warning:  include_once(): Failed opening 'includes/videos.php' for inclusion (include_path='.:/usr/share/php') in /var/www/html/index.php on line 88
2021-10-10 10:20:08: (mod_fastcgi.c.421) FastCGI-stderr: PHP Warning:  include_once(includes/keograms.php): failed to open stream: No such file or directory in /var/www/html/index.php on line 89
2021-10-10 10:20:08: (mod_fastcgi.c.421) FastCGI-stderr: PHP Warning:  include_once(): Failed opening 'includes/keograms.php' for inclusion (include_path='.:/usr/share/php') in /var/www/html/index.php on line 89
2021-10-10 10:20:08: (mod_fastcgi.c.421) FastCGI-stderr: PHP Warning:  include_once(includes/startrails.php): failed to open stream: No such file or directory in /var/www/html/index.php on line 90
2021-10-10 10:20:08: (mod_fastcgi.c.421) FastCGI-stderr: PHP Warning:  include_once(): Failed opening 'includes/startrails.php' for inclusion (include_path='.:/usr/share/php') in /var/www/html/index.php on line 90
2021-10-10 10:20:08: (mod_fastcgi.c.421) FastCGI-stderr: PHP Notice:  Undefined variable: networks in /var/www/html/includes/configure_client.php on line 179
2021-10-10 10:20:08: (mod_fastcgi.c.421) FastCGI-stderr: PHP Warning:  Invalid argument supplied for foreach() in /var/www/html/includes/configure_client.php on line 179
2021-10-10 10:20:09: (mod_fastcgi.c.421) FastCGI-stderr: PHP Warning:  include_once(includes/videos.php): failed to open stream: No such file or directory in /var/www/html/index.php on line 88
2021-10-10 10:20:09: (mod_fastcgi.c.421) FastCGI-stderr: PHP Warning:  include_once(): Failed opening 'includes/videos.php' for inclusion (include_path='.:/usr/share/php') in /var/www/html/index.php on line 88
2021-10-10 10:20:09: (mod_fastcgi.c.421) FastCGI-stderr: PHP Warning:  include_once(includes/keograms.php): failed to open stream: No such file or directory in /var/www/html/index.php on line 89
2021-10-10 10:20:09: (mod_fastcgi.c.421) FastCGI-stderr: PHP Warning:  include_once(): Failed opening 'includes/keograms.php' for inclusion (include_path='.:/usr/share/php') in /var/www/html/index.php on line 89
2021-10-10 10:20:09: (mod_fastcgi.c.421) FastCGI-stderr: PHP Warning:  include_once(includes/startrails.php): failed to open stream: No such file or directory in /var/www/html/index.php on line 90
2021-10-10 10:20:09: (mod_fastcgi.c.421) FastCGI-stderr: PHP Warning:  include_once(): Failed opening 'includes/startrails.php' for inclusion (include_path='.:/usr/share/php') in /var/www/html/index.php on line 90
2021-10-10 10:20:29: (mod_fastcgi.c.421) FastCGI-stderr: PHP Warning:  include_once(includes/videos.php): failed to open stream: No such file or directory in /var/www/html/index.php on line 88
2021-10-10 10:20:29: (mod_fastcgi.c.421) FastCGI-stderr: PHP Warning:  include_once(): Failed opening 'includes/videos.php' for inclusion (include_path='.:/usr/share/php') in /var/www/html/index.php on line 88
2021-10-10 10:20:29: (mod_fastcgi.c.421) FastCGI-stderr: PHP Warning:  include_once(includes/keograms.php): failed to open stream: No such file or directory in /var/www/html/index.php on line 89
2021-10-10 10:20:29: (mod_fastcgi.c.421) FastCGI-stderr: PHP Warning:  include_once(): Failed opening 'includes/keograms.php' for inclusion (include_path='.:/usr/share/php') in /var/www/html/index.php on line 89
2021-10-10 10:20:29: (mod_fastcgi.c.421) FastCGI-stderr: PHP Warning:  include_once(includes/startrails.php): failed to open stream: No such file or directory in /var/www/html/index.php on line 90
2021-10-10 10:20:29: (mod_fastcgi.c.421) FastCGI-stderr: PHP Warning:  include_once(): Failed opening 'includes/startrails.php' for inclusion (include_path='.:/usr/share/php') in /var/www/html/index.php on line 90
2021-10-10 10:20:38: (mod_fastcgi.c.421) FastCGI-stderr: PHP Warning:  include_once(includes/videos.php): failed to open stream: No such file or directory in /var/www/html/index.php on line 88
2021-10-10 10:20:38: (mod_fastcgi.c.421) FastCGI-stderr: PHP Warning:  include_once(): Failed opening 'includes/videos.php' for inclusion (include_path='.:/usr/share/php') in /var/www/html/index.php on line 88
2021-10-10 10:20:38: (mod_fastcgi.c.421) FastCGI-stderr: PHP Warning:  include_once(includes/keograms.php): failed to open stream: No such file or directory in /var/www/html/index.php on line 89
2021-10-10 10:20:38: (mod_fastcgi.c.421) FastCGI-stderr: PHP Warning:  include_once(): Failed opening 'includes/keograms.php' for inclusion (include_path='.:/usr/share/php') in /var/www/html/index.php on line 89
2021-10-10 10:20:38: (mod_fastcgi.c.421) FastCGI-stderr: PHP Warning:  include_once(includes/startrails.php): failed to open stream: No such file or directory in /var/www/html/index.php on line 90
2021-10-10 10:20:38: (mod_fastcgi.c.421) FastCGI-stderr: PHP Warning:  include_once(): Failed opening 'includes/startrails.php' for inclusion (include_path='.:/usr/share/php') in /var/www/html/index.php on line 90
2021-10-10 10:26:37: (mod_fastcgi.c.421) FastCGI-stderr: PHP Warning:  include_once(includes/videos.php): failed to open stream: No such file or directory in /var/www/html/index.php on line 88
2021-10-10 10:26:37: (mod_fastcgi.c.421) FastCGI-stderr: PHP Warning:  include_once(): Failed opening 'includes/videos.php' for inclusion (include_path='.:/usr/share/php') in /var/www/html/index.php on line 88
2021-10-10 10:26:37: (mod_fastcgi.c.421) FastCGI-stderr: PHP Warning:  include_once(includes/keograms.php): failed to open stream: No such file or directory in /var/www/html/index.php on line 89
2021-10-10 10:26:37: (mod_fastcgi.c.421) FastCGI-stderr: PHP Warning:  include_once(): Failed opening 'includes/keograms.php' for inclusion (include_path='.:/usr/share/php') in /var/www/html/index.php on line 89
2021-10-10 10:26:37: (mod_fastcgi.c.421) FastCGI-stderr: PHP Warning:  include_once(includes/startrails.php): failed to open stream: No such file or directory in /var/www/html/index.php on line 90
2021-10-10 10:26:37: (mod_fastcgi.c.421) FastCGI-stderr: PHP Warning:  include_once(): Failed opening 'includes/startrails.php' for inclusion (include_path='.:/usr/share/php') in /var/www/html/index.php on line 90
2021-10-10 10:30:57: (mod_fastcgi.c.421) FastCGI-stderr: PHP Warning:  include_once(includes/videos.php): failed to open stream: No such file or directory in /var/www/html/index.php on line 88
[... cut ...]
2021-10-10 10:30:57: (mod_fastcgi.c.421) FastCGI-stderr: PHP Warning:  include_once(): Failed opening 'includes/keograms.php' for inclusion (include_path='.:/usr/share/php') in /var/www/html/index.php on line 89
2021-10-10 10:30:57: (mod_fastcgi.c.421) FastCGI-stderr: PHP Warning:  include_once(includes/startrails.php): failed to open stream: No such file or directory in /var/www/html/index.php on line 90
2021-10-10 10:30:57: (mod_fastcgi.c.421) FastCGI-stderr: PHP Warning:  include_once(): Failed opening 'includes/startrails.php' for inclusion (include_path='.:/usr/share/php') in /var/www/html/index.php on line 90
~~~

EricClaeys, 2021-10-11:
@maserowik Well this is embarrassing.  the errors about "No such file or directory" are my fault.  I deleted some files but didn't tell the script so it went looking for them.  I'll submit a bug fix.  Sorry about that.
The others I haven't seen before so I'll fix them as well.
Thanks for letting me know.

EricClaeys, 2021-10-11:
The updates to remove these error messages is available in allsky-portal.
Also, debug statements at debug level 4 are now displayed when an image resize, crop, or stretch is done.