AllskyTeam/allsky #636: Install fails on fresh RPi

ssagerian, 2021-10-11:
After re imaging an SD card with the latest Rapian OS, and allowing it to update, I attempted to re install allsky. Following the instructions on the wiki for installing the allsky (not the instructions show on the left), bullet 1 (mkdir) and 4 (git clone) fail without a sudo command, but then later the build fails to compile. I believe the issue is that the folder which the git uses needs to be owned by pi or given more liberal access permissions (that would need to be documented)
[image]
.

ssagerian, 2021-10-11:
just a follow up to my last comment, the camera is running. running command raspistill -o bob.jpg does work.
[image]
.

ssagerian, 2021-10-12:
Line 3 of the install tips doesnt set the permissions correct.
[image]
.

EricClaeys, 2021-10-12:
@ssagerian What's wrong with the permissions?  Doesn't it set the owner and group to "pi" and make it mode 664?

EricClaeys, 2021-10-13:
@ssagerian Steve, that file is nothing but a pain in the butt.  It is initially created empty by `allsky.sh` then after camera detection completes the file is populated.  The newest allsky-portal doesn't use the file anymore.
If you set the CAMERA variable in config/config.sh using `nano` you can run allsky.sh once to populate the file.  You should the be able to use the WebUI.

ssagerian, 2021-10-13:
@linuxkidd @EricClaeys  The most recent instructions in the wiki install tips work, to the extend that the install scripts ran without error. When I opened the page for the first time it exited with a note indicating that CAMERA was not set and to manually update it. After manually updating the setting, and refreshing the browser the page came up correctly. Everything appears to be working correctly however I have a few more things to check.  autocam.sh is still zero length

ssagerian, 2021-10-14:
In order to close this out, I think it still needs a mention in the wiki install tips to edit the config/config.sh file to add the camera type, as mentioned in the previous comment.

EricClaeys, 2021-10-14:
@ssagerian Steve, thanks for the suggestion.  Check out the first section in "02 Installation Tips" of the Wiki.  If it's sufficient, please close this issue.

ssagerian, 2021-10-15:
comment in 02 is clear, thanks