AllskyTeam/allsky #637: Dark Frame question

maserowik, 2021-10-12:
So this is more of a question on the dark frame option.  Currently on the pi there is dark images from -13 to 31. the number go from -13 to 31 in one step settings i have the dark frame option select in the config.sh.
'''
# Set to "true" if you want to enable dark frame subtraction on your night-time images.
# You must first create dark frames - see the README.md file for instructions./

DARK_FRAME_SUBTRACTION="true"
'''
I read the README.md but still a confused but it. 
So since I have 46 dark images the software will try all 46 and see which one is the best are removing hot pixels.
Mike

EricClaeys, 2021-10-12:
@maserowik No, the software will get the current sensor temperature from the `temperature.txt` file then find the closes file in darks/* to that temperature.  For example, current temp in temperature.txt is 21.  You have 3 darks:  17.jpg, 20.jpg, and 23.jpg.  The software will first look for an exact match with 21.jpg, but since it's not found it will pick 20.jpg because it's only 1 degree off from the current temp of 21.
It never looks in the files - it simply looks at the names of the files.  Unless something is really weird with the camera, or your darks are pretty old, the closest dark frame will give the best results.  Assuming the dark and the light have the same settings, like exposure time, binning, gain, etc.  And they should, since when darks are initially taken they use the max night exposure and all the other night settings.

Does this answer your question?

FYI, the Pi you're letting me use has the fixed version of darkCapture.sh and darkSubtract.sh that correctly implements this.

techd51, 2021-10-12:
Here are examples of the files I'm comparing, when I open them both up in Photoshop and create layers and compare the two I do see some of the hot pixels being removed but not all.
![43](https://user-images.githubusercontent.com/92370510/136946861-f118cd0c-29ee-4b61-bd03-076b6dd978ec.jpg
[image]
)

maserowik, 2021-10-12:
> @maserowik No, the software will get the current sensor temperature from the `temperature.txt` file then find the closes file in darks/* to that temperature. For example, current temp in [...]

Thanks  I have a better understanding on how the darks work with the program.  

Mike

techd51, 2021-10-21:
I know the issues I was having have been resolved.