AllskyTeam/allsky #643: Website not tidy as All sky Admin Panel

shiner501, 2021-10-12:
http://skycam.zapto.org/allsky/

I set to keep only last two days in config.  This is reflected in Admin Panel.  Yet on website above I have junk icons from long deleted videos...... How do I clean up this please?  This make GUI look messy.

Thanks, Steve

maserowik, 2021-10-12:
@shiner501 
I believe you have to manually delete the files on you hosted side.  The gui only takes care of files local to you pi.

shiner501, 2021-10-12:
Hi maserowik, the "site" is also hosted on the RPi - the one Thomas made.  When you connect to my URL above it got through my router and onto http server on the RPi.  So is this still a "site" and needs to be manually deleted?  The "time to keep" only applies to inside allskycam application?

shiner501, 2021-10-12:
Confirmed.  You have to go into /var/www/html/allsky/keograms, /var/www/html/allsky/startrails, /var/www/html/allsky/videos and then delete the older ones.  i.e. the automation in the allsky application does not also delete them in the allsky website too.

I will set up a cron that runs at midday to delete all files in these directories older than the number of days I configure in the allsky application.

EricClaeys, 2021-10-13:
@shiner501 I just added issue #644 ("NIGHTS_TO_KEEP should clean up allsky-website") to clean up old files in /var/www/html/allsky.
You are correct that NIGHTS_TO_KEEP currently only applies to directories in ~/allsky/images.
When I implement the ability to clean up /var/www/html/allsky I'll probably add separate variables for the website so you can specify a different length of time.  The website uses a lot less space since it only has up to 6 files per day rather than thousands.

Automatically cleaning up a remote web site (like @maserowik has) would be really cool.  I'll have to see how difficult that will be.  At a minimum I'll add @maserowik's comments to the Wiki.

rabit9, 2021-10-13:
what do you have to set to see the current live image on the webserve. I always say Loading!