AllskyTeam/allsky #650: exposures past 1 sec fail with ASI_ERROR_TIMEOUT

EricClaeys, 2021-10-14:
@maphilli14 is having problems with exposures greater than 1 second producing ASI_ERROR_TIMEOUTs.  This also happens with oacapture, which points to a hardware problem, not a problem with AllSky.

See also the end of [https://github.com/thomasjacquin/allsky/issues/417](url).

maphilli14, 2021-10-17:
It does seem to work on windows - https://youtu.be/IN2qUpWdckc

Also recorded in FC briefly:

[image]

In the past I had used the SDK but was never proficient at installing them.  Please advise!

maphilli14, 2021-10-19:
Also worth noting the oacapture messages as seen

https://user-images.githubusercontent.com/6034105/136716511-81053340-1d2a-4f9b-a076-ea8ebe759c8a.png

Transcribed to
`
ASIGetExpStatus returned status 3`

EricClaeys, 2021-10-20:
@maphilli14, ASIGetExpStatus doesn't return error 3 (ASI_ERROR_INVALID_CONTROL_TYPE, which means the camera was asked to do something it can't, like gamma, which most cameras don't support), so that might be a bug in oacapture.  The message above about "AutoExpMaxGain" are error 3.

Anyway, it works in Windows, and works in FireCapture (on the Pi, or Windows?), but does not work in oacapture and Allsky.   And on Allsky you can successfully take exposures < 1 second but not over 1 second, using either the version 0.8 exposure or the 0.7.  Is that correct?

EricClaeys, 2021-10-21:
@maphilli14,
This camera USED to work on **this** Pi, right?  Did anything change with the Pi?  Do you have access to a different one you can try?

If the camera did used to work on this same Pi, the only thing I can suggest is to try it with this version of the capture program, which is the version 0.7 one that knows about the newer options.  If THIS also doesn't work, it could be a problem with the Pi itself.  Perhaps the USB bus isn't getting enough power?
[capture0.7with0.8arguments.zip](https://github.com/thomasjacquin/allsky/files/7293039/capture0.7with0.8arguments.zip)

maphilli14, 2021-11-06:
Just getting back to trying this again, it fails as follows:

```
capture                         capture0.7with0.8arguments.zip
capture0.7with0.8arguments.cpp  capture.cpp
pi@allsky:~/allsky $ cp NEW/capture0.7with0.8arguments.cpp ./capture.cpp
pi@allsky:~/allsky $ make capture
make[1]: Entering directory '/home/pi/allsky/src'
2021-11-05 22:06:21 Building capture program...
2021-11-05 22:08:18 Done.
make[1]: Leaving directory '/home/pi/allsky/src'
make[1]: Entering directory '/home/pi/allsky/config_repo'
2021-11-05 22:08:18 nothing to do for capture
make[1]: Leaving directory '/home/pi/allsky/config_repo'
make[1]: Entering directory '/home/pi/allsky/notification_images'
2021-11-05 22:08:18 nothing to do for capture...
make[1]: Leaving directory '/home/pi/allsky/notification_images'
make[1]: Entering directory '/home/pi/allsky/scripts'
2021-11-05 22:08:18 nothing to do for capture
make[1]: Leaving directory '/home/pi/allsky/scripts'

sudo service allsky stop ; sleep 5 ;  sudo service allsky start
pi@allsky:~/allsky $ tail /var/log/allsky.log
Nov  5 22:09:23 allsky allsky.sh[3556]:   > Sleeping from auto exposure: 200 ms (0 sec)
Nov  5 22:09:23 allsky allsky.sh[3556]: STARTING EXPOSURE at: 2021-11-05 22:09:23
Nov  5 22:09:27 allsky allsky.sh[3556]:      ***** Stopping AllSky *****
Nov  5 22:09:46 allsky allsky.sh[9016]:      ***** Starting AllSky *****
Nov  5 22:09:46 allsky allsky.sh[9016]: CAMERA: ZWO
Nov  5 22:09:46 allsky allsky.sh[9016]: CAMERA_SETTINGS: /etc/raspap/settings_ZWO.json
Nov  5 22:09:49 allsky allsky.sh[9016]: Starting allsky camera...
Nov  5 22:09:51 allsky allsky.sh[9016]: /home/pi/allsky/allsky.sh: line 166: /home/pi/allsky/capture: No such file or directory
Nov  5 22:09:51 allsky allsky.sh[9016]: 'capture' exited with RETCODE=127
Nov  5 22:09:51 allsky allsky.sh[9016]: *** After fixing, restart the allsky service. ***

```

maphilli14, 2021-11-06:
Some things I tried ....

```

pi@allsky:~ $ cd allsky/
pi@allsky:~/allsky $ mkdir NEW
pi@allsky:~/allsky $ cd NEW/
pi@allsky:~/allsky/NEW $ wget https://github.com/thomasjacquin/allsky/files/7293039/capture0.7with0.8arguments.zip
--2021-11-05 22:04:59--  https://github.com/thomasjacquin/allsky/files/7293039/capture0.7with0.8arguments.zip
Resolving github.com (github.com)... 140.82.112.4
Connecting to github.com (github.com)|140.82.112.4|:443... connected.
HTTP request sent, awaiting response... 302 Found
Location: https://github-repository-files.githubusercontent.com/71661619/7293039?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=AKIAIWNJYAX4CSVEH53A%2F20211106%2Fus-east-1%2Fs3%2Faws4_request&X-Amz-Date=20211106T020500Z&X-Amz-Expires=300&X-Amz-Signature=946b98c0be0e16c5501970c64870f4a9f6e49844e4fc4930866665b7596f1382&X-Amz-SignedHeaders=host&actor_id=0&key_id=0&repo_id=71661619&response-content-disposition=attachment%3Bfilename%3Dcapture0.7with0.8arguments.zip&response-content-type=application%2Fx-zip-compressed [following]
--2021-11-05 22:05:00--  https://github-repository-files.githubusercontent.com/71661619/7293039?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=AKIAIWNJYAX4CSVEH53A%2F20211106%2Fus-east-1%2Fs3%2Faws4_request&X-Amz-Date=20211106T020500Z&X-Amz-Expires=300&X-Amz-Signature=946b98c0be0e16c5501970c64870f4a9f6e49844e4fc4930866665b7596f1382&X-Amz-SignedHeaders=host&actor_id=0&key_id=0&repo_id=71661619&response-content-disposition=attachment%3Bfilename%3Dcapture0.7with0.8arguments.zip&response-content-type=application%2Fx-zip-compressed
Resolving github-repository-files.githubusercontent.com (github-repository-files.githubusercontent.com)... 185.199.109.154, 185.199.110.154, 185.199.111.154, ...
Connecting to github-repository-files.githubusercontent.com (github-repository-files.githubusercontent.com)|185.199.109.154|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 12773 (12K) [application/x-zip-compressed]
Saving to: ‘capture0.7with0.8arguments.zip’

capture0.7with0.8arguments.zi 100%[==============================================>]  12.47K  --.-KB/s    in 0.004s

2021-11-05 22:05:00 (2.76 MB/s) - ‘capture0.7with0.8arguments.zip’ saved [12773/12773]

pi@allsky:~/allsky/NEW $ ls
capture0.7with0.8arguments.zip
pi@allsky:~/allsky/NEW $ unzip capture0.7with0.8arguments.zi
unzip:  cannot find or open capture0.7with0.8arguments.zi, capture0.7with0.8arguments.zi.zip or capture0.7with0.8arguments.zi.ZIP.
pi@allsky:~/allsky/NEW $ unzip capture0.7with0.8arguments.zip
Archive:  capture0.7with0.8arguments.zip
  inflating: capture0.7with0.8arguments.cpp
pi@allsky:~/allsky/NEW $ ls
capture0.7with0.8arguments.cpp  capture0.7with0.8arguments.zip
pi@allsky:~/allsky/NEW $ cd ..
pi@allsky:~/allsky $ mv capture^C
pi@allsky:~/allsky $ ls capture*
capture  capture.cpp  capture_RPiHQ
pi@allsky:~/allsky $ ls -la
total 4468
drwxr-xr-x 14 pi   pi      4096 Nov  5 22:04 .
drwxr-xr-x 19 pi   pi      4096 Oct 29 06:17 ..
-rwxr-xr-x  1 pi   pi      5935 Oct  9 12:39 allsky.sh
-rwxr-xr-x  1 pi   pi   2155012 Oct  9 12:44 capture
-rwxr-xr-x  1 pi   pi     57358 Oct  9 12:43 capture.cpp
-rwxr-xr-x  1 pi   pi    524372 Oct  9 12:44 capture_RPiHQ
-rw-r--r--  1 pi   pi      1560 Oct  9 12:39 .clang-format
drwxr-xr-x  2 pi   pi      4096 Oct  9 15:26 config
drwxr-xr-x  2 pi   pi      4096 Oct  9 12:44 config_repo
drwxr-xr-x  9 pi   pi      4096 Oct  9 12:39 .git
-rw-r--r--  1 pi   pi       158 Oct  9 12:39 .gitattributes
-rw-r--r--  1 pi   pi       405 Oct  9 12:39 .gitignore
-rw-r--r--  1 pi   pi        97 Oct  9 12:39 .gitmodules
drwxr-xr-x  2 pi   pi      4096 Oct  9 12:53 gui
-rw-r--r--  1 pi   pi         0 Nov  5 22:05 image.jpg
drwxr-xr-x  4 pi   pi      4096 Nov  5 18:41 images
-rwxr-xr-x  1 pi   pi      1351 Oct  9 12:39 install.sh
-rwxr-xr-x  1 pi   pi    648664 Oct  9 12:44 keogram
-rw-r--r--  1 pi   pi      1070 Oct  9 12:39 LICENSE
-rw-r--r--  1 pi   pi    685777 Nov  5 22:05 liveview-image.jpg
-rw-r--r--  1 pi   pi      1291 Oct  9 12:39 Makefile
drwxr-xr-x  2 pi   pi      4096 Nov  5 22:05 NEW
drwxr-xr-x  2 pi   pi      4096 Oct  9 12:39 notification_images
-rw-r--r--  1 pi   pi     32172 Oct  9 12:39 README.md
drwxr-xr-x  2 pi   pi      4096 Oct 20 08:47 scripts
drwxr-xr-x  5 pi   pi      4096 Oct  9 21:41 src
-rwxr-xr-x  1 pi   pi    344864 Oct  9 12:44 startrails
drwxr-xr-x  2 root root    4096 Nov  5 22:05 TEMP
-rw-r--r--  1 pi   pi         5 Nov  5 22:05 temperature.txt
drwxr-xr-x  2 pi   pi      4096 Oct 29 11:59 tmp
-rwxr-xr-x  1 pi   pi      1019 Oct  9 12:39 uninstall.sh
[... 114 lines cut ...]
config         images       liveview-image.jpg  README.md            TEMP        variables.sh
pi@allsky:~/allsky $ ls src
capture      capture_RPiHQ      include  keogram.cpp  Makefile    startrails.cpp  sunwait-src
capture.cpp  capture_RPiHQ.cpp  keogram  lib          startrails  sunwait
pi@allsky:~/allsky $ cp src/capture .
pi@allsky:~/allsky $ ls
allsky.sh      config       images      liveview-image.jpg   README.md   TEMP             variables.sh
capture        config_repo  install.sh  Makefile             scripts     temperature.txt  website
capture.cpp    gui          keogram     NEW                  src         tmp
capture_RPiHQ  image.jpg    LICENSE     notification_images  startrails  uninstall.sh
pi@allsky:~/allsky $ sudo service allsky stop ; sleep 5 ;  sudo service allsky start
pi@allsky:~/allsky $ tail /var/log/allsky.log
[... cut ...]
Nov  5 22:15:01 allsky allsky.sh[9390]: Auto Exposure value: 1000 ms
Nov  5 22:15:01 allsky allsky.sh[9390]:   > Sleeping from auto exposure: 200 ms (0 sec)
Nov  5 22:15:01 allsky allsky.sh[9390]: /home/pi/allsky/allsky.sh: line 166:  9485 Segmentation fault      "${ALLSKY_HOME}/${CAPTURE}" "${ARGUMENTS[@]}"
Nov  5 22:15:01 allsky allsky.sh[9390]: 'capture' exited with RETCODE=139
Nov  5 22:15:02 allsky allsky.sh[9390]: *** After fixing, restart the allsky service. ***
pi@allsky:~/allsky $

```

EricClaeys, 2021-11-08:
@maphilli14, you want to copy the capture.cpp file to the `allsky/src` directory, and run "make capture" from within that directory, then run "make install".

The `WARNING: currently saving an image; can't save new one.` message is one I added in version 0.8.  It usually indicates your delay isn't long enough, so the system is still writing one image when the next one is ready to write.

Before trying the capture0.7with0.8arguments.zip file,  try the attached file instead.  If that doesn't help, then try capture0.7with0.8arguments.zip file.

[capture.cpp.txt](https://github.com/thomasjacquin/allsky/files/7493503/capture.cpp.txt)

maphilli14, 2021-11-10:
With capture.cpp.txt compiled...

```

Nov  9 17:59:06 allsky allsky.sh[4846]:   > Sleeping: 251.0 ms
Nov  9 17:59:06 allsky allsky.sh[4846]:   > Saving NIGHT image 'image.jpg'
Nov  9 17:59:07 allsky allsky.sh[4846]: STARTING EXPOSURE at: 2021-11-09 17:59:07   @ 948,502 us
Nov  9 17:59:07 allsky allsky.sh[4846]:   > Saving NIGHT image 'image.jpg'
Nov  9 17:59:07 allsky allsky.sh[4846]:   > Sleeping: 251.0 ms
Nov  9 17:59:07 allsky allsky.sh[4846]: scripts/saveImageNight.sh: line 91: 12381 Aborted                 convert "$IMAGE_TO_USE" -resize "${THUMBNAIL_SIZE_X}x${THUMBNAIL_SIZE_Y}" "$THUMBNAILS_DI
R/$SAVED_FILE"
Nov  9 17:59:07 allsky allsky.sh[4846]: *** saveImageNight.sh: WARNING: THUMBNAIL resize failed with RET=134; continuing.
Nov  9 17:59:08 allsky allsky.sh[4846]: convert-im6.q16: Premature end of JPEG file `image.jpg' @ warning/jpeg.c/JPEGWarningHandler/387.
Nov  9 17:59:08 allsky allsky.sh[4846]: convert-im6.q16: Corrupt JPEG data: premature end of data segment `image.jpg' @ warning/jpeg.c/JPEGWarningHandler/387.
Nov  9 17:59:08 allsky allsky.sh[4846]: STARTING EXPOSURE at: 2021-11-09 17:59:08   @ 948,502 us
Nov  9 17:59:08 allsky allsky.sh[4846]:   > Saving NIGHT image 'image.jpg'
Nov  9 17:59:08 allsky allsky.sh[4846]:   > Sleeping 200.0 ms from auto exposure
Nov  9 17:59:09 allsky allsky.sh[4846]: STARTING EXPOSURE at: 2021-11-09 17:59:09   @ 1,024,382 us
Nov  9 17:59:09 allsky allsky.sh[4846]: convert-im6.q16: Premature end of JPEG file `image.jpg' @ warning/jpeg.c/JPEGWarningHandler/387.
Nov  9 17:59:09 allsky allsky.sh[4846]: convert-im6.q16: Corrupt JPEG data: premature end of data segment `image.jpg' @ warning/jpeg.c/JPEGWarningHandler/387.
Nov  9 17:59:10 allsky allsky.sh[4846]: convert-im6.q16: Premature end of JPEG file `image.jpg' @ warning/jpeg.c/JPEGWarningHandler/387.
Nov  9 17:59:10 allsky allsky.sh[4846]: convert-im6.q16: Corrupt JPEG data: premature end of data segment `image.jpg' @ warning/jpeg.c/JPEGWarningHandler/387.
Nov  9 17:59:16 allsky allsky.sh[4846]:   > ERROR: Failed getting image: ASI_ERROR_TIMEOUT #2 (with 0.8 exposure = NO) (11)
Nov  9 17:59:18 allsky allsky.sh[4846]: STARTING EXPOSURE at: 2021-11-09 17:59:18   @ 1,024,382 us
Nov  9 17:59:25 allsky allsky.sh[4846]:   > ERROR: Failed getting image: ASI_ERROR_TIMEOUT #3 (with 0.8 exposure = NO) (11)
Nov  9 17:59:27 allsky allsky.sh[4846]: STARTING EXPOSURE at: 2021-11-09 17:59:27   @ 1,024,382 us
Nov  9 17:59:34 allsky allsky.sh[4846]:   > ERROR: Failed getting image: ASI_ERROR_TIMEOUT #4 (with 0.8 exposure = NO) (11)
Nov  9 17:59:36 allsky allsky.sh[4846]: STARTING EXPOSURE at: 2021-11-09 17:59:36   @ 1,024,382 us
Nov  9 17:59:43 allsky allsky.sh[4846]:   > ERROR: Failed getting image: ASI_ERROR_TIMEOUT #5 (with 0.8 exposure = NO) (11)
Nov  9 17:59:45 allsky allsky.sh[4846]: STARTING EXPOSURE at: 2021-11-09 17:59:45   @ 1,024,382 us
Nov  9 17:59:52 allsky allsky.sh[4846]:   > ERROR: Failed getting image: ASI_ERROR_TIMEOUT #6 (with 0.8 exposure = NO) (11)
Nov  9 17:59:54 allsky allsky.sh[4846]: Maximum number of consecutive errors of 5 reached; exiting...
Nov  9 17:59:56 allsky allsky.sh[4846]:   > Saving NIGHT image 'image.jpg'
Nov  9 17:59:59 allsky allsky.sh[4846]:      ***** Stopping AllSky *****
Nov  9 18:00:04 allsky allsky.sh[4846]: 'capture' exited with RETCODE=2

```

maphilli14, 2021-11-10:
the 0.7 with 08 args does this...

```

Nov  9 21:36:10 allsky allsky.sh[6527]:   > Sleeping from auto exposure: 200 ms (0 sec)
Nov  9 21:36:11 allsky allsky.sh[6527]: Exposure value: 1,000,000 µs (1,000.0 ms)
Nov  9 21:36:11 allsky allsky.sh[6527]: 20211109 21:36:11
Nov  9 21:36:11 allsky allsky.sh[6527]: Auto Exposure value: 1000 ms
Nov  9 21:36:11 allsky allsky.sh[6527]:   > Sleeping from auto exposure: 200 ms (0 sec)
Nov  9 21:36:13 allsky allsky.sh[6527]: Exposure value: 1,000,000 µs (1,000.0 ms)
Nov  9 21:36:13 allsky allsky.sh[6527]: 20211109 21:36:13
Nov  9 21:36:13 allsky allsky.sh[6527]: Auto Exposure value: 1000 ms
Nov  9 21:36:13 allsky allsky.sh[6527]:   > Sleeping from auto exposure: 200 ms (0 sec)
Nov  9 21:36:13 allsky allsky.sh[6527]: /home/pi/allsky/allsky.sh: line 166:  6628 Segmentation fault      "${ALLSKY_
HOME}/${CAPTURE}" "${ARGUMENTS[@]}"
Nov  9 21:36:13 allsky allsky.sh[6527]: 'capture' exited with RETCODE=139
Nov  9 21:36:14 allsky allsky.sh[6527]: *** After fixing, restart the allsky service. ***

```

maphilli14, 2021-11-10:
FYI this is a mono camera, LMK if the settings should be changed at all...

EricClaeys, 2021-11-10:
@maphilli14, it shouldn't matter that it's a mono camera.  Try running `vcgencmd get_throttled` - it will give a code that states whether or not your Pi has low voltage, which could be the issue with your camera.

Can you attach (don't copy/paste) the complete /var/log/allsky.log file from the 0.7 with 08 args run?  You can do this:
~~~
# stop the service
sudo truncate -s 0 /var/log/allsky.log
# start the service
# let it run until you have issues, then attach the file.
~~~
It's very interesting that you go the "Segmentation fault" with the version 0.7 file.  Someone else is getting that with the newest version 0.8 file, and I've never seen it before.

EricClaeys, 2021-11-10:
@maphilli14, another thing to try if you can.
Stop the service.
In your home directory is a subdirectory called `.ZWO` (with a leading period).  It's hidden by default so you'll need to show the hidden files by pressing CTRL-H when in the explorer window.
Enter the .ZWO directory and open `ASIconfig.xml`, then change DebugPrint’s value from 00 to 01.
Then restart the service.  This will add a bunch of additional debugging statements from the camera driver to /var/log/allsky.log.  I'm hoping they'll give us an idea why you're getting the ASI_ERROR_TIMEOUTs.

maphilli14, 2021-11-11:
I will try some of the other items  you suggested but here's what the capture0.7with0.8arguments.zip is doing for me tonight....
[allsky-20211110.log](https://github.com/thomasjacquin/allsky/files/7517662/allsky-20211110.log)
.

EricClaeys, 2021-11-11:
@maphilli14, 50005 means throttling and low voltage are currently happening, and happened in the past.
If you move the attached file (without the .txt on the end) to /var/www/html/includes, then go to the "System" link in the WebUI you'll see the throttle status.

If you are getting under-voltage, [this power unit](https://lifepo4wered.com/lifepo4wered-pi.html) may help.  Or use a more powerful power supply or power cables with thicker wires.

[system.php.txt](https://github.com/thomasjacquin/allsky/files/7521968/system.php.txt)

maphilli14, 2021-11-14:
I'm happy to play with the hardware a bit more but it's confusing to me why the same hardware used to work and now does not.  could the camera be at any fault?
`
Throttle Status | Not able to get throttle status:sudo: no tty present and no askpass program specified
`

EricClaeys, 2021-11-14:
@maphilli14, to get the WebUI to correctly display the throttle status, download the latest `includes/system.php` from `allsky-portal` and the latest `gui/sudoers` from `allsky`.  Then:
`sudo cp gui/sudoers /etc/sudoers.d/allsky`

I don't know why things used to work but don't now.  One possibility is they didn't always work, but the software didn't tell you and the problems weren'tusually bad enough for most people to notice.  The 0.8 version added a LOT of error checking so problems are more noticable now.
It's also possible 0.8 added a lot of bugs, some of which are still present.
It would be interesting to revert to 0.7 and see if you are still having issues.  I know that's possible but don't know how.

EricClaeys, 2021-11-14:
@maphilli14 Are you sure you were running the version 7 with 8 arguments code when you got the Segmentation fault?  The was a bug introduced in 0.8 that caused a segmentation fault, but it wasn't in 0.7.
Could you do me a favor and set Debug Level to 4 and run with that version for a minute, then attach the log file?
Thanks

maphilli14, 2021-11-14:
I pulled the setup off the roof and put it into the lab.  Yanking the uSD from the Pi3 and putting into a Pi4 with proper power supply on the wall, I find some odd behavior.  The issues of no exp >1s are consistent with the Pi4 and a new USB3 cable on the ASI174mm.  The issue goes away after replacing with an ASI290.  I'll test more but I'm thinking it's either drivers/config or hardware/camera issues.

EricClaeys, 2021-11-15:
@maphilli14, pretty strange.  The ASI174 works fine in Windows but not on either Pi using oacapture OR AllSky, which implies it's not a Pi hardware issue (although it could be).  It's very possible an issue with ZWO drivers.

maphilli14, 2021-11-15:
That is my guess as well, this is long and extensive but likely no resolution since June 2021?   https://bbs.astronomy-imaging-camera.com/d/12315-libasicamera2-for-aarch64/58  I'll keep searching but that....
`
Exposure Status: 3`

sounds similar to what I had as well.  I wonder if some change in Buster made it malfunction beyond >1sec?

maphilli14, 2021-11-15:
https://github.com/OpenPHDGuiding/phd2/issues/936
what's all this mean?

maphilli14, 2021-11-15:
Just catching a daytime crash too....

```
Nov 15 09:27:55 allsky allsky.sh[21527]: [WorkingFunc]: drop frames:66
Nov 15 09:27:55 allsky allsky.sh[21527]: [WorkingFunc]: get len:0x0 drop:67  frametime:19520us waittime:89ms
Nov 15 09:27:55 allsky allsky.sh[21527]: [WorkingFunc]: nothing get count:1
Nov 15 09:27:55 allsky allsky.sh[21527]: [WorkingFunc]: head:0x1a1b COUNT:6426 tail:0x1b1c count:7453
Nov 15 09:27:55 allsky allsky.sh[21527]: [WorkingFunc]: drop frames:68
Nov 15 09:27:55 allsky allsky.sh[21527]: [WorkingFunc]: get len:0x0 drop:69  frametime:19520us waittime:89ms
Nov 15 09:27:55 allsky allsky.sh[21527]: [WorkingFunc]: nothing get count:1
Nov 15 09:27:56 allsky allsky.sh[21527]: [WorkingFunc]: get len:0x1a4084 drop:70  frametime:19520us waittime:89ms
Nov 15 09:27:56 allsky allsky.sh[21527]: [WorkingFunc]: get len:0x1a4084 drop:70  frametime:19520 waittime:89
Nov 15 09:27:56 allsky allsky.sh[21527]: [WorkingFunc]: drop frames:70
Nov 15 09:27:56 allsky allsky.sh[21527]: Maximum number of consecutive errors of 5 reached; exiting...
Nov 15 09:27:56 allsky allsky.sh[21527]: [WorkingFunc]: get len:0x0 drop:71  frametime:19520us waittime:89ms
Nov 15 09:27:56 allsky allsky.sh[21527]: [WorkingFunc]: nothing get count:1
Nov 15 09:27:56 allsky allsky.sh[21527]: [SaveSetting]: save REG
Nov 15 09:27:56 allsky allsky.sh[21527]: [StopCapture]: stop capture
Nov 15 09:27:56 allsky allsky.sh[21527]: [StopCapture]: >> wait Working thread exit
Nov 15 09:27:56 allsky allsky.sh[21527]: [Stop]: WaitForSingleObject >
Nov 15 09:27:56 allsky allsky.sh[21527]: [WorkingFunc]: head:0x1c1c COUNT:7197 tail:0x909 count:2313
Nov 15 09:27:56 allsky allsky.sh[21527]: [WorkingFunc]: drop frames:72
Nov 15 09:27:56 allsky allsky.sh[21527]: [ResetCirBuff]: Rst
Nov 15 09:27:56 allsky allsky.sh[21527]: [releaseAsyncXfer]: free transfer!
Nov 15 09:27:56 allsky allsky.sh[21527]: [Stop]: WaitForSingleObject >
Nov 15 09:27:56 allsky allsky.sh[21527]: [SetGainExpFunc]: Thread SetGainExpFunc Exit!
Nov 15 09:27:56 allsky allsky.sh[21527]: [Stop]: WaitForSingleObject <
Nov 15 09:27:56 allsky allsky.sh[21527]: [WorkingFunc]: working thread exit!
Nov 15 09:27:56 allsky allsky.sh[21527]: [Stop]: WaitForSingleObject <
Nov 15 09:27:56 allsky allsky.sh[21527]: [StopCapture]: << wait Working thread exit
Nov 15 09:27:59 allsky allsky.sh[21527]:      ***** Stopping AllSky *****
Nov 15 09:27:59 allsky allsky.sh[21527]: destructor called!
Nov 15 09:27:59 allsky allsky.sh[21527]: 'capture' exited with RETCODE=2
```

EricClaeys, 2021-11-16:
@maphilli14, Would you please attach the version of capture.cpp you are using, that's giving the errors?  I want to make a change and have you try it.

The OpenPHDGuiding/phd2#936 stuff is an issue in PHD2, not in AllSky.  They are calling some ZWO functions but not checking the return status.  I added a lot of that status checking in version 0.8.

I'm reading through the long ZWO post to see if there's anything that might help us.

maphilli14, 2021-11-16:
I sure can!  I did a fresh git clone when I switched to the Pi4....
[capture.cpp.txt](https://github.com/thomasjacquin/allsky/files/7543087/capture.cpp.txt)

EricClaeys, 2021-11-16:
@maphilli14 thanks Mike.  Would you please try with this file.  All I did was change the timeout of a capture to "wait forever", and changed the version string to include "timeout=-1" so we know which version you are running.
Please stop the service, run `sudo truncate -s 0 /var/log/allsky.log`, then compile this and copy allsky/src/capture to allsky.  Then post a log.

[Mike's capture.cpp.txt](https://github.com/thomasjacquin/allsky/files/7543124/Mike.s.capture.cpp.txt)

maphilli14, 2021-11-16:
seems to fail even at <=1sec still, but all day it was stable on oacapture with same hardware!  :(
[allsky-log-20211115.txt](https://github.com/thomasjacquin/allsky/files/7543479/allsky-log-20211115.txt)

maphilli14, 2021-11-16:
what are these?
```

  D:\E-Delete\allsky-log-20211115.txt (5 hits)
	Line 251: Nov 15 23:47:36 allsky allsky.sh[6448]: [WorkingFunc]: BAD delta time:7769 pkg:800
	Line 269: Nov 15 23:47:39 allsky allsky.sh[6448]: [WorkingFunc]: BAD delta time:2627 pkg:800
	Line 288: Nov 15 23:47:42 allsky allsky.sh[6448]: [WorkingFunc]: BAD delta time:2635 pkg:800
	Line 308: Nov 15 23:47:45 allsky allsky.sh[6448]: [WorkingFunc]: BAD delta time:2841 pkg:800
	Line 327: Nov 15 23:47:48 allsky allsky.sh[6448]: [WorkingFunc]: BAD delta time:3075 pkg:800
```

maphilli14, 2021-11-16:
```

Linux allsky 5.10.63-v7l+ #1457 SMP Tue Sep 28 11:26:14 BST 2021 armv7l

The programs included with the Debian GNU/Linux system are free software;
the exact distribution terms for each program are described in the
individual files in /usr/share/doc/*/copyright.

Debian GNU/Linux comes with ABSOLUTELY NO WARRANTY, to the extent
permitted by applicable law.
Last login: Mon Nov 15 09:24:00 2021 from 192.168.43.33
pi@allsky:~ $ sudo service allsky stop
pi@allsky:~ $ cd allsky/src/
pi@allsky:~/allsky/src $ rm capture
pi@allsky:~/allsky/src $ mv M
Makefile                Mike.s.capture.cpp.txt
pi@allsky:~/allsky/src $ mv Mike.s.capture.cpp.txt capture.txt
pi@allsky:~/allsky/src $ rm ../capture
pi@allsky:~/allsky/src $ cd ..
pi@allsky:~/allsky $ make capture
make[1]: Entering directory '/home/pi/allsky/src'
2021-11-15 23:44:24 Building capture program...
^Cmake[1]: *** [Makefile:122: capture] Interrupt
make: *** [Makefile:24: capture] Interrupt

pi@allsky:~/allsky $ ^C
pi@allsky:~/allsky $ ^C
pi@allsky:~/allsky $ ^C
pi@allsky:~/allsky $ sudo truncate -s 0 /var/log/allsky.log
pi@allsky:~/allsky $ make capture
make[1]: Entering directory '/home/pi/allsky/src'
2021-11-15 23:44:43 Building capture program...
2021-11-15 23:44:59 Done.
make[1]: Leaving directory '/home/pi/allsky/src'
make[1]: Entering directory '/home/pi/allsky/config_repo'
2021-11-15 23:44:59 nothing to do for capture
make[1]: Leaving directory '/home/pi/allsky/config_repo'
make[1]: Entering directory '/home/pi/allsky/notification_images'
2021-11-15 23:44:59 nothing to do for capture...
make[1]: Leaving directory '/home/pi/allsky/notification_images'
make[1]: Entering directory '/home/pi/allsky/scripts'
2021-11-15 23:44:59 nothing to do for capture
make[1]: Leaving directory '/home/pi/allsky/scripts'
pi@allsky:~/allsky $ ls
allsky.sh      gui         keogram             notification_images  startrails       variables.sh
capture_RPiHQ  image.jpg   LICENSE             README.md            temperature.txt  website
config         images      liveview-image.jpg  scripts              tmp
config_repo    install.sh  Makefile            src                  uninstall.sh
pi@allsky:~/allsky $ ls src/
capture      capture_RPiHQ      capture.txt  keogram      lib       mode_RPiHQ_mean.cpp  startrails.cpp  sunwait-src
capture.cpp  capture_RPiHQ.cpp  include      keogram.cpp  Makefile  startrails           sunwait
pi@allsky:~/allsky $ cp src/capture .
pi@allsky:~/allsky $ cat /var/log/allsky.log
pi@allsky:~/allsky $ sudo service allsky start

```

EricClaeys, 2021-11-16:
@maphilli14, Hi, the log file you recently posted isn't from `Mike's capture.cpp.txt`; it's from the version you had been running.  After you do a `cd src; make capture` you need to `cp capture ..` then restart the service.  Sorry I forgot to mention that.  So the commands would be:
~~~
# stop the service
sudo truncate -s 0 /var/log/allsky.log
cd allsky/src
# Copy the new capture.cpp program here and rename to "capture.cpp"
make capture
cp capture ..
# restart the service
~~~
If you run "make capture" in the "allsky" directory it tries to make everything under the sun.
The version displayed near the top of the log file should be
`*** Allsky Camera Software v0.8.1b, timeout=-1 |  2021 ***`

I don't know what the `BAD delta time` messages are for.  It's not clear what most of the entries in ZWO's log file are...

maphilli14, 2021-11-16:
I feel like a n00b for compiling, I've been doing it since slackware 2.4, but still don't get it all right!  😔
I think I did it right but don't see timeout=-1 in the logs...
it seemed to do a daytime capture for a bit longer than normal, but did eventually die.  
I feel like there something underlying for libraries or drivers that is causing a difference between the 174 and the 290 on buster.  Both cameras are fine on Win10 with FC for exp > 1sec, but exhibit different behavior on the Pi 😭

Thanks for all the hard work making allsky better, hopefully for all not just me!

[allsky-cli-20211116.txt](https://github.com/thomasjacquin/allsky/files/7546929/allsky-cli-20211116.txt)
[allsky-log-20211116.txt](https://github.com/thomasjacquin/allsky/files/7546963/allsky-log-20211116.txt)

EricClaeys, 2021-11-16:
@maphilli14 Try `cd src; grep "Allsky Camera" capture.cpp` and see if it has the correct version.

If you do another run, please set the Debug Level to 4.  It was 0 in your last run which hides almost all the output I need.

Thanks.

maphilli14, 2021-11-16:
printf("%s *** Allsky Camera Software v0.8.1b |  2021 ***\n", KGRN);
and it seems pretty stable right now

maphilli14, 2021-11-16:
[allsky-log-20211116b.txt](https://github.com/thomasjacquin/allsky/files/7550400/allsky-log-20211116b.txt)

with failure

EricClaeys, 2021-11-17:
@maphilli14, The "grep" for the file you want will return
`printf("%s *** Allsky Camera Software v0.8.1b, timeout=-1 |  2021 ***\n", KGRN);`
I attached that file.
[maphilli14-capture.cpp.txt](https://github.com/thomasjacquin/allsky/files/7551283/maphilli14-capture.cpp.txt)

EricClaeys, 2021-11-17:
@maphilli14, if the file I just attached doesn't help, something you can try is to add `swiotlb=131072` to end of the line (NOT on a new line) in `/boot/cmdline.txt`, then reboot.

maphilli14, 2021-11-17:
I hope I didn't confuse myself with capture.cpp and capture.txt in the src directory... I think i have the right one this time...

[image]

maphilli14, 2021-11-17:
3hrs and stable for daytime!  🤞

EricClaeys, 2021-11-17:
@maphilli14 woo-hoo!  This is the best it's ever done, right?
Can you send me the log file?

maphilli14, 2021-11-17:
[allsky-log-20211117.txt](https://github.com/thomasjacquin/allsky/files/7557543/allsky-log-20211117.txt)

maphilli14, 2021-11-17:
[allsky-log-20211117b-nighterr.txt](https://github.com/thomasjacquin/allsky/files/7558631/allsky-log-20211117b-nighterr.txt)

now it just seems to stop new captures once it hits an error.  It doesn't error out, just holds the last known good image

IE froze on frame 18:28:45 and kept displaying that instead of any new ones and also didn't display the red stopped in liveview

EricClaeys, 2021-11-19:
@maphilli14, the change I made in the file was to have it wait forever for the camera to return an image.  This was on the off-chance that the ASI_ERROR_TIMEOUTs you were seeing were really timeouts.

What happened when you ran it is that all the sub-second pictures worked fine, but once it tried a picture > 1 second the camera never returned.  It went for about 1 hour 8 minutes, then I assume you stopped the service.
It looks like you restarted the service, and it was nighttime, and the exact same thing happened.
In both cases, the camera would display some log messages every second or two, then repeat.

Please try again but set the "Version 0.8 exposure" to OFF.  If it hangs, let it go a couple minutes then stop the service and send the log file.

Unless there's some setting that we're supposed to set before taking pictures that I'm not aware of, this looks like an issue with the ZWO driver or the camera itself.  Since it works on Windows, that would imply it's the ZWO Linux driver.

EricClaeys, 2021-11-19:
@maphilli14, try this version AFTER your current version with "Version 0.8 exposure" OFF.
The timeout in this newer version is back to normal

[capture.cpp.txt](https://github.com/thomasjacquin/allsky/files/7569737/capture.cpp.txt)

maphilli14, 2021-11-19:
Will do, I'm eclipse tired today, hope to try this weekend!  Thanks much @EricClaeys

maphilli14, 2021-11-22:
I had some initial good success <1s but it seems it might have died at some point... I am restarting and will try again tonight <1s and then later with >1s.... logs are 0bytes :(  stay tuned

maphilli14, 2021-11-23:
I retested last night with exp=999 after finding I had set the exp to 1999ms and it failed... woke up this am to find it's in a "Allsky software is not running" state.  With logs.  I'm tempted to redo the entire uSD with fresh raspbian and use my old methods of pulling down the SDK and testing with oacapture before any allsky git clones..
[allsky-log-20211123a.zip](https://github.com/thomasjacquin/allsky/files/7589273/allsky-log-20211123a.zip)
..

EricClaeys, 2021-11-24:
@maphilli14, it was in the "Allsky software is not running" state because as soon as daytime came, the "night max exposure" of 999 you set didn't apply so it quickly tried to take a picture over 1 second and died due to ASI_ERROR_TIMEOUTs.
Attached is a version of the camera_options_ZWO.json file that supports a daytime max exposure.
Copy it to /etc/raspap (will need sudo), then in the WebUI go to the Camera Settings link (may need to refresh your browser) and you'll see it.  You can set it to 999 also.
[camera_options_ZWO.json.txt](https://github.com/thomasjacquin/allsky/files/7594661/camera_options_ZWO.json.txt)

maphilli14, 2021-12-02:
I tried a fresh copy of raspbian and the SDK with oacapture, which didn't recognize the camera as it usually did for me in the past.  I grabbed the openastro libasi-xxx.deb and then oacapture did find the camera.  NO allsky git clones at all.  Same issue.  I think this is a libASI problem and not sure where to take it next, but likely not related to allsky per se.

maphilli14, 2021-12-02:
Trying the ASI forums - https://bbs.astronomy-imaging-camera.com/d/13593-asi174mm-stopped-working-on-pi34

EricClaeys, 2022-05-17:
@maphilli14   Are you still having this problem, or can we close the Issue?
Eric

maphilli14, 2022-05-17:
I do still have this issue but it seems to be related to Linux and not to allsky I think we're safe to close this

closed 2022-05-17 by EricClaeys

pdelorme, 2023-08-24:
@maphilli14 one year later, have you resolved this ?

maphilli14, 2023-08-24:
I've swapped to a QHY and indi-allsky we can close this.

Mike