AllskyTeam/allsky #652: How to find out what teim night capture starts?

robmorin, 2021-10-14:
Hey all... 

Still trying to figure out all the subtleties of this cool program, one of them being start and stop times of night capture.
I am assuming  allsky knows where I am by the  Latitude & Longitude i entered under camera settings-->Other settings.

But how does it decide when it should start and or stop capturing? Based on sunset & sunrise times of my area?
If so what if I have no internet where I plan to put this contraption?

Thanks..
Have a great day everyone!

bbillp, 2021-10-14:
> One other thing - you might want to add an I2C clock and/or a GPS so that you have the correct time without network, even after a reboot.

Those $20 USB GPS found on ebay are not the best, sometimes they lock on the satellites in seconds and other times it takes an hour to lock onto the satellites.   If you know how to connect it use a Garmin GPS puck or an old hand held GPS, older models are often serial data while the newer models are USB .  The serial data connection is not that hard to setup and OH8STN has very good GPS instructions on his web site.    

The alternative is easy enough, use the command line ‘date’ to set the Pi date and time.  

 Bill K6ACJ

EricClaeys, 2021-10-15:
@aleveilleVE2DTL Are you looking for sunrise/sunset or when the user has defined daytime and nighttime, which is what the Angle variable uses?  An Angle of 0 gives sunrise/sunset but most people use an angle of -6 or more to define when nighttime begins.

robmorin, 2021-10-15:
> @robmorin<https://github.com/robmorin> It does indeed use the latitude and longitude you set, as well as the Angle you set, which many people set to -6 degrees. So "nighttime" is defined as when the Sun is 6 degrees BELOW the horizon, and "daytime" is when the Sun is 6 degrees below the horizon.
> The software uses the "sunwait" command, passing it the latitude, longitude, angle, and an argument that says what to do with that information.
> I'll add something to the Wiki on this. When I'm done I'll ask you to review it, and if you agree it answered your question, you can then close the Issue. Sound reasonable?

Thanks for the the info! My angle is set at the default, - 9 I shall change it to - 6

Just got my fisheye in today, so set is coming along.
Have a great weekend!

Rob Morin
Montreal, Canada
HTTP://1RADIO.CA

robmorin, 2021-10-15:
@Chris
Oooo, that's a cool idea!

Rob Morin
Montreal, Canada
HTTP://1RADIO.CA

robmorin, 2021-10-15:
@EricClaeys 
Ya sure let me know when you get a wiki going with that explanation.

A quick question is there a forum somewhere where we can discuss non software related topics rather than fill up the "issues" with non software related  talk?

Thanks, have a great evening.

EricClaeys, 2021-10-16:
@maserowik Mike, for not being a programmer you did one heck of a nice job with your script!

@robmorin Rob, The "Allsky Camera" Facebook page at [https://www.facebook.com/groups/172438633343696](url) is an ok place to discuss non-issues.  It's not specific to Thomas' software so has people using Windows software as well.  I wish GitHub had a discussions forum because I prefer the "Issues" to be true bugs or potential bugs, but we're doing a pretty good job answering questions that are in Issues, then closing them, so most of the Issues are things we need to look at.  A couple months ago there were over 200 Issues and I've been closing outdated ones and resolving others.

Speaking of which, if your original question was answered, would you please close this Issue?

closed 2021-10-16 by robmorin

EricClaeys, 2022-01-05:
@robmorin Rob, I added a page explaining "DAY", "NIGHT", "sunrise", and "sunset" to the Wiki, including how to determine what your values are.

The Discussion link at the top of the page is useful for non-Issue related discussion, like you asked about.

robmorin, 2022-01-06:
Thanks, I'll check it out very soon, I have been busy with work stuff lately, sorry for not being too active.

Happy belated holidays!

Rob Morin
Montreal, Canada
HTTP://1RADIO.CA

EricClaeys, 2022-01-06:
@robmorin Happy New Year to you, too.  When you get a chance to view [the page](https://github.com/thomasjacquin/allsky/wiki/DAY-and-NIGHT-vs.-sunrise-and-sunset), let me know if you have any comments or suggested improvements.  It makes sense to me, but then again, I wrote it.  I want it to make sense to others.