AllskyTeam/allsky #655: Error: one bright and one good image in alternation

Tartouffe, 2021-10-14:
Fresh install today of v0.8.1 .. now got one bright and one good image in alternation.
Any idea how to fix or which setting causes this?

[image]

Tartouffe, 2021-10-17:
@Tartouffe the limit only applies to auto exposure, not manual exposure. 
Aaah. So when i turn autoexposure off i can use it up to 75s? 
...wondering myself. Made screenshots of the settings screen before i made the update and there were autoexposure ON - and the 75s images worked. But when its only this option ... easy to switch off to get my 75s night images again. :)  

Videos-Dir
Hm, then there must be a glitch at the ftp settings. All versions are having (**on the webserver**) a videos dir. Misspelled that 'webdir' fact. Anyway, the ftp copying routine has some hicup:  

Images and video dir is set at the ftp-settings 'LOCAL' as:
IMAGE_DIR="/home/SkyCamWeb/allsky/"

# The remote directory where the timelapse video should go.
VIDEOS_DIR="/home/SkyCamWeb/allsky/videos/"
VIDEOS_DESTINATION_NAME=""
WEB_VIDEOS_DIR=""
[ some lines below: MP4DIR="${VIDEOS_DIR}"  - this is the one i have to use for target path]

All files (MP4, keno, images, startrails) were stored inside and below the daily images-dir. 
But the videos were not copied/moved to the place as set above, the videos-dir, which exists on the allsky webpages as an extra directory.

(BTW: whats the difference between videos dir and web videos dir?)

jsanchez73, 2021-11-01:
Hello,

I am having  the same issue here. Any updates on how to fix it?

EricClaeys, 2021-11-01:
@jsanchez73 try the version of capture.cpp we just uploaded.  If that still gives problems let me know (and include the log file).  I'm working on a version that dampens the changes by implementing an "aggression" setting.  Foe example, it will only make half the expected change in brightness rather than the full change.  I'm also trying to detect wild swings in brightness and deal with them.

Tartouffe, 2021-11-02:
Trying out the updated capture.cpp. But it is (very) bad weather. Had to wait for sun, to reach some brightness level. Could go a few days until I have a result. (Note: version 0.8 expure has the same problem)

EricClaeys, 2021-11-02:
@Tartouffe, are you saying version 0.8 is causing your bad weather?????  We are working on a cloud-removal filter :-)

Are you running with the version exposure mon ON or OFF?

Selse reported good results with 0.8 exposyee ON, but I realize not everyone can run with it ON.

Tartouffe, 2021-11-03:
"Version 0.8 Exposure" was OFF, now ON. 

Had tested this (set ON) with the version before, no difference. 
Lets see what the latest capture.cpp will produce over the day tomorrow with now switched on.

EricClaeys, 2021-11-03:
@Tartouffe, please increase your `Debug Level` to 3 in the WebUI.  It will give me much-needed information, like what was happening with the dark/bright images.

The "stutter" in your GIF is because some images were being skipped due to ASI_ERROR_TIMEOUTs even though you are using the older version 0.7 exposure algorithm.  Try decreasing your `USB Bandwidth` to 40.

The strangest image is the one that doesn't include the overlay.  I have no idea how that can happen.

Tartouffe, 2021-11-04:
First debug level 3 run with the parameters: USB Bandwidth: 40 / debug Level 3 / Version 0.8 Exposure: YES
A too bright image was shot at timecode 15:17:59. Only one. I had to run it on brighter sky (when your "clear sky automation" is working error free ;) ). 

I noticed when a bright image was produced, an image in the set Intervall is missing!
E.g. every 1min one picture, the bright image is 2 minutes old/er, the one-minute-before it is missing. After the bright one, the time Intervall gets a few seconds longer. But also keep an eye on the time Intervall of the Images. 
But had to reproduce/prove this on better conditions. 
For now the sky makes no sense. For the next run at brighter weather i will set it to 60000ms per image for better calculation, wheather there is really a frame before the bright one missing. Then it could be a cumulation of some values or variables.  

Attached the log. Timecode of the wrong one: see above.  
Meanwhile: waiting for better sky. 

[allsky.log](https://github.com/thomasjacquin/allsky/files/7475535/allsky.log)

Tartouffe, 2021-11-07:
Eric, you know what? NOT A SINGLE overexposed picture since the modified parameters 3 day ago. I think the default USB value of 80 (if i remember right?) causes these bright images. Now with 40 the images are how they should be. Also no frames were missing. 
Had to prove this on bright weather. Will keep you updated (waiting for bright sky/sunshine).

EricClaeys, 2021-11-07:
@Tartouffe, that's great to hear.

The brightness of the sky appears to have been changing quite a bit.  The mean brightness of the picture before 15:17:59 was 81 which is below the acceptable range, then the 15:17:59 picture's mean was 129 which is inside the acceptable range.  The next several pictures had mean of 124 - 134 which are all in the allowed range.
If those pictures were way too bright you may want to move the histogram box to include more of the sky.

EricClaeys, 2021-11-19:
@Tartouffe, is it still working well?  If so, would you please close this issue?  Thanks.

Tartouffe, 2021-11-20:
Sunny weather didn't happend these days, so i can't proof it by 100%. But with reduced USB bandwidth no more brigthness issues. It has to be that. 
Thanks for the patience identifying this issue. :) 

So if you got some too bright images, reduce the USB bandwidth. 
I will close here.

closed 2021-11-20 by Tartouffe