AllskyTeam/allsky #658: error on install of allsky

kenkolen-neal, 2021-10-15:
This is a new raspberry pi device. I updated operating system update and an update of sudo.
I did a clean install of allsky and I can see install.sh in the directory.
I run sudo ./install.sh and I receive an error message "This script must NOT be run as root"

maserowik, 2021-10-15:
@kenkolen-neal 
That is correct the developers change the install to **_not_** run the main AllSky as root.  
if you install the Admin gui you **_still need to install it as root_**

kenkolen-neal, 2021-10-15:
Sorry I am new to this, can you walk me throgh what I need to do?I am fillowing Patriot Astro youtube video

Neal Kleinman 
ReMax Advance Realty 
Platinum Club, Hall of Fame
Southern Cross Astronomical Society 
Board Member, established 1922

kenkolen-neal, 2021-10-15:
> @kenkolen-neal, @maserowik is correct that installing allsky no longer requires sudo. Step 4 on the main page says to execute ./install.sh. I will take a look at Patriot Astro instructions - we may not need to reference them anymore since they are somewhat out of date and in my opinion the Wiki should be sufficient.
>
> If you're able to install the software after looking at the Wiki would you please close this Issue?

I installed the software but I cannot open it using a browser.

Neal Kleinman 
ReMax Advance Realty 
Platinum Club, Hall of Fame
Southern Cross Astronomical Society 
Board Member, established 1922

EricClaeys, 2021-10-18:
@kenkolen-neal,
How did you install the software initially?  You had to have been on the Pi somehow, either in via a monitor and keyboard, or via a remote interface like ssh or VNC client.
Use that same message to edit the `/home/pi/allsky/config/config.sh file, setting the CAMERA variable at the top to either ZWO or RPiHQ.

When prompted for your password using ssh, it won't display the characters you are typing or even a "*" like wesites do.  When done entering the password just hit the Enter key.

kenkolen-neal, 2021-10-19:
> @kenkolen-neal <https://github.com/kenkolen-neal>
> I'm in Naperville IL (30 miles west of Chicago). Michael and his wife live in a nice mobile bus so his address is where ever they happen to be. Not sure about Chris but I'm pretty sure he's in the US.
> Thomas is in Canada.
>
>
> This email has been checked for viruses by Avast antivirus software.
> https://www.avast.com/antivirus

I presume you guys are amateur astronomers?

 

Neal

EricClaeys, 2021-10-19:
Michael and I are.  Not sure about Chris but I'm 99% sure he's not a profession astronomer.
I have a remote observatory in NM next to New Mexico Skies that Mike and Linda (?) Rice own.
Michal has an observatory in southwest NM that he's moving about 15 miles down the road.