AllskyTeam/allsky #660: V 0.8 exposure

Clutch57, 2021-10-15:
Last night I used my ASI174MM camera and got some great imagery.  Today I wanted to try out my ASI178MC to get some color.  I ran into an issue were daytime exposure was bouncing around.

.34 ms
209.38 ms
.16 ms
30 s
.34 ms
1.05 s
.50 ms
 482,03 ms

I went into the advanced settings and found V 0.8 exposure and turned it on.  It is working great during day time.
I just wanted to give some feed back.

EricClaeys, 2021-10-16:
@Clutch57 Thanks for the feedback.  So you're saying that with `Version 0.8 Exposure` set to Yes the exposures don't jump around but with it set to No they do jump around?  That's very interesting.

If that's correct, I'd like to ask a favor.
Basically, you'll run both ways with debugging turned on, and send the log files to me:

During the day when it's clear or fully cloudy (so the brightness won't be changing much), stop the service, then execute `sudo rm /var/log/allsky.log`.
In the WebUI, set the `Debug Level` to 4 and `Version 0.8 Exposure` to **No**, then click on `Save changes`.
Let it go for about 30 minutes.
Make a copy of the log file:  `cp /var/log/allsky.log allsky-0.7.log`.

Now, stop the service and execute `sudo rm /var/log/allsky.log`.
In the WebUI set `Version 0.8 Exposure` to **Yes**, then click on `Save changes`.
Let it go for about 30 minutes.
Make a copy of the log file: `cp /var/log/allsky.log allsky-0.8.log`.

Attach (don't copy/paste) both log files to your reply in GitHub, and let me know how both runs went.
Restore your settings to what you had before the test.

I'm really curious to see how the Version 0.8 Exposure impacts the "bouncing" exposure.

Clutch57, 2021-10-16:
Attached are two files.  Not 30 minutes long each but long enough I think you can see what is happening.  I'll do longer later today if you need them.

[allsky-0.7.log](https://github.com/thomasjacquin/allsky/files/7357611/allsky-0.7.log)
.
[allsky.08.log](https://github.com/thomasjacquin/allsky/files/7357612/allsky.08.log)

Clutch57, 2021-11-10:
Do you want me to use V7 or V8 settings.  Likely won't get to trying it until Friday.

EricClaeys, 2021-11-10:
@Clutch57, please try Version 0.8 Exposure first, and only set it to "No" if you get ASI_ERROR_TIMEOUTs with it On.

Clutch57, 2021-11-12:
> 
> 
> @Clutch57, please try Version 0.8 Exposure first, and only set it to "No" if you get ASI_ERROR_TIMEOUTs with it On.

The sky was clear with some clouds so I worked my way though the instructions.  The weather turned to solid clouds and rain while I was preparing the test.  I'm not sure if this is a good test case as the sky is currently a close approximation of a grey card.

[allsky.log](https://github.com/thomasjacquin/allsky/files/7529748/allsky.log)

I can run this again on a nicer day.

Thanks,

EricClaeys, 2021-11-12:
@Clutch57, can I see the video created from the images taken that the log file is from?  Basically 1 pm - 2 pm today.
One way to do this is the create `allsky/images/test` and copy files from allsky/images/20211112 taken between 1 - 2 pm to the "test" directory.  Then run `scripts/generateForDay.sh -t test`.  When done the timelapse video will be in allsky/images/test.

It shouldn't be that big so you can attach it here (append ".txt" to the name).

Clutch57, 2021-11-19:
I wasn't saving daylight files.  I started getting black lines in the images that were being captured and thought I had water getting into my camera so I brought it in.  Last night I put it back out to capture the eclipse and I still had the lines even though I didn't see lines when I had my camera inside and looked at it using Sharpcap.  I copied in the capture program I was using before this test and they went away.

I'm going to close this as I have a different camera I want to run so I can use the 178 for other things.

closed 2021-11-19 by Clutch57