AllskyTeam/allsky #676: sunwait compiler warnings

EricClaeys, 2021-10-17:
Just did an install of "allsky" and got this when Building sunwait:
~~~
make[2]: Entering directory '/home/pi/allsky/src/sunwait-src'
gcc -c -Wall  sunwait.cpp -o sunwait.o
sunwait.cpp: In function ‘int main(int, char**)’:
sunwait.cpp:665:62: warning: value computed is not used [-Wunused-value]
     if (arg[0] == '-' && arg[1] != '\0' && !isdigit(arg[1])) *arg++;
                                                              ^~~~~~
gcc -c -Wall  sunriset.cpp -o sunriset.o
sunriset.cpp: In function ‘void sun_RA_dec(double, double*, double*, double*)’:
sunriset.cpp:151:18: warning: variable ‘zs’ set but not used [-Wunused-but-set-variable]
   double xs, ys, zs;
                  ^~
~~~

I suggest we fix this before 0.8.5 to avoid confusing users.

ckuethe, 2021-10-17:
Not our bug. We no longer keep a copy of sunwait in our repo since @linuxkidd fixed the submodule.

closed 2021-10-17 by ckuethe

EricClaeys, 2021-10-17:
@ckuethe 
@linuxkidd 
I don't agree that it's not our bug.  sunwait.cpp and sunrise.cpp were in the distribution I got a couple hours ago so from a user's point of view, the code is "ours" and so are the confusing warnings.  What is a user supposed to do with those warnings?
I can edit those files and fix the warnings.

EricClaeys, 2021-10-17:
I guess I can't edit the files, so can the Makefile do a `> /dev/null 2>&1` ?

ckuethe, 2021-10-17:
They're a submodule - they come from an entirely different repo owned by someone else.