AllskyTeam/allsky #698: ALLSKY_HOME not found when new install on raspi

int48, 2021-10-20:
when the install starts with install.sh a error message appears that allsky_home path is not set. I help myself by adding the path on variables.sh on a new Raspi os installation. I found out a another way is to store the path for allsky_home in file called bashrc but this have no effect on intall.sh. What is the correct solution or what is to do, i a am new on linux shell scripting.

shiner501, 2021-10-20:
I have same issue after doing a reinstall:
Pull it down from GitHub, cd to allsky then ./install.sh and I get:

**install.sh: ERROR: ALLSKY_HOME not set!  Exiting...**

EricClaeys, 2021-10-21:
We posted a new allsky/install.sh that fixes this issue.  It sets the ALLSKY_HOME variable at the top of the script.
Let us know if you still have an issue with this.

closed 2021-10-21 by EricClaeys