AllskyTeam/allsky #750: Nice to have Request

calberts, 2021-10-31:
Hoi 

I have a question regards new features to be added

1. on the website of the allssky I like to have the last video/images displayed first and , is this already possible
2. Create a notification message (with MQTT) when it's clear, I think it should be possible (The user should be able  to select box (and size) where this will be measured.
(I decide on clear skies always checking the all-sky also, besides my weather station) 

Thanks in advance

Chris

maserowik, 2021-10-31:
@calberts 
Check out the following and see if this answers you first question.
https://github.com/thomasjacquin/allsky-website/issues/54

@EricClaeys 
Can this be added to the wiki if not there
https://github.com/thomasjacquin/allsky-website/issues/54

calberts, 2021-10-31:
The suggestion didn;t work 

thomasjacquin/allsky-website#54

Chris

EricClaeys, 2021-11-01:
@calberts, @maserowik,
The ability to sort is on our [Project "to do"](https://github.com/thomasjacquin/allsky/projects/37/edit) list.

EricClaeys, 2021-11-01:
@calberts, I added the "send message when clear" to our Project list as well, so we'll track both requests there.  Hence, I'm closing this Issue.

closed 2021-11-01 by EricClaeys