AllskyTeam/allsky #755: Appear to be a issue on my end with Videos

maserowik, 2021-11-02:
I just noticed this a few day ago and thought it was due to the position of my camera.  On my timelapse videos they appear to flicker. 
Here is a sample of the issue I am seeing
http://n3bsq.com/allsky/videos/allsky-20211031.mp4
attached my settings
[settings_ZWO.json.txt](https://github.com/thomasjacquin/allsky/files/7460758/settings_ZWO.json.txt)
[config.sh.txt](https://github.com/thomasjacquin/allsky/files/7460759/config.sh.txt)
[camera_options_ZWO.json.txt](https://github.com/thomasjacquin/allsky/files/7460760/camera_options_ZWO.json.txt)

Thanks
Mike

EricClaeys, 2021-11-02:
@maserowik, all the flickering is at night when the software uses ZWO's exposure mode, so there's nothing we can currently do about it.  We may eventually use our own exposure algorithm at night as well, but I wouldn't expect that anytime soon.

A fun test would be to set "angle" to either 23.9 or -23.9 to effectively make the whole day "daytime" as far as the software is concerned. This would cause the histogram algoithm to be used all the time.  I wonder how it would handle night shots.

EricClaeys, 2021-11-05:
@maserowik, Mike, @linuxkidd pointed out my mistake saying "23.9 or -23.9.  I was thinking hours when "angle" is an angle.  Try +/- 89.9 degrees.  Basically you want everything to be "daytime".  Note I haven't tested this myself...

maserowik, 2021-11-08:
@EricClaeys 

Not sure what happened.  but the flickering went away.  Closing this issue.

closed 2021-11-08 by maserowik