AllskyTeam/allsky #756: Confusing menu option text

Tartouffe, 2021-11-02:
At the  Configure Camera Settings  it says as first:
_DAYTIME SETTINGS
50 - 160. Changes the amount of light in the image.Low value = dark; high value = bright. **Only works with auto-exposure.**_

The phrase _Only works with auto-exposure_ should be complemented with '(enable hidden option to change)' or something like this. Otherwise, not so familiar with allsky and the existence of a hidden options switch, you search and will not find a daytime auto exposure. Or hit instead the NIGHT auto exposure, which causes other 'problems' hours later beside the default _everything on automatic_...

Just not to confuse beginners.

EricClaeys, 2021-11-02:
4rr4n NB bbbbbbb NB j BBC
.b uh
o.if
N

Tartouffe, 2021-11-02:
> 4rr4n NB bbbbbbb NB j BBC .b uh

??
I think your cat has occupied the keyboard 😂

EricClaeys, 2021-11-02:
@Tartouffe sorry about that.  I think it was my butt...

Thanks for the feedback.  It will help make the software and documentation better.

EricClaeys, 2022-06-30:
Closing - this request was added to the Projects list - see the `Projects` link at the top of any Allsky GitHub page.
Project Title: **Improve message about "Only applies if 'Auto-Exposure' is enabled" #756**

closed 2022-06-30 by EricClaeys