AllskyTeam/allsky #757: lighttpd.conf for gui has /home/pi hard coded

JosWennmacker, 2021-11-03:
I run allsky on Ubuntu 21.10 using user allsky.
The lighttpd.conf for gui has /hom/pi/allsky and /home/pi/allsky/images hard coded on lines 27 and 28, so neither 'Live View' nor 'Images' nor 'Editor' work. Changing pi to allsky on both lines and restarting lighttpd things are ok.

EricClaeys, 2021-11-03:
@JosWennmacker Line 73 of gui/install.sh should convert `/home/pi/allsky` to whatever directory your `gui` directory is in, so if you have allsky in `/home/allsky/allsky` it should use that.

Did that not happen?

JosWennmacker, 2021-11-03:
Obviously not. But I removed the gui and installed it again. Now it did change the directory. Problem solved.

closed 2021-11-03 by EricClaeys